hrvach/deskhop #1: Additional FAQs

mossholderm, 2023-12-26:
These are some questions I had, that probably would be good to add to the FAQs. I don't actually have the answers :)

1. When the active screen is changed via the mouse, does the keyboard follow (and vice versa)?
2. Will this work with keyboard/mouse combo dongles, like the Logitech Unifying receiver?
3. Will this work with wireless mice and keyboards that have separate wireless receivers (one for the mouse, another for the keyboard)?

KanyonKris, 2023-12-26:
1. Yes, the keyboard input follows the mouse (active screen). See the animated GIF.
2. It won't work with mouse and keyboard combined on one USB-A input. There are 2 different firmwares for each Pico - one for keyboard, one for mouse. It might be possible to have a unified firmware on both Picos that can recieve and process both keyboard and mouse input, but that is not how the current firmware works. So must use a mouse connected to the mouse USB-A port, and a keyboard connected to the keyboard USB-A port.
3. Yes, it should work with wireless mice and keyboards that have USB-A dongles (receivers), as long as they use standard USB HID protocols and not some custome communication that requires a driver on the computer.

hrvach, 2023-12-26:
Thanks @KanyonKris for providing a quick and accurate response to the questions. Much appreciated!

I don't have any Logitech gear to test on so I don't know how they implemented it - my guess is that it enumerates as both keyboard and mice, but I'd have to check first. If I'm able to borrow one, I'll try it!

mossholderm, 2023-12-26:
Just to continue the train of thought... it might be worth it to move to a unified firmware anyway... it simplifies the design (one firmware for both Picos), and potentially opens up other possibilities.  

For example, some people might have multiple pointing devices (mouse, keyboard, and tablet or touchpad type device or even a second mouse).  Having the ability to have them all connected via a hub, or 2 in one port, and one in the other would increase the usability.

a5m0, 2023-12-26:
Could also make this scale to more than 2 computers with a more unified firmware.

pacjo, 2023-12-27:
Hey, I have additional question. Is it possible to add the button anyway? I know it's not the goal of the project, but it could be considered as additional functionality.

I personally prefer the automated switching (I will probably switch to inputleap when Wayland support is finished), but since the minimum PCB order is 5 I'm thinking about building one for a family member and my concern is that it will be useless for them if they don't like automatic switching.

HoustonDad, 2023-12-27:
I saw the FAQ issues on this, so I wanted to ask, would this support ChromeOS / ChromeBooks?

hrvach, 2023-12-27:
> I saw the FAQ issues on this, so I wanted to ask, would this support ChromeOS / ChromeBooks?

I believe it should work just fine, but I have no way to know since no Chromebooks to test on :)

hrvach, 2023-12-27:
> Hey, I have additional question. Is it possible to add the button anyway? I know it's not the goal of the project, but it could be considered as additional functionality.
>
> I personally prefer the [...]

I have written this down for v2 of the PCB, so I'll try to include a place where a microswitch would go and it's up to the user to install it or not. That way it wouldn't increase the cost of device and would still provide you an option to use whatever is more convenient.

eproxus, 2023-12-27:
Another question: does it work if one of the computer is off? I usually don't have both on. Can it still switch between them in that case?

hrvach, 2023-12-27:
> Another question: does it work if one of the computer is off? I usually don't have both on. Can it still switch between them in that case?

Great question - sadly, no, because both Picos have to be powered and they get their power from the respective computers they are connected to. Let me add that to the FAQ so nobody gets disappointed!

luciusf, 2023-12-28:
> Great question - sadly, no, because both Picos have to be powered and they get their power from the respective computers they are connected to. Let me add that to the FAQ so nobody gets disappointed!

I'd assume, that in many setups a device2 (eg. a work laptop) would not be present at all times. In order to not change a clean keyboard+mouse cable arrangement, the DeskHop so conveniently makes possible, could we then power the second pico with a standard charger? Or would this mess with the firmware?

hrvach, 2023-12-28:
> I'd assume, that in many setups a device2 (eg. a work laptop) would not be present at all times. In order to not change a clean keyboard+mouse cable arrangement, the DeskHop so conveniently makes [...]

Right now there is a dependency that the device first needs to successfully connect to its output for the host to start working. I have to admit I haven't considered the scenario you are suggesting - but it makes a lot of sense and I'll try to fix that :) Will also look into the option of connecting both devices on one side with a hub - the current solution was chosen because it's cheap and doesn't depend on anything else :)

luciusf, 2023-12-28:
> Right now there is a dependency that the device first needs to successfully connect to its output for the host to start working. I have to admit I haven't considered the scenario you are suggesting [...]

Thank you for looking into this :). I believe this device is desperately needed, and any way of reducing caveats and making it more convenient is much appreciated!

hrvach, 2023-12-28:
> Thank you for looking into this :). I believe this device is desperately needed, and any way of reducing caveats and making it more convenient is much appreciated!

If it helps anyone else be more productive and less annoyed with their workspace, I'll be very happy! :)

KanyonKris, 2023-12-28:
> Another question: does it work if one of the computer is off? I usually don't have both on. Can it still switch between them in that case?

I opened [issue 9](https://github.com/hrvach/deskhop/issues/9#issue-2058792277) to discuss this issue further.

WmOne, 2023-12-31:
hi, great job.
as i'm new on Raspberry (and i never used them), but i'm able to work with electric components (and soldjering), can u please:

1) specify wich rasperri PI shoul i buy (mybe could we have a moel number ? i looked at amazon (IT) and i found a lot of boards very different from the one posted here (and in most case, without USB-A , but microusb. maybe some link to the correct article ?)
2) have a tutorial on how to connect the two boards and wich material we must buy to do it (models, or links)
3) have a tutorial on how to build the image (for those, like me, than never did it)

As i told, i'm a very newbye about Raspberry
tanks a lot

hrvach, 2023-12-31:
>     1. specify wich rasperri PI shoul i buy

It's not a full blown Pi computer board but a [Raspberry Pi Pico](https://www.berrybase.de/raspberry-pi-pico-rp2040-mikrocontroller-board) microcontroller, much more affordable and super cool too.

>     2. have a tutorial on how to connect the two boards and wich material we must buy to do it (models, or links)
>     3. have a tutorial on how to build the image (for those, like me, than never did it)

This is a good idea, I'll try to make a step-by-step assembly tutorial to make it easier to put together.

driftvk, 2024-01-01:
Thanks in advanced! Super excited to see this evolve, I desperately need something like this for my work.
 I'll put an order with JLCPCB including assembly as I'm not keen on SMD soldering. I will download KiCAD at some point to generate the BOM&CPL files, this is all new to me :)
https://jlcpcb.com/help/article/81-How-to-generate-the-BOM-and-Centroid-file-from-KiCAD

hrvach, 2024-01-02:
> Thanks in advanced! Super excited to see this evolve, I desperately need something like this for my work. I'll put an order with JLCPCB including assembly as I'm not keen on SMD soldering. I will [...]

Readme updated with a link to a short "how to assemble" video. It's not hard at all to assemble it yourself, maybe you'll feel encouraged :)

WmOne, 2024-01-02:
i'm trying to order the pbc from JLC
i uloaded all the Gerber folder file, and this is what i see is:
[image]
is all correct ?
or have i also to click on PCB Assembly (i don't understant what is it for...)

hrvach, 2024-01-02:
[image]

This was mine - looks OK. Just make sure it's 1.6 mm and you should be fine. 

PCB assembly is to have them assemble the board for you (can be some extra cost due to if using non-basic parts that need pick-and-place component feed change etc). You buy parts at JLC prices (which are kinda OK) and they assemble and inspect your board professionally. If you can wait, you can save on shipping with the slower option so it turns out reasonable.

WmOne, 2024-01-05:
the two pc are to be connected via micro usb port, so i need a micro usb cable (one to the first pc, the second to the other) , isn't it ?

hrvach, 2024-01-05:
> the two pc are to be connected via micro usb port, so i need a micro usb cable (one to the first pc, the second to the other) , isn't it ?

Yes, just an ordinary microusb cable you probably have lying around. Please note, some cables are for charging only (to save cost on copper?) so if your device doesn't show up when holding the BOOTSEL button and plugging in, try another cable. Cheers!

thrazznos, 2024-01-10:
So excited about this project! I bought the components and was testing as I was assembling. I am getting behavior where when I drag the compiled binaries onto the pico, it disconnects from my pc and when I reconnect it the file seems to be missing (crash?). I don't have the IC hooked up because its still in the mail. Am I doing something wrong? 

So impressed with this project and cant wait to get it working! Thanks!

hrvach, 2024-01-10:
> So excited about this project! I bought the components and was testing as I was assembling. I am getting behavior where when I drag the compiled binaries onto the pico, it disconnects from my pc and when I reconnect it the file seems to be missing (crash?). I don't have the IC hooked up because its still in the mail. Am I doing something wrong?
> 
> So impressed with this project and cant wait to get it working! Thanks!
> 
> Edit: After more reading I understand that loading a file while in mass storage mode reboots the device to run the files, so I shouldn't expect to see them when I remount it afterward, so i'm probably fine!

That's how it's supposed to work! You hold the button and plug in the cable, that puts the pico in bootloader mode (button disables chip select line for the flash IC) and now it's a glorified usb stick. You drag the uf2 file there, it gets written to flash and device reboots to application mode and runs your code. When you remount it again, it basically waits for another .uf2 firmware file you might have for it, otherwise you won't see anything there. Don't worry, so far - so good. Good luck and let me know how it goes!

WmOne, 2024-01-20:
just a question, as i'm not an electrician nor a technician
Does the ADuM1201BRZ have a verse ? how can i know in wich position (rotation...) have i to soldjer it into the pbc ?

hrvach, 2024-01-20:
> just a question, as i'm not an electrician nor a technician Does the ADuM1201BRZ have a verse ? how can i know in wich position (rotation...) have i to soldjer it into the pbc ?

The little dot on the chip (pin 1) goes towards the B board (mouse). Silkscreen has a longer line marking pin 1, but that's clearly not obvious enough, sorry about that. The subsequent board revision will have a large visible dot. Also, it will tell you which pico pins to solder and which are OK to skip.

Check the [assembly guide video](https://www.youtube.com/watch?v=LxI9NYi_oOU) for instructions how to put it together.

Tom1827, 2024-02-03:
Thanks for this - I'm planning on combining one of these with  a PBP monitor in the near future (which I assume should _just work_).

A couple of questions:

- Is there (could there be?) an option to quickly switch back to a single computer mode, if one of the two is not needed?  (In my PBP use case, I'd turn off a work computer at the end of the day, and then devote the whole screen to my personal computer.)
- Is there any noticable latency with this approach, versus a direct cable?  I guess there must be some small amount, even if imperceptible?

hrvach, 2024-02-04:
@Tom1827 thanks for the questions - let me answer without quoting to make it a bit more readable.

So, first question - if you want just one computer, you have a reasonable chance of getting it working with a cheap usb hub plugged into the pico you plan on using for single computer mode. Hub support is not perfect, but I've tested it on my devices and it kind of works. Some usb3.0 hubs reportedly don't, so ymmv.

If you don't physically unplug the computer but only shutdown, it would most likely work as most computers nowadays keep providing USB power. It's easy to power one pico through the other, but this approach was chosen on purpose to provide maximal safety and full isolation between outputs. If one of your computers has a catastrophic failure, it should not kill the other through the box in any scenario. I am looking into reworking the power section by e.g. just isolating one USB port for a future board release, maybe with ADuM3160 USB isolator but that one is not super affordable.

I opted out of sockets/headers to bring the cost (and box size) down. It's a *very* good idea though from an availability point of view - "just add your own picos". I can't notice any latency myself, I'll try measuring.

OrpheeGT, 2024-02-04:
I'm currently using a desktop computer, and a laptop that I un/plug time to time.

The Board A is plugged to the laptop (right screen)
The Board B is plugged to the desktop computer

Luckyly for me, the keyboard has an additional USB cable plugged to a powered hub. and It actually gives back power from keyboard to board A when laptop is unplugged.
As the mouse is plugged to board B, which is plugged to desktop computer always on, for me, it currently works even if I unplug the laptop.

I just need to Right Shift + L to lock on desktop computer while laptop is unplugged.

hrvach, 2024-02-04:
Implementing heartbeat + switch lock enable when the other side is not detected has been on the to-do for a while now.
Btw, what's also under construction is a single pico w version that connects to output B via bluetooth and consequently has no dual-power requirements. :)

hammaal, 2024-03-04:
Hello. As far as I am gathering now there is no way this will work with two screens on windows correct? Is there even any possible solution that maybe can be worked out? I have a windows laptop and a windows PC with 2 monitors on the PC. Thanks for the answer in advance.

hrvach, 2024-03-04:
Correct. For the time being, windows with multiple screens will have the mouse confined to the main display. Possible solutions are being tested, but the progress is slower than expected due to personal/work constraints.

spanzetta, 2024-03-05:
> Correct. For the time being, windows with multiple screens will have the mouse confined to the main display. Possible solutions are being tested, but the progress is slower than expected due to personal/work constraints.

I am available to do some beta testing on this argument .. if you want/you think it can help..
I realized 2 devices .. so I can use 1 for beta "testing"
Just let me know..
I have one Win10 Laptop (+ external monitor) + a Samsung Smart Monitor M50C (Tizen 7.0) and currently it works but on the WIndows Laptop the mouse is confined to main display..

hrvach, 2024-03-05:
Thank you for the offer - much appreciated! Will work on multiple displays soon, this seems to be the most requested feature so it makes sense to prioritize it.

jstorrs, 2024-08-28:
Is there a way to use two "keyboards" and one mouse? I have an inputstick and use the Keypass2Android inputstick extension (https://github.com/inputstick/KP2APlugin) to type passwords stored in keypass2android on my phone. inputstick is USB dongle that presents as a dumb USB keyboard to devices when plugged in (communicates with phone using an encrypted bluetooth protocol).

hrvach, 2024-08-28:
I haven't tested it so I'm just talking out of my ass, but I think it might even work out of the box if you use some dumb usb hub on one side and e.g. plug a mouse and that stick in mouse port. Both boards know the active output so the one on the mouse side would send keystrokes over serial and should work.

j-f-d, 2025-06-25:
Is there a way to keep the screen awake on the computer I'm not working on currently?
While I'm at my desk working at one computer, I don't want the other computer to go into screen saver, as (due to policies) I need to use authenticator to unlock each time. Presumably this could work like the existing screensaver mode, but only if active, rather than always, or only when inactive.