hrvach/deskhop #3: Multiple monitors on each PC

shantur, 2023-12-27:
Hi,

Excellent idea and implementation.
As of now, does it support multiple monitors with different resolutions connected on each PC ?

Thanks

hrvach, 2023-12-27:
Resolutions should not be a problem, but I never tried it on a multi-monitor setup. I'll try and report back :)

fayfay78, 2024-01-08:
hi there, any news about multiple screens?

hrvach, 2024-01-08:
> hi there, any news about multiple screens?

I believe it should work, provided that the orientation/layout is correct, but I'm traveling and have no way to test yet. 

My assumption is that if you have 2 machines with 2 FHD screens each, let's call them A and B, setting them up like:
```
-------------------------------
B left | B right | A left | A right |
-------------------------------
```
would be identical as having e.g. 3840 x 1080 screen on the left and a similar one on the right. However, there are 3 major OSes to try on, as well as mismatched resolution scenario (one "left" monitor UHD and "right" monitor FHD). Unless somebody beats me to it, I'll try to see what happens in a couple of days :)

jcassan, 2024-01-09:
Hi,
Thank you for providing such an amazing project.
I just tried what you mentionned :
---------------------------------------
| B left | B right | A left | A right |
---------------------------------------
My mouse is swapping well from B right to A left but other screens do not work.

I have a "logical barrier" at borders of the main screen on both pc. I also tried a setup with vertical stack resulting in the same behavior. If I use my trackpad to get out of the barrier and then re-use the mouse connected to the pico, cursor is being teleported to the main screen.

All screens are FHD and pcs are running windows 10.

My soldering skills are pretty bad so let me know if any bad solder could make the picos behave like this.
I also should mention the keyboard does not work on pc connected to board B but this might be a soldering issue somewhere.

hrvach, 2024-01-10:
I've tested the multiple screen scenario and it works for me with no issues [tm] (as suggested above). 

I don't know what a logical barrier is in this context, please be so kind to explain (I have nothing similar configured, just a plain ubuntu with two monitor output on one side and the mouse just skips screens as expected).

if your mouse works on PC connected to A but keyboard doesn't work on board connected to B, check solder/contacts on pins 7 and 2 of the ADuM1201 chip. Also check if you soldered pin 16 on pico A (marked GP12 on solder mask) and pin 22 on pico B (marked GP17 on solder mask).

jcassan, 2024-01-10:
Sorry for not being precise on my explanation, I will try again.

From B right to A Left, everything works just fine.
When I try to move the mouse from B right to B left, the mouse is blocked at the left border of B right. I am not able to move the cursor to B left. (Same behavior for A device)

I am using two laptops for my tests. I can use my trackpad to move the mouse to my second screen (eg B right to B left). When I do so and use the mouse back, cursor is being teleported back to B right. 

Anyway, I think solders are not clean enough to learn much from my tests. I will try to assemble a second unit this week end as I bought enough component to do so. I will let you know how it turns out.

hrvach, 2024-01-10:
Now I understand, sorry for being slow :) It seems Microsoft KB5003637 update to Win 10 [brok^H^H^H^H changed the HID system behavior](https://answers.microsoft.com/en-us/windows/forum/all/kb5003637-ms-usb-hid-driver-absolute-positioning/90be17fd-45f3-4de4-8e5b-cc1c465324b5). I have no idea if it's going to be reverted at some point. In theory this could be worked around with a horrible hack by having a relative mode mouse just for screen hopping, or perhaps easier - activate accessibility mode to control mouse with numpad and then patch the firmware to send a numpad right/left when you reach internal screen border.

Thank you for finding this and sharing,, I had no idea Windows behaved this way!

jcassan, 2024-01-11:
Wow, well spotted ! Let me know if you want to build a new custom firmware someday to correct this. I would be happy to test it if needed.

hrvach, 2024-01-11:
> Wow, well spotted ! Let me know if you want to build a new custom firmware someday to correct this. I would be happy to test it if needed.

I'm focusing on basic functionality first, to cover most users and most hardware. Then we tackle the edge cases, like specific OS multi-desktop handling. I think it should be doable :)

hrvach, 2024-01-13:
As of now, it works on Linux, Windows 10 might have issues after KB5003637 and so might macOS.

Added to the list of feature requests/ideas, as this requires some workarounds to support.

https://github.com/hrvach/deskhop/discussions/27

hrvach, 2024-01-18:
@Trekky12 this is a great find, thank you. There is very little actual info on this anywhere.

I have some ideas how to address that potentially, and avoiding any software would be great (because very often you can't install drivers, and for good reason). 

This brings up the complexity, as you have to model multiple screens on multiple outputs, but it's doable. Definitely in the roadmap :)

Trekky12, 2024-01-19:
I totally agree with you that it's best done without needing to install any software (especially on work devices where this is often not possible).

Great to hear that this is on the roadmap 👍

forrestguest, 2024-01-23:
Would it be possible to have a mode where the mouse isn't used for hopping?  I am mostly interested in the fast switching and removing the need to restart an electrical connection with each USB device when switching desktops.  I don't really need the cursor based hopping.  Happy to open another issue if needed.

hrvach, 2024-01-23:
> Would it be possible to have a mode where the mouse isn't used for hopping? I am mostly interested in the fast switching and removing the need to restart an electrical connection with each USB [...]

You can do this by hitting Right CTRL + L to "lock" the mouse to the current screen and not have it hop. Would this be something along the lines of what you're after?

forrestguest, 2024-01-24:
That may work, I meant more along the lines of suppressing the absolute positioning and just passing along the mouse inputs.  But thinking it through, you'd still have to be able to pass something through the pico, so I probably just need to be patient and wait for MS to roll back that KB

infosecconsultant, 2024-02-16:
I've got what's probably an odd setup with multiple monitors/displays.

Currently I've got | Windows-1 | MacOS-1 | Windows-2 |
And underneath MacOS-1 is | MacOS-2 |

Currently, I can't seem to 'reach' MacOS-2 (the inbuilt MacOS display) from MacOS-1 or any other display. 

I use a custom mechanical keyboard and I don't actually have a Right CTRL key, I went to https://usevia.app/ to modify that, but the keyboard isn't detected (Which makes sense and I can plug it in and do that, but just thought I'd point that out as well) 
 
Is there anything I can configure to allow this or is the solution gonna be Win1|Win2|Mac1|Mac2? 

(Great project btw, and if anyone needs a spare board, I got a couple dozen made to gift to friends and I've still got spares)

infosecconsultant, 2024-02-17:
> I feel almost embarrassed how basic my two-monitor setup is compared to some of you folks! :) I'm sorry but I simply have to ask out of curiosity - can you can share a picture of your setup? :)))

Sure, I've kinda redacted a bit of stuff out of it though. The mount is custom built from aluminium extrusions for my 3x 27" monitors. 

[image]

> work was super-busy lately and left zero energy to work on personal stuff.

No worries. I realise this is a pet project of yours. If you ever open up a link for donations though, please let me know. This is a game-changer of a device for me :) 

This isn't even my ideal setup, I'd actually like to chain a couple of other laptops and a raspi which is what I used to use Synergy for but a hardware based solution (with a focus on security like you seem to have) is way more desirable. That said, the config files for synergy defining the layout was very handy and an a suggestion, perhaps that's a way to go about things (define resolution and position in a text file on the boards?).

I realise i'm probably the minority by a long shot here but I'll be following developments closely. If there's anything I can test please let me know! 

I also should have mentioned in my original port Windows-1/Mac-1 can't reach Windows-2 either. So I'm confined to Windows 1 and Mac 1 (middle and left screens in that picture). 

Thanks!

hrvach, 2024-02-17:
@infosecconsultant this setup is just ... wow. I feel like a caveman now! Actually there is something you could test for me. Please test windows and mac like this: try plugging in another mouse and when you reach the right edge of screen 1 using deskhop, move the "other" mouse to the right a bit to hop on screen 2. Then continue moving the normal mouse connected through deskhop. What happens? Are you now stuck on screen 2 or returned back to screen 1? Screen 1 and screen 2 here mean multiple screens on the same output/os.

This would be valuable info to determine how to try addressing this multiple monitors issue.

Thanks in advance!

infosecconsultant, 2024-02-18:
Ahhh thanks lol. It took a while to find what worked great for me and honestly, there was a lot of building my own solution for things because there is no commercial alternative 🥲 Some people regularly use upwards of 6 monitors (stock traders, security guards, people in the matrix) and some people are happy with just one, whatever works 🤷 

Yeah so if I use a second mouse on windows, it lets me scroll around on that screen no worries, but as soon as I touch the deskhop mouse it snaps back to the first monitor.

MacOS on the other hand lets me switch between screens with the trackpad and I end up 'stuck' on that screen with the deskhop mouse. (it doesn't switch me back to the first monitor although it does reset the position of the cursor on the new screen to the last known position of the original screen).

hrvach, 2024-02-19:
That's great information, thanks - so macos could benefit from that hypothetic workaround but windows would still remain broken. Oh well, it's either installing a driver for windows or going back to relative mode there. 

There seem to be no elegant solutions here :(

infosecconsultant, 2024-02-19:
Well, a work around for Mac would definitely be better than nothing. 

A driver is less than ideal. Is there any chance there could be some client side agent on windows? Also not _ideal_ but getting FOSS drivers to work on Windows is not fun. It seems KB5003637 addressed some HID specific vulnerabilities so I'm not really in a position to roll it back either. Are there any other potential solutions?

infosecconsultant, 2024-02-19:
I can potentially send you some additional monitors if that would help? Your setup would have to support that though.

hrvach, 2024-02-19:
> I can potentially send you some additional monitors if that would help? Your setup would have to support that though.

Nah, I couldn't take your monitors. I'd have nowhere to put them! I need to find something tiny that could fit on my desk, or use 1-2 cheap usb hdmi capture devices from Amazon. Those seem to be the most promising approach, as they don't take any additional space and you can fake 2 more screens easily.

infosecconsultant, 2024-02-19:
Well, alternatively, it is possible to buy ~5-8 inch HDMI LCD's for fairly cheap (usually used with Raspberry Pi's but would work with any other OS as well).

galooph, 2024-02-19:
Looks like commercial projects are struggling with this, too, see [https://forum.level1techs.com/t/usb-roaming-mouse-km-switch-monitor-question/203273](url).

hrvach, 2024-02-22:
Yep, it's not an easy problem to solve, especially when your premise is not installing any drivers. I have a few ideas to try, maybe one of them will work OK.

AJolly, 2024-03-13:
[image]
@infosecconsultant keyboard twins! I created a magnetic suspension for mine + tpu keys. Also used to be an infosec guy.

Came here because I want a kvm solution that doesnt lag, which input director, synergy, barrier, etc all do.  (mouse movement latency, not switching latency).
But I obviously need multimon support.  Input director has the most flexible multimon positioning of anything I've found.

I'm ok with a hybrid hw + software solution though, I'm thinking about essentially 
1) Software to detect when at edge of screen.
2) Find a software controllable KVM - maybe trigger it via a hotkey via a pi or something, or ill have to check if theres a way to get bidirectional comms via QMK or VIA.
3) some software script to sync up the mouse position for that transfer, then let the rest of the movement be handled via direct USB hid input.

jalmeroth, 2024-03-18:
> Would you be interested to do some testing?

@hrvach I would be willing to do multiple screen-testing on MacOS too (my setup is here: https://github.com/hrvach/deskhop/issues/56), if it's of any help.

lvntbkdmr, 2024-03-21:
is there any software solution (workaround, patch etc.) for windows 10 as it currently disables the second monitor when desktop is connected

hrvach, 2024-03-21:
> is there any software solution (workaround, patch etc.) for windows 10 as it currently disables the second monitor when deskhop is connected

No software solutions are planned - the point of this is to do it in hardware as a self-contained thing. Windows workaround will basically be: for desktop 2 and beyond switch to relative mouse mode. Switching back from win screen 2  to screen 1 will not be very accurate but that's why it's a workaround. Work is crazy these days, so progress is slow... thank you for your patience.

lvntbkdmr, 2024-03-23:
> No software solutions are planned - the point of this is to do it in hardware as a self-contained thing. Windows workaround will basically be: for desktop 2 and beyond switch to relative mouse mode. [...]

Ok, so is it possible to disable the instant switch when mouse movement is detected on the monitor edges and instead rely on only shortcut (some keyboard keypresses) and make the switch, but also be able to use 2 monitors on a windows pc? I do not know whether it is related with edge detection or not, but maybe at least we can use deskhop as a regular USB switch with a shortcut instead of an external hardware switch to trigger switching ?

Trekky12, 2024-03-27:
Thank you very much for the experimental code for multiple monitors @hrvach!

I have two secondary screens on each of my Windows-PCs and would like to give some feedback:
In general it is working very well 🥳 

Sometimes when moving the mouse from the secondary to the primary screen the mouse jumps back to the border where the secondary and the primary screen meet. It seems to occur after a specific time and not on a specific position, so when the mouse is already on the middle of the primary screen it is more obvious then when the mouse is still near the border. 

It seems that this behaviour is almost every time on the "right PC" (A) and not so often on the "left PC" (B) but it occurs on both.

Sometimes it is not possible to move windows from the primary to the secondary screen. Moving windows from the secondary to the primary screen is always possible. It seems there is a link betweent his two behaviours because "after" the mouse jumping to the border the window movement is never possible but when no jump happened the window movement is possible.

Is there anything I help to debug this and improve the multiple monitor feature?

hrvach, 2024-03-27:
One thing you can try is turning off the mouse acceleration in windows if possible. Also, if you keep moving the pointer past the screen edge on the secondary windows screen, it will accumulate positioning error which then demonstrates as "mouse jumps back to the border" since the box can't know you're pushing the mouse beyond the screen edge in relative mode. What would potentially help is defining screen resolution and capping the movements past that point. One could make Windows use relative mode on all screens but that would make the experience of jumping FROM windows to another computer a bit worse. I can try that too :)

Trekky12, 2024-03-27:
Unfortunately turning mouse acceleration off changes nothing. It would be great if you could try your suggested solution :-)

hrvach, 2024-03-27:
I'll see what can be done! There will probably be no magic fix that makes Windows behave like Linux, it will be a workaround but it can surely be improved at least a bit. Good thing is at least the initial plan somewhat works, so the effort was not pointless :)

spanzetta, 2024-04-08:
Dear @hrvach 
I did update one of my 2 deskhop device to latest 0.51 FW (which if I understood well should add workaround for multiple desktop) but it doesn't work..

Maybe I am doing something wrong.. but on my setup it behaves like previous FW

My setup is: 1 Win10 Laptop (A) with internal + external monitor (which is above the Laptop, but I tried also to reocnifgure it to be on the right side) and 1 Tizen Smart Monitor (B)

On A the mouse remains always on main internal screen..

Should I do something manually to enable the support of the 2nd screen on A?

Thanks in advance

spanzetta, 2024-04-08:
ops.. I missed the instructions at https://github.com/hrvach/deskhop?tab=readme-ov-file#multiple-screens-per-output
Will test it soon.. 
Sorry

hrvach, 2024-04-08:
There is still room for improvement, hopefully I get it to a usable state...

Trekky12, 2024-08-19:
hey @hrvach, 

with the latest release 0.63 when using two monitors on windows when the mouse is moved away from the primary screen everything is working as expected. As soon as the mouse cursor is moved in the direction of the primary screen the mouse jumps to the primary screen.

[image]

With this version it is possible to move windows from the primary to the secondary screen which was not possible previously but the mouse jumping back to the primary screen was less frequently and the jumps were smaller. 

Could you look into this?

hrvach, 2024-08-19:
@Trekky12 I'll take a look, this should be fixed soon with a completely different approach. Would you be up for testing this new solution since I don't have a windows setup?

Trekky12, 2024-08-19:
@hrvach Sure! I am happy to help 😊 
Feel free to give a test version.

The only annoyance of the version 0.62 was moving the window from the primary to the secondary screen. There were occasional jumps near the border but this was no big deal. When the big jumps on the current version could be fixed this would be really great 👍

nguyenhaiac, 2024-08-23:
I'd be happy to test as well. This is the only pain point I have, currently my work around is to use only one screen :D

ak66666, 2024-08-23:
I  can test that too.
In my setup it happens on both sides when the mouse on the secondary screen is moved towards the line between the computers.

hrvach, 2024-08-24:
> hey @hrvach,
>
>
>
> with the latest release 0.63 when using two monitors on windows when the mouse is moved away from the primary screen everything is working as expected. As soon as the mouse cursor [...]

Thank you for a great explanation and graphics. Can you check v0.64 (pre release), is it working for you now? Sorry for breaking it.

ak66666, 2024-08-24:
Yes, with 0.64 that issue is resolved.

hrvach, 2024-08-24:
I was testing the digitizer solution, it works great for moving the cursor, but I can't get mouse wheel to work without additional report, need to play more with it.

nguyenhaiac, 2024-08-25:
initial testing: it's working as expected now. I will try it for some day and report if there is any unexpected behavior.

Trekky12, 2024-08-26:
I can confirm what @ak66666 and @nguyenhaiac already reported: with v0.64 the big mouse jumping issue I described is now fixed 🥳 Thank you very much!

I am excited what improvements are possible with the digitizer solution, thank you for your great work!

jalmeroth, 2024-11-28:
Follow-up from our discussion here https://github.com/hrvach/deskhop/issues/56#issuecomment-2504760157

> So tl;dr, there are 3 cases - Linux just works, macOS has a thing where absolute coords apply to the current screen so a "helper mouse" could be used to nudge it between the two (this thread). Windows is the third case, since KB5003637 where Microsoft broke things and effectively made absolute mode apply to the primary screen output only (if you use a relative mouse to move to screen 2, any input from the absolute one returns you to screen 1). The current workaround is basically switch to using relative mode on screens 2 and beyond in Windows. It's not perfect but better than nothing.
> 
> However, there is a potential workaround with the digitizer pen for moving around, as that specific combo of collection/usagepage/usage will apply to a virtual desktop stretching through all of the screens (just like on Linux). I was able to move the cursor perfectly accurate on multi-screen windows, but couldn't figure out buttons properly (and I don't use Windows either, haha). If you have any ideas, I'm all ears :)

Well, to be honest I would put setups that involve windows in gaming mode by default*, until we find a better solution.

A possible solution may look like this:
- we use a rel-mouse on Windows
- we know the dimensions of the screen(s) from the config
- we run in training mode (no AI please) for a while:
  - we track all moves from our unknown start position
  - the user has to move all the way to all boarders so we can find our actual position through this heat map
- once we are confident enough we enable hop to next pc again
- we continue to track where we are at all times

Probably this will be quite error prone, but maybe it is an idea worth trying.
 

\* As author of DeskhopL I say: Gaming mode should be the default anyway :stuck_out_tongue:

eltariel, 2024-12-05:
Another possible option for windows: present a separate digitizer (absolute-mode pointer) for each deskhop screen. _**Note:**_ this is just an idea, it might not be a solution and I don't currently have a windows machine to test with.

There's apparently a built-in but well-hidden calibration tool (`multidigimon -touch`) that lets you associate each connected digitizer with a display device. We could add a "calibration mode" to work with this:
- (optional) launch the calibration tool (e.g. send win+r, type command, send enter)?
- when prompted to touch a screen, user presses 1, 2, or 3 to select screen position
- send tap event from digitizer for selected position
- repeat until we've run out of screens

Deskhop then tracks screen position the way it already does, but we'd send movement when screen 1 is selected using digitizer 1, screen 2 -> digitizer 2, etc.

More details about monitor->digitizer mapping:
- [Superuser post about multiple touchscreens](https://superuser.com/questions/1158302/windows-10-pc-with-two-touchscreens-touch-only-on-one-display)
- [Article describing the way the tool works](https://www.luisllamas.es/en/configure-multiple-touch-screens-windows/)
- [Stack overflow post about how the mapping is stored in the registry](https://stackoverflow.com/questions/74562479/how-can-two-of-the-same-model-usb-touch-screens-be-differentiated-on-windows-10)

hrvach, 2024-12-05:
I have a *highly experimental* Windows multi-monitor test, for now mouse wheel isn't supported but clicking and dragging should work. Can anyone help and test?

[deskhop-win-experiment.zip](https://github.com/user-attachments/files/18024328/deskhop-win-experiment.zip)

Instructions: Only one side should be upgraded manually (the output where windows has multiple monitors). Should be configured as OS = Windows and 1 screen only and mouse plugged directly into that board. Not ideal, but might be usable?

ak66666, 2024-12-05:
I tried this (as I understood the instructions):
Connected the side where the mouse is to a computer, left other side disconnected. Flashed this firmware to that one side only. Connected both sides to Windows computers: side A with 2 displays, side B with 3. Launched the configuration page, set side B as having 1 display.

It appears that the side to which mouse is connected to is always treated as single-display.  When mouse reaches the end of that screen, the pointer appears at the edge of the next one, but does not move farther.  
When I swap mouse and keyboard - the side with the mouse works with a single monitor, and the side with keyboard - with all of them on that side computer.

Also on side A the cursor goes through the primary screen twice. I.e. I try to move cursor left to right, from side B to A, it switches to the first monitor on side A, when reaches right side of that display it jumps back to the eft side of it, and when still moved to the right, only then it goes to the second monitor's left edge, and could not be moved to the right side any further.

On the multi-monitor side the windows could be dragged between the monitors ok.

Other observations:
- mouse speed has changed, got faster.

hrvach, 2024-12-05:
Some of this behavior is still super-unfinished and needs to be patched around the code, the focus is merely on the multi-monitor and windows behavior, i.e. can you drag windows across desktops and can you use all mouse buttons ok.

This is the digitizer hybrid mode attempt which might even actually work for Windows...

jalmeroth, 2024-12-05:
This is just a minor thing (and I'm currently on a train so I can't verify this), but I noticed one thing:

```cpp
void update_mouse_position(device_t *state, mouse_values_t *values) {
	// snip…

  /* Calculate movement */
  int offset_x = accelerate(values->move_x) * (current->speed_x >> reduce_speed);
  int offset_y = accelerate(values->move_y) * (current->speed_y >> reduce_speed);

  /* Update movement */
  state->pointer_x = move_and_keep_on_screen(state->pointer_x, offset_x);
  state->pointer_y = move_and_keep_on_screen(state->pointer_y, offset_y);

	// snip…
}
```

In `update_mouse_position` we already set the new value for `state->pointer_x`.

```cpp
void check_screen_switch(const mouse_values_t *values, device_t *state) {
  int new_x        = state->pointer_x + values->move_x;
	// snip…
}
```

And then in `check_screen_switch` we add `values->move_x` to `state->pointer_x` again. Isn't `values->move_x` added twice now?

ak66666, 2024-12-05:
Forgot to mention - all mouse keys do work: left/center/right, plus I have two extra to navigate back and forward.

infosecconsultant, 2025-01-24:
I've got 5 plugged into my PC currently (using 4090gpus) and thats running 3x 4k and 2x 2k monitors. With a laptop in the middle or off to the side which I'd like to use with this. But the wide and stacked setup is definitely confusing deskhop.

[image]

hrvach, 2025-01-24:
> I've got 5 plugged into my PC currently (using 4090gpus) and thats running 3x 4k and 2x 2k monitors. With a laptop in the middle or off to the side which I'd like to use with this. But the wide and stacked setup is definitely confusing deskhop.

Vertical switching is something I'll probably start working on soon, if the work doesn't kill me first :) Just wanted to comment I'm impressed with your setup ... it's like you're air traffic control 👍

infosecconsultant, 2025-01-27:
Haha, thanks. Different devices for different bits of work/clients. 😅

kkourin, 2025-03-07:
> Ok, so is it possible to disable the instant switch when mouse movement is detected on the monitor edges and instead rely on only shortcut (some keyboard keypresses) and make the switch, but also be [...]

Anyone find a way to do this? Is there a config to disable the mouse switching but still keep the combination for switching between the two machines?

I guess one way is to use the lock combination after switching. Then every time a switch is needed to unlock, then switch, then lock again. Anyway to do it in one key press? Or to configure the monitor settings so that switching cannot occur?

Scalpos, 2025-03-08:
> > > > is there any software solution (workaround, patch etc.) for windows 10 as it currently disables the second monitor when deskhop is connected
> > > 
> > > 
> > > No software solutions are planned - the point of this is to do it in hardware as a self-contained thing. Windows workaround will basically be: for desktop 2 and beyond switch to relative mouse mode. Switching back from win screen 2 to screen 1 will not be very accurate but that's why it's a workaround. Work is crazy these days, so progress is slow... thank you for your patience.
> > 
> > 
> > Ok, so is it possible to disable the instant switch when mouse movement is detected on the monitor edges and instead rely on only shortcut (some keyboard keypresses) and make the switch, but also be able to use 2 monitors on a windows pc? I do not know whether it is related with edge detection or not, but maybe at least we can use deskhop as a regular USB switch with a shortcut instead of an external hardware switch to trigger switching ?
> 
> Anyone find a way to do this? Is there a config to disable the edge-detection based switching but still keep the combination for switching between the two machines?
> 
> I guess one way is to use the lock combination after switching. Then every time a switch is needed to unlock, then switch, then lock again. Anyway to do it in one key press? Or to configure the monitor settings so that switching cannot occur?

Check [Jan Almeroth's fork **DeskHopL**](https://github.com/jalmeroth/deskhopl). There's no auto-switch between computers, but the mouse can access all computer's screens whatever their positions are. Pressing Caps Lock switch between computers.

[2 more replies]