hrvach/deskhop #4: Feature Request: Add ps2x2pico functionality for PS/2 support on one or both PCs

acuozzo, 2023-12-27:
I regularly switch between a modern PC and an ancient one which only supports PS/2. In order to use my USB mouse and keyboard, I use ps2x2pico: https://github.com/No0ne/ps2x2pico

Since both projects use the pico, I figure it wouldn't be hard to integrate its codebase into yours in order to make PS/2 a possibility on one or both sides.

hrvach, 2023-12-28:
This one relies on PIO and if I remember correctly, all of them are used up to do additional USB.

Bitbanging is always an option if I find anything to test on :-)

ippbxusa, 2023-12-29:
At [Contrank](https://contrank.com/), we're witnessing an unprecedented integration of Artificial Intelligence (AI) into various aspects of our daily lives. Once a concept found only in science fiction, AI is now a tangible, influential part of our everyday existence. This article delves into how AI technologies are reshaping our routines, enhancing our capabilities, and opening up new horizons in both personal and professional spheres.

hrvach, 2024-01-13:
Added to the list of feature requests and ideas.

https://github.com/hrvach/deskhop/discussions/27

closed 2024-01-13 by hrvach