hrvach/deskhop #5: Feature Request: Desk-hop between different HDMI displays

isConic, 2023-12-27:
I love this project. I have two computers, a big desktop computer, and a small work laptop.   

When Im in work mode I would love to have all displays to extend my work laptop, but after work, I would love to switch back to my PC without having to unplug and replug HDMI cables.   Do you foresee any issues with making this work? Maybe for something like this it can't be digitally defined, but would be physical relay switches

jdmg94, 2023-12-28:
I think you might be looking for an [HDMI switch](https://www.amazon.com/UGREEN-Bi-Directional-Splitter-Switcher-Projector/dp/B07JQH1L83/ref=sr_1_6?crid=RH65329S7WGH&keywords=ugreen%2Bhdmi%2Bswitch&qid=1703724828&sprefix=ugreen%2Bhdmi%2Bswitch%2Caps%2C151&sr=8-6&th=1) It's gonna cost you about $12

hrvach, 2023-12-28:
For simply doing the switching, Amazon is full of low-cost switches that work fairly well. 

Overall, doing it yourself is quite tricky - most folks are not happy with 1080p anymore and want 4k (and rightly so), and it's both challenging and expensive to switch that kind of bandwidth without a dedicated ASIC.

Externally controlling an affordable HDMI switch might be a decent compromise :)

closed 2023-12-28 by hrvach