hrvach/deskhop #6: Barrier..

pierrenel, 2023-12-28:
Just curious if you tried https://github.com/debauchee/barrier ?

hrvach, 2023-12-28:
That's a great solution and if your workflow can accommodate for it - by all means!

Sometimes you don't want to / can't install stuff on the computers you want to switch between, have security concerns etc. 

Software solution offers way more options (like clipboard), and it's easier to add more - no wonder the repo is super popular and it seems very well received by the community.

closed 2023-12-28 by hrvach

reorx, 2023-12-31:
I tried using Barrier for a while, but it is not suitable for frequent everyday use. Additionally, I have a KVM which, as the author mentioned, has a lag problem. I am excited about the potential of this project, but I lack the skills and tools to build one myself. Are there any plans to make it available as a Kickstarter project or something similar, so that we can purchase it?

jmaldon1, 2024-01-02:
I tried barrier and it was unusable when you do something like sharing your screen on Zoom and using barrier at the same time. Hardware solutions are way better.