hrvach/deskhop #7: How does your system know it's at the edge of screen 1 and hence, should enter screen 2?

205g0, 2023-12-28:
Doesn't get really clear from the README. Even if the mouse submitted absolute values, e.g.. `x`, it doesn't know when the edge of a screen is reached. So, you'd need always software knowing that the mouse is at the edge. But guess I missed something.

205g0, 2023-12-28:
Ok I think I get it: Your device converts to abs. values. But then, I can't lift my mouse and reposition it without moving the mouse cursor. So, the mouse then act like the pen on a typical Wacom tablet which is also absolute unlike a mouse (or the finger on an iPad), right?

205g0, 2023-12-28:
No, now I get it, the mouse stays relative and your device is kind of a virtual screens with its abs. coordinates catching the crossing.

Whatever, would be great if you could elaborate on this. Just to see if the user faces any compromises compared to a software solution.

205g0, 2023-12-28:
My last comment doesn't make sense, how you should your device know the real screen aspect ratios which it needs to convert to abs. value, otherwise cursor movement is skewed.

hrvach, 2023-12-28:
So, your mouse sends "move up by 10", or "move right by 5" messages. Device will read all of them and adjust internal position locations it keeps (0 - 32767 for X and also 0-32767 for Y) and then report those location to the computer which will, in turn, do the mapping to the actual screen, e.g. [0, 32767] -> [0, 1919] for X axis. For now, the offset is the same for both axes (didn't notice much difference in behavior) but making screen ratios configurable is among the planned updates.

Edge of the screen is always 0 (left) and 32767 (right), regardless of the actual resolution - that's taken care by the computer when mapping absolute coords to actual screen geometry.

closed 2023-12-28 by hrvach

205g0, 2023-12-29:
> For now, the offset is the same for both axes (didn't notice much difference in behavior)

This is interesting and surprising but if it works why not. Is the resolution (32,767) enough to provide a buttersmooth and absolute lag-free experience on high-refresh screens (+144hz) and gaming mice (e.g. 4,000hz polling rate)? Not that everyone would need this, though it's nice, rather to see if the 32,767 is enough for a smooth experience and your chosen hardware is fast-enough for a latency-free operation.

hrvach, 2023-12-29:
> This is interesting and surprising but if it works why not. Is the resolution (32,767) enough to provide a buttersmooth and absolute lag-free experience on high-refresh screens (+144hz) and gaming [...]

I don't really know much about gaming gear, my mouse cost 5$ and  its polling rate is not high. This would have to be tested to make sure it can even poll that fast. For now it's a pretty basic setup (only boot protocol) but I'm working to expand it and properly parse report protocol in order to offer wider compatibility and features. I'm learning as I go along :)