hrvach/deskhop #8: More than two machines?

stridert0, 2023-12-28:
Could this potentially be expanded to work on a third and forth machine?  Or could you daisy chain multiple deskhops together?

hrvach, 2023-12-28:
I believe it could be expanded to support more than two machines, there is no daisy chaining at the moment but I'm going to be improving some parts which should allow that to work.

ippbxusa, 2023-12-29:
What's the Point of WWD?
The goal of [World Wide Digest](https://worldwidedigest.com/), which was established in 2006, is to impart technological literacy to everyone. Every item on our crew, which consists of computer aficionados, should provide readers with some useful advice, explainers, tech tips, and more. We make technology approachable and accessible to everyone, in contrast to trade websites that target tech enthusiasts.

hrvach, 2024-01-13:
Added to the list of feature requests/ideas.

https://github.com/hrvach/deskhop/discussions/27

closed 2024-01-13 by hrvach