hrvach/deskhop #9: Function with one computer connected

KanyonKris, 2023-12-28:
In issue XXXX, YYY asked if the DeskHop would work with only one computer connected (for example, a laptop has been disconnected from the DeskHop while a desktop computer remains connected to the DeskHop). With the current design, the DeskHop will not function with only one computer connected. This is because the DeskHop comprises two Picos, each powered independently by the computer connected to it via USB.

With one computer connected, it might be possible to use that single USB to power both Picos (with diode or-ing), but it would require connecting USB A and USB B grounds together and that may create a ground loop.

Could use an external 5V supply to power both Picos.

Discuss the problem and suggest solutions below.

KanyonKris, 2023-12-28:
This article about desgning a USB hub may have some ideas. The circuit for external 5V power looks useful.
https://circuitdigest.com/electronic-circuits/diy-hub-port-extender

hrvach, 2023-12-28:
I'm trying to get tinyusb host for pio-usb working (so far so good), which would then allow for using a small usb hub on one side to plug both keyboard and mouse, and continue using them even if the other one is disconnected. There would have to be some sort of heartbeat that disables the switching when it is not received more than <TIMEOUT> value, but that should be easy.

This is just the first iteration. Honestly, I'm shocked to learn that people actually want to use this so I'll definitely work on making it better!

KanyonKris, 2023-12-28:
A USB hub connected to Pico A sounds like a good idea.

Let me explain to see if I understand: The DeskHop still uses 2 Picos (Pico A and Pico B). Pico A is powered by the computer it's plugged into USB. Pico A is connected to a USB hub IC (or cheap external USB hub?) that connects to the USB keyboard and USB mouse. Now both keyboard and mouse are connected to Pico A. If a 2nd computer is USB connected to Pico B, it powers up Pico B. When Pico A sees that Pico B is "alive" (via the serial connection between Pico A and Pico B) it knows a 2nd computer is connected so it does mouse screen spanning and keyboard output going to the computer that is "active."

The only caveat is: if only computer B is connected, the DeskHop doesn't work. I think that's a reasonable limitation. I guess a hub could also be added to Pico B so all combinations would work: Computer A only, computer B only, both computer A & B.

hrvach, 2023-12-28:
Yes, you are pretty good at explaining! :) Cheap external USB hub everyone has in their "messy drawer" is what I had in mind if you need this functionality, since it is optional and doesn't increase complexity of the base circuit.  :)

KanyonKris, 2023-12-28:
> Yes, you are pretty good at explaining! :) Cheap external USB hub everyone has in their "messy drawer" is what I had in mind if you need this functionality, since it is optional and doesn't increase complexity of the base circuit. :)

So would you remove the USB-A socket connected to Pico B, leaving just one USB-A socket connected to Pico A (for both keyboard and mouse via a USB hub)? Or would you leave a USB-A socket connected to Pico B and change the firmware so it can handle either situation, hub and no hub?

hrvach, 2023-12-29:
I would leave it in the current setup - does not force you to get a hub but you could if you need to unplug one side regularly.

KanyonKris, 2023-12-29:
Bonus: Support for a USB hub (because Pico A will be able to accept both keyboard and mouse input) also enables wireless USB "unified" dongles that can connect to both a wireless mouse and wireless keyboard, as was asked about in issue #1.

eproxus, 2023-12-29:
Some context with my current use case: I have a desktop and a laptop that share one keyboard and a USB microphone. They’re seldom on at the same time (the laptop is most often asleep, or powered off when the desktop is on). I currently use a [UGREEN USB Switch](https://www.amazon.de/-/en/gp/aw/d/B0C8MSP967) which works fine, but is very slow with switching.

555vedant, 2023-12-30:
hello respected , i would like to work with this project  is there any work  that i can fix?

s51517765, 2024-01-03:
Raspberrypi Pico can be started by supplying power (3.3V) to the Vsys pin.
Additionally, this can coexist with power supply from USB.
So, if we make a circuit from Pico A's 3.3V pin to Pico B's Vsys pin, and from Pico B's 3.3V pin to Pico A's Vsys pin, it would work even when only one of the PCs is powered on.
This should be addressed in the future with PCB modifications, but can be tried with a jumper for testing purposes.
https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf
[image]

hrvach, 2024-01-03:
> hello respected , i would like to work with this project is there any work that i can fix?

Sure, I'll try to compile a list of open issues.

hrvach, 2024-01-03:
> Raspberrypi Pico can be started by supplying power (3.3V) to the Vsys pin. Additionally, this can coexist with power supply from USB. So, if we make a circuit from Pico A's 3.3V pin to Pico B's Vsys [...]

The goal was to have a full galvanic separation between output computers, that's why it wasn't included. One option is to use a usb hub for such a requirement, but another option is a cheap isolated PCB mounted DC-DC converter, like Traco TEA 1-0505, Through Hole 1W 4.5-5.5Vin 5Vout 200mA and costs 1.87$ on Mouser. Pico should not exceed 100mA according to datasheet. Not a bad idea, this would not raise the cost significantly, would still provide proper galvanic insolation and would make it more convenient to use.

tomjohnstone, 2024-07-05:
> The goal was to have a full galvanic separation between output computers, that's why it wasn't included. One option is to use a usb hub for such a requirement, but another option is a cheap isolated [...]

Would you be able to provide schematics for this approach?