hrvach/deskhop #10: Power Toys

abbaselmas, 2023-12-29:
Just leting you know that there exist Power Toys Mouse without Borders, do exactly what you need.

https://learn.microsoft.com/en-us/windows/powertoys/mouse-without-borders

seanwcom, 2023-12-29:
> Just leting you know that there exist Power Toys Mouse without Borders, do exactly what you need.
> 
> https://learn.microsoft.com/en-us/windows/powertoys/mouse-without-borders

Mouse without borders is fantastic, but it only works with Windows machines and (obviously) requires a software install. The project here is for a hardware solution that accomplishes the same without a software install and no OS requirement on each end.

closed 2023-12-29 by abbaselmas

hrvach, 2023-12-29:
> Just leting you know that there exist Power Toys Mouse without Borders, do exactly what you need.
> 
> https://learn.microsoft.com/en-us/windows/powertoys/mouse-without-borders

Btw thanks for the recommendation! I keep getting pleasantly surprised with nice things coming out of Microsoft lately :)