hrvach/deskhop #13: Button to lock yourself to one computer

greengarlic, 2023-12-30:
Hi! Although I have not been able to build this one yet I was thinking how well does it work when you play e.g. games? I always have to disable Barrier when playing or my mouse switches to other computer. A physical button to lock yourself to current computer would do or a key combination on keyboard. Physical button is hardware change but with a button or two there are other possible uses too.

hrvach, 2023-12-30:
Thank you for the suggestion, this will surely be useful and I'll definitely add it in an upcoming version. A keyboard shortcut to lock is probably the easiest!

iooner, 2023-12-31:
An "anti-hop" button or key combination could really come in handy for fullscreen apps.

johschmitz, 2023-12-31:
Could this be implemented as a simple "pass through" which does not introduce any delay whatsoever (in the millisecond range) to use this as a "gaming mode"? 

Also (but tbh less important) two LEDs which show the currently active mode would be useful. I imagine there would be 3 states:

1. Automatic seamless switching (both LEDs on)
2. Pass through A (LED A on)
3. Pass through B (LED B on)

hrvach, 2023-12-31:
That is a good idea, I will try to see how this might be implemented. For now, I'm trying to catch up on basic functionality so more universal mouse support happens, as well as ironing out some bugs.

squarism, 2024-01-02:
This was my first thought when I saw this because this is what happened when using Synergy way back in the day (not the recent buy-out version of Synergy).  I had Scroll Lock mapped as lock mouse.  It was great.  Glad to see this entire idea continue on, nice project.

hrvach, 2024-01-06:
Right CTRL + L in latest firmware should do the trick.

closed 2024-01-06 by hrvach