hrvach/deskhop #14: Question regarding UART

Trekky12, 2024-01-04:
Hey @hrvach,

thank you very much for the effort of creating this piece of hardware. I think you created a solution many of us were looking for 🥳

I have a short question regarding the UART Pins. Could you explain why Pico A is using GPIO 12,13 and Pico B is using GPIO 16,17? Wouldn't it be possible to use the same Pins in both Picos?

Trekky12, 2024-01-04:
Okay good to know 👍 I think this is one of the last barriers to have one unified code for both Picos, am I right?

hrvach, 2024-01-06:
> Okay good to know 👍 I think this is one of the last barriers to have one unified code for both Picos, am I right?

Correct, adding e.g. a simple pullup or pulldown to one board would enable it to figure out if it's A or B on boot and act accordingly, so a single firmware could be used. This will most likely be one of the features in the PCBs next revision.

closed 2024-01-07 by Trekky12