hrvach/deskhop #16: Not working when compiled by source

topaxi, 2024-01-05:
I'm not sure what I am doing wrong, the provided binaries from the beta release work, but I ran into issue #15 and thought I'd increase the mouse acceleration in `user_config.h`.

I built it on my arch linux machine using the following commands:

```sh
PICO_BOARD=pico PICO_SDK_PATH=/usr/share/pico-sdk cmake -S . -B build
cmake --build build
```

I have pico-sdk 1.5.1 installed via aur.

When I flash my pico A, the led keeps blinking in an interval.

Flashing back the provided binaries, functionality returns :sweat_smile:

hrvach, 2024-01-05:
Hey! Thanks for the feedback, can you first please share your mouse brand and model, and raw descriptor (you should be able to find devices in /sys/bus/hid/devices/ and for each one there is a report_descriptor in there you can hexdump -C, for example hexdump -C /sys/bus/hid/devices/XXXX\:XXXX\:XXXX.XXXX/report_descriptor)

It's hard to say what's wrong, it builds without warnings or any information? I'd advise to first git clone --recursive pico-sdk from GitHub and try with that. 

On the mouse being slow, I have likely made some bug that affects certain (more sophisticated?) mice because I have just 2 devices to test on and they are both basic and 8-bit. Please be so kind to share the descriptor and I'll do my best to issue a fix :)

hrvach, 2024-01-05:
@topaxi I believe I might have figured it out - tinyusb used both as device and host seems to have an issue - tuh_task() will get stuck because it checks for tusb_inited() which will be true because of device mode, so it will not return immediately if tuh not inited yet but proceed and enter a while(1) loop, so watchdog will not be reset.

The most interesting question is - what on earth did my compiler do to make the code fast enough to catch up with watchdog expiry timeout :-)

Please try now and let me know. Cheers!

closed 2024-01-10 by hrvach

topaxi, 2024-01-11:
Thank you for the update, I managed to rebuild it by manually checking pico-sdk and tinyusb, which looks better, haven't had the time yet to properly test and run it though.

Here also for completeness, my hexdump of my `Bus 003 Device 008: ID 046d:c069 Logitech, Inc. M-U0007 [Corded Mouse M500]`

```hexdump
00000000  05 01 09 02 a1 01 09 01  a1 00 05 09 19 01 29 08  |..............).|
00000010  15 00 25 01 95 08 75 01  81 02 95 00 81 03 05 01  |..%...u.........|
00000020  09 30 09 31 16 01 f8 26  ff 07 75 0c 95 02 81 06  |.0.1...&..u.....|
00000030  09 38 15 81 25 7f 75 08  95 01 81 06 05 0c 0a 38  |.8..%.u........8|
00000040  02 95 01 81 06 c0 c0                              |.......|
00000047
```