hrvach/deskhop #17: Slow mouse: Logitech G502

unxmaal, 2024-01-06:
On MacOS 14.2.1, Logitech G502. Very slow mouse. 

Used https://github.com/todbot/mac-hid-dump to fetch the descriptors.

```
# note the Logitech reports 2 descriptors

046D C332: Logitech - Gaming Mouse G502
DESCRIPTOR:
  05  01  09  06  a1  01  85  01  05  07  19  e0  29  e7  15  00
  25  01  75  01  95  08  81  02  81  03  95  06  75  08  15  00
  26  a4  00  19  00  2a  a4  00  81  00  c0  05  0c  09  01  a1
  01  85  03  75  10  95  02  15  01  26  8c  02  19  01  2a  8c
  02  81  00  c0  05  01  09  80  a1  01  85  04  75  02  95  01
  15  01  25  03  09  82  09  81  09  83  81  60  75  06  81  03
  c0  06  00  ff  09  01  a1  01  85  10  75  08  95  06  15  00
  26  ff  00  09  01  81  00  09  01  91  00  c0  06  00  ff  09
  02  a1  01  85  11  75  08  95  13  15  00  26  ff  00  09  02
  81  00  09  02  91  00  c0
  (151 bytes)
046D C332: Logitech - Gaming Mouse G502
DESCRIPTOR:
  05  01  09  02  a1  01  09  01  a1  00  05  09  19  01  29  10
  15  00  25  01  95  10  75  01  81  02  05  01  16  01  80  26
  ff  7f  75  10  95  02  09  30  09  31  81  06  15  81  25  7f
  75  08  95  01  09  38  81  06  05  0c  0a  38  02  95  01  81
  06  c0  c0
  (67 bytes)
```

hrvach, 2024-01-06:
Thanks for the report! It suggests this might affect higher-end mice with 16-bit x/y. I'm going to get one of these to test properly. In the meantime, can you check if 
[test_16bit.zip](https://github.com/hrvach/deskhop/files/13851548/test_16bit.zip) makes any difference and how much? Just a warning, this build is *completely* untested since I don't have anything to test on here.

hrvach, 2024-01-06:
Also, please try [test_boot.zip](https://github.com/hrvach/deskhop/files/13851565/test_boot.zip) and I'd appreciate some info if your mouse wheel scroll works. Thanks!

unxmaal, 2024-01-07:
test_16bit didn't work at all.

The Logitech mouse worked with test_boot, but it was a little laggy and slow. The mouse wheel worked fine. However, the keyboard didn't work. 

If you want to push changes to a branch, I can build firmware locally.

hrvach, 2024-01-07:
Thank you for the feedback! Which keyboard model are you using? test_boot behaved the same keyboard-wise even after a hard power cycle?

It's hard to make changes in the blind, I'll try to recreate the issue when I'm back. Just out of curiosity, with the release fw that has the slow mouse symptom, does holding right alt make it even slower or has no effect? Thanks for your patience and understanding, it's not easy to make it work universally :)

chris17453, 2024-01-07:
I've done some debugging with my logitech mouse. 

I could not get the SCALE by size to work at all. for example 8 or 12 or 16.

But this worked for me for my logitech MX. I just removed the switch and set it to a default.

Strangly enough, I can compile and load the project. But the kb does not work on my build. even with 0 changes in the repo.

```c
void extract_values_report_protocol(uint8_t* report,
                                    device_state_t* state,
                                    mouse_values_t* values) {
    /* If Report ID is used, the report is prefixed by the report ID so we have to move by 1 byte */
    if (state->mouse_dev.uses_report_id) {        
        /* Move past the ID to parse the report */
        report++;
    }

    values->move_x = get_report_value(report, &state->mouse_dev.move_x);
    values->move_y = get_report_value(report, &state->mouse_dev.move_y);
    values->wheel = get_report_value(report, &state->mouse_dev.wheel);
    values->buttons = get_report_value(report, &state->mouse_dev.buttons);
    values->move_x <<= 4;
    values->move_y <<= 6;
}
```

hrvach, 2024-01-07:
> Debugged Logitech mouse issues.

Thanks for the help, it's awesome to see someone else poking around this!

> I am able to compile, load project, but keyboard fails work on build, despite no repo changes. Temporary fix: keep Board A with default image, modify Board B accordingly.

This is something totally unexpected I yet have to figure out when I get back from my trip. What versions are you using for gcc, pico-sdk? What is the exact keyboard model? 

>     values->buttons = get_report_value(report, &state->mouse_dev.buttons);
>     values->move_x <<= 4;
>     values->move_y <<= 6;
> }

This is a very useful piece of information, thanks! So, your mouse being slow issue goes away if you shift LEFT by 4? That would mean it was likely flipped to HID report mode OK. When my Logitech mouse arrives I will knew more, but I assumed a 16 bit mouse would basically output a Q8.8 number, and 12-bit mouse Q8.4 ... maybe my assumption was wrong or some other step is missing. 

Please share your mouse model and hid descriptor if possible.

> I'm trying to figure out how to get debugging to work, but one thing I noticed with the scale loop in this function, is it scales to the X size. The Y size might be different, it should have seperate scale switch's If thats the case.

You are right, the most proper thing would be to treat these independently, but I will be really surprised if there exists a mouse that has different resolutions for X and Y.

chris17453, 2024-01-08:
Bus 003 Device 011: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 013: ID 3434:0160 Keychron Keychron Q6

HID DESCRIPTORS for Logitech Unifying Receiver
003:011:002:DESCRIPTOR         1704713201.483199
 06 00 FF 09 01 A1 01 85 10 75 08 95 06 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 75 08 95 13 15 00 26 FF 00 09 02 81
 00 09 02 91 00 C0 06 00 FF 09 04 A1 01 85 20 75
 08 95 0E 15 00 26 FF 00 09 41 81 00 09 41 91 00
 85 21 95 1F 15 00 26 FF 00 09 42 81 00 09 42 91
 00 C0

003:011:001:DESCRIPTOR         1704713201.487198
 05 01 09 02 A1 01 85 02 09 01 A1 00 05 09 19 01
 29 10 15 00 25 01 95 10 75 01 81 02 05 01 16 01
 F8 26 FF 07 75 0C 95 02 09 30 09 31 81 06 15 81
 25 7F 75 08 95 01 09 38 81 06 05 0C 0A 38 02 95
 01 81 06 C0 C0 05 0C 09 01 A1 01 85 03 75 10 95
 02 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
 09 80 A1 01 85 04 75 02 95 01 15 01 25 03 09 82
 09 81 09 83 81 60 75 06 81 03 C0 06 BC FF 09 88
 A1 01 85 08 19 01 29 FF 15 01 26 FF 00 75 08 95
 01 81 00 C0

003:011:000:DESCRIPTOR         1704713201.490202
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 03 95 05 05 08 19 01 29 05
 91 02 95 01 75 03 91 01 95 06 75 08 15 00 26 A4
 00 05 07 19 00 2A A4 00 81 00 C0

HID DESCRIPTORS for KeyChron
003:013:002:DESCRIPTOR         1704713201.492201
 05 01 09 02 A1 01 85 02 09 01 A1 00 05 09 19 01
 29 08 15 00 25 01 95 08 75 01 81 02 05 01 09 30
 09 31 15 81 25 7F 95 02 75 08 81 06 09 38 15 81
 25 7F 95 01 75 08 81 06 05 0C 0A 38 02 15 81 25
 7F 95 01 75 08 81 06 C0 C0 05 01 09 80 A1 01 85
 03 19 01 2A B7 00 15 01 26 B7 00 95 01 75 10 81
 00 C0 05 0C 09 01 A1 01 85 04 19 01 2A A0 02 15
 01 26 A0 02 95 01 75 10 81 00 C0 05 01 09 06 A1
 01 85 06 05 07 19 E0 29 E7 15 00 25 01 95 08 75
 01 81 02 05 07 19 00 29 EF 15 00 25 01 95 F0 75
 01 81 02 05 08 19 01 29 05 95 05 75 01 91 02 95
 01 75 03 91 01 C0

003:013:001:DESCRIPTOR         1704713201.495197
 06 60 FF 09 61 A1 01 09 62 15 00 26 FF 00 95 20
 75 08 81 02 09 63 15 00 26 FF 00 95 20 75 08 91
 02 C0

003:013:000:DESCRIPTOR         1704713201.498204
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 01 05 07 19 00
 29 FF 15 00 26 FF 00 95 06 75 08 81 00 05 08 19
 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01 C0

Software Versions
```
annobin-plugin-gcc-12.28-1
arm-image-installer-3.8-1
arm-none-eabi-binutils-cs-2.41-1
arm-none-eabi-gcc-cs-13.2.0-1
arm-none-eabi-gcc-cs-c++-13.2.0-1
arm-none-eabi-newlib-4.1.0-6
gcc-12.3.1-1
gcc-gdb-plugin-12.3.1-1
gcc-plugin-annobin-12.3.1-1
gcc-c++-12.3.1-1
```

BUILD OUTPUT
```
[root@nd-box-watkinslabs-com deskhop]# make build
PICO_SDK_PATH is /home/nd/repos/github/pico-sdk
PICO platform is rp2040.
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/arm-none-eabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/arm-none-eabi-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Build type is Debug
Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
PICO target board is pico.
Using board configuration from /home/nd/repos/github/pico-sdk/src/boards/include/boards/pico.h
TinyUSB available at /home/nd/repos/github/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
Compiling TinyUSB with CFG_TUSB_DEBUG=1
BTstack available at /home/nd/repos/github/pico-sdk/lib/btstack
cyw43-driver available at /home/nd/repos/github/pico-sdk/lib/cyw43-driver
Pico W Bluetooth build support available.
lwIP available at /home/nd/repos/github/pico-sdk/lib/lwip
mbedtls available at /home/nd/repos/github/pico-sdk/lib/mbedtls
-- Configuring done (0.8s)
-- Generating done (0.0s)
-- Build files have been written to: /home/nd/repos/github/deskhop/build
[9/209] Performing configure step for 'PioasmBuild'
loading initial cache file /home/nd/repos/github/deskhop/build/pico-sdk/src/rp2_common/pico_cyw43_driver/pioasm/tmp/PioasmBuild-cache-Debug.cmake
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The CXX compiler identification is GNU 12.3.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.3s)
-- Generating done (0.0s)
-- Build files have been written to: /home/nd/repos/github/deskhop/build/pioasm
[10/209] Performing build step for 'PioasmBuild'
ninja: no work to do.
[46/209] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 12.3.1
-- The CXX compiler identification is GNU 12.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done (0.5s)
-- Generating done (0.0s)
-- Build files have been written to: /home/nd/repos/github/deskhop/build/elf2uf2
[48/209] Performing build step for 'ELF2UF2Build'
ninja: no work to do.
[209/209] Linking CXX executable board_B.elf
```

hrvach, 2024-01-08:
> The X and Y are shifted by different values, to match the acceleration on my desktop. A 4 shift wasnt enough the Mouse Y, however it was perfect for the X. I must have redeployed a dozen times to figure this out.
> ## Devices
> 
> ```
> Bus 003 Device 011: ID 046d:c52b Logitech, Inc. Unifying Receiver
> Bus 003 Device 013: ID 3434:0160 Keychron Keychron Q6
> ```

I'm a bit confused with the Keychron descriptor, as it also specifies a mouse? So now you'd have two mice connected (not to the same board, but I have no idea what the keyboard is doing as a mouse). Also the descriptor is pretty long and has so many usages there is a chance it would break the current parser. I have to fix that and fortunately I have something to test with now, thanks.

For the Logitech unifying receiver, I have a better idea how it works now and with some refactoring on the way, it should reliably support keyboard and mouse plugged in the same side (and with it, the unifying receiver)

The different X and Y shift is still a mystery to me, I wonder if there is an alternative explanation.

```
        Usage Page (Desktop),           ; Generic desktop controls (01h)
        Logical Minimum (-2047),
        Logical Maximum (2047),
        Report Size (12),
        Report Count (2),
        Usage (X),                      ; X (30h, dynamic value)
        Usage (Y),                      ; Y (31h, dynamic value)
        Input (Variable, Relative),
```

It defines both axis with the same min/max. Of course, that doesn't prove anything and I wonder if it's some Logitech thing... I defined separate speed config in the configuration header file for each axis, but on my mouse anything other than keeping them the same didn't seem natural. 

Might be handy to do some usbhid-dump on the mouse while doing circles (to roughly keep movement the same in both axes)?

chris17453, 2024-01-09:
This morning my Logitech Unifying Receiver is at this address 
- Bus 003 Device 012: ID 046d:c52b Logitech, Inc. Unifying Receiver

Cat'ing the events for endpoint A or B return nothing. However C seems to be the winner
```
cat /sys/kernel/debug/hid/0003\:046D\:C52B.000A/events 

cat /sys/kernel/debug/hid/0003\:046D\:C52B.000B/events 

cat /sys/kernel/debug/hid/0003\:046D\:C52B.000C/events 

report (size 15) (numbered) =  20 02 02 00 00 00 f0 ff 00 00 4f 00 00 00 00
ff00.0041 = 0

report (size 7) (numbered) =  10 02 49 03 00 58 00

report (size 7) (numbered) =  10 02 49 03 00 08 00

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00
ff00.0041 = 1

report (size 7) (numbered) =  10 02 49 03 00 58 00

report (size 7) (numbered) =  10 02 49 03 00 08 00

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 02 00 00 00 00 3c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 02 00 00 00 00 3c 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 02 00 00 00 00 3c 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 00 00 00 00 3c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 00 00 00 00 3c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 00 00 00 00 3c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 10 00 00 00 2c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 02 10 00 00 00 2c 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 01 00 00 00 00 3d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 10 00 00 00 2c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 02 20 00 00 00 1c 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 02 10 00 00 00 2c 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 01 10 00 00 00 2d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 20 00 00 00 1c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 01 20 00 00 00 1d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 20 00 00 00 1c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 01 20 00 00 00 1d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 02 20 00 00 00 1c 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 01 20 00 00 00 1d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 00 10 00 00 00 2e 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 01 20 00 00 00 1d 00 00 00 00
ff00.0041 = 1

report (size 15) (numbered) =  20 02 02 00 00 00 10 00 00 00 2e 00 00 00 00
ff00.0041 = 0

report (size 15) (numbered) =  20 02 02 00 00 00 10 00 00 00 2e 00 00 00 00

report (size 7) (numbered) =  10 02 49 03 00 58 00

report (size 7) (numbered) =  10 02 49 03 00 08 00

report (size 15) (numbered) =  20 02 02 00 00 01 10 00 00 00 2d 00 00 00 00
ff00.0041 = 1

report (size 7) (numbered) =  10 02 49 03 00 58 00

report (size 7) (numbered) =  10 02 49 03 00 08 00

report (size 15) (numbered) =  20 02 02 00 00 ff 1f 00 00 00 20 00 00 00 00
ff00.0041 = 0

report (size 7) (numbered) =  10 02 49 03 00 58 00

report (size 7) (numbered) =  10 02 49 03 00 08 00

report (size 15) (numbered) =  20 02 02 00 00 ff 0f 00 00 00 30 00 00 00 00

report (size 7) (numbered) =  10 02 49 03 00 58 00

report (size 7) (numbered) =  10 02 49 03 00 08 00

report (size 15) (numbered) =  20 02 02 00 00 ff 0f 00 00 00 30 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 ff 0f 00 00 00 30 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 fe 0f 00 00 00 31 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 fe 0f 00 00 00 31 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 fe 0f 00 00 00 31 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 fe 0f 00 00 00 31 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 ff 0f 00 00 00 30 00 00 00 00

report (size 15) (numbered) =  20 02 02 00 00 ff 0f 00 00 00 30 00 00 00 00

report (size 7) (numbered) =  10 02 49 03 00 58 00

report (size 7) (numbered) =  10 02 49 03 00 08 00

report (size 15) (numbered) =  20 02 02 00 00 ff 0f 00 00 00 30 00 00 00 00

report (size 7) (numbered) =  10 02 49 03 00 58 00
```

hrvach, 2024-01-10:
@chris17453 great idea to use the python code to get the mouse position, very cool!

These reports above seem Logitech-specific somehow, it's in vendor-defined page ff00, ones that start with 20 should be report id 32:

Collection (Application),
    Report ID (32),
    Report Size (8),
    Report Count (14),
    Logical Minimum (0),
    Logical Maximum (255),
    Usage (41h),
    Input,
    
... which matches the output - report id (20 hex) and 14 x 1 byte after, but I have no idea what's it supposed to be, need to check some Logitech drivers. 

For mouse, I'd expect report ID 2, then 2 bytes for the buttons, 3 for the coords and 2 for wheel/pan.  
    
Perhaps try usbhid-dump --entity=all --address  your:addr and aim for the one that has report ID 2 in the descriptor?

hrvach, 2024-01-11:
Finally got my hand on a Logitech mouse, can anyone please try [test_highres.zip](https://github.com/hrvach/deskhop/files/13905647/test_highres.zip)? I've also upped the polling rate for these modern mice.

[image]

chris17453, 2024-01-11:
I tried it.

On Computer A: 
- A orange Pi 5 - Armbian
- The x and y are much faster, but still seem a bit slow
- x and y movement are smooth in their operation
- keyboard works

On Computer B:
- Threadrippper - Fedora 37
- X is much faster, y is significantly slower than it needs to be, possibly still @ 1x
- Keyboard works

hrvach, 2024-01-17:
Slow mouse issue should be fixed by now, individual speeds can be tweaked per users preference.

closed 2024-01-17 by hrvach