hrvach/deskhop #19: 3-4 desktop for 1mouse 1 Keyboard ?

fayfay78, 2024-01-08:
Dear Hrvach, first of all i wanna tell you how i am happy when i found your project, it reply to a need that i had since many years.

I have 4 computers and i wanna try to connect all of them with the same mouse and keyboard ( the first computer has 2 screens, the second 1 the third one and the last one screen also).

Do you think that you will be able to extend your project on this need?

I will be very happy to buy all what you ill tell me to buy and test it, 2 3 4 5 computer, i can make all the tests you need and i will be very quick to reply.

Thank you for your reply, hope you will be enthusiast on testing it with me ^^

On the waiting of the pleasure of reading you.

hrvach, 2024-01-08:
I'm really happy if it will make your user experience better and more convenient, that was the whole point!

Multiple computer support seems to be a popular demand, and it's definitely on the roadmap. My idea is to try making it as stable and usable as possible in a 2-machine scenario first, since that is a use-case most people care for, and then use that as a base to expand on and increase the number of devices supported.

My wishful thinking is to not create another device in a different size, but have just one model that's extendable, so you can hook two boxes together with some cable and have 4 devices supported. 

I'll do it as soon as I can, but my daily job gets the priority since that puts bread on the table. :)

fayfay78, 2024-01-10:
Hi dear Hrvach, 

Thank you so much for you fast reply, i will wait so, your idea is a great one, using two boxes and add with a mulitport USb others dongles from other computers. 

Please tell me when you will have the time for checking it, i will be available to make the tests with you whenever you want !

hrvach, 2024-01-13:
Added to the list of feature requests/ideas.

https://github.com/hrvach/deskhop/discussions/27

closed 2024-01-13 by hrvach