hrvach/deskhop #20: suggestion - one of the devices as bluetooth

dkuku, 2024-01-09:
It would be cool to use a bluetooth module for one of the devices. 
Then all the stuff can be inserted into a keyboard and the keyboard cable reused for power. The second device could be connected by bluetooth. Maybe even allow to pair with multiple devices.

hrvach, 2024-01-10:
I've considered that (using a Pico W, no PCB redesign needed and no additional hardware) - it's like 2$ more than a plain pico and you could perhaps use a wireless keyboard AND mouse! The plan is to first get things stable and working for e.g. Logitech mice and popular gear, then add more advanced features (including bluetooth).

dkuku, 2024-01-10:
Cool, then you don't need to separate both of the picos so you can save on that :). Anyway good project, will try to build it.

hrvach, 2024-01-13:
Added to the list of feature requests/ideas.

https://github.com/hrvach/deskhop/discussions/27

closed 2024-01-13 by hrvach