hrvach/deskhop #21: Development/debug tips

mparrett, 2024-01-10:
Hello,

Thank you for sharing your project! I'm building a deskhop this week and I'm curious to see if it solves some of the pain I've been putting up with for years.

I'd love to support and have a handful of Logitech and other devices I can test with. Do you have any suggestions for development setup? Specifically any guidance around debugging would be greatly appreciated! So far I'm able to build and flash the devices but end up with flashing LEDs. I'm guessing this is due to some MacOS idiosyncrasy or the fact that it's all roughly cobbled together on a breadboard until Friday when the PCB arrives, but I am eager to debug. I have a FTDI device, and other pis I could use as serial debuggers... but somewhat uncharted territory at the moment. 

Thanks again!

hrvach, 2024-01-10:
Try checking if you pico-sdk has a recent tinyusb in the lib folder. 

For debugging, SWD works great (you can use another pico as ["picoprobe"](https://github.com/raspberrypi/picoprobe/releases/tag/picoprobe-cmsis-v1.0.3) and that works great!)

USB stdio is tricky when you already have tinyusb doing a device, but you can add a CDC with #define CFG_TUD_CDC 1 in tusb_config.h, you'd need to add a descriptor, something like:
```
    TUD_CDC_DESCRIPTOR(ITF_NUM_CDC,
                       STRID_CDC_INTERFACE,
                       EPNUM_CDC_NOTIF,
                       8,
                       EPNUM_CDC_OUT,
                       EPNUM_CDC_IN,
                       CFG_TUD_CDC_EP_BUFSIZE),
```
... check some of the CDC examples for tinyusb, then you can tud_cdc_n_write() to output strings and use a serial device created to receive them... of course, uart1 is available if that's the preferred route (and it's way easier than CDC).

Do binaries in the "release" section work on your breadboard? Don't worry, I'll try to help and we'll get it working for you somehow :)

mparrett, 2024-01-13:
Thanks for the quick and helpful response. I'll have some time this weekend to hopefully make progress.

> Do binaries in the "release" section work on your breadboard? 

I can load the binaries and device A LED is illuminated. Device B LED is not illuminated.
When I flash with a .uf2 that I built myself, I get a flashing LED.

mparrett, 2024-01-13:
Update: Investigating a possible root cause with `gcc-arm-embedded` vs `gcc-arm-none-eabi` (had issues with the latter on macos initially and followed some instructions to use the -embedded package. But I see now that could be causing issues with my builds.

mparrett, 2024-01-13:
Progress - solid LEDs after flashing with my own build! Uninstalled all *eabi* packages in homebrew, and reinstalled arm toolchain with `brew install --cask gcc-arm-embedded`

hrvach, 2024-01-13:
Yep, for mac os it seems you need to take the gcc-arm-embedded route, also you need to git pull latest tinyusb in pico-sdk/lib/tinyusb). Fingers crossed! :)

mparrett, 2024-01-14:
I appear to have a viable debug setup using a pi zero! For a "hello world" program I'm able to connect to gdb and also read stdio using minicom. Next I'll see if I can do the same with Deskhop!

mparrett, 2024-01-15:
Happy to report I've got a foothold on debugging. Thanks for the help! Relevant snippets below:

Relevant snippets:

**build.sh**
```
cmake -DCMAKE_BUILD_TYPE=Debug
```

**CMakeLists.txt**
```
# debug setup: enable stdio via uart on board A
pico_enable_stdio_uart(board_A 1)
```

**main.h**
```
#define SERIAL_DEBUG_TX_PIN 4 // GPIO 4 -> pin 6
#define SERIAL_DEBUG_RX_PIN 5 // GPIO 5 -> pin 8
```

**setup.c**
```
stdio_uart_init_full(uart1, 115200, SERIAL_DEBUG_TX_PIN, SERIAL_DEBUG_RX_PIN);
printf("hello from pico\n");
```

jalmeroth, 2024-04-06:
Hi there :wave: ,

I tried to follow along your discussion. I see the CDC part has been somewhat implemented and is hidden behind an ifdef. I tried to set `#define DH_DEBUG 1` in `user_config.h`. Unfortunately, I can't build it with the latest version as it looks to me like e.g. this [whole block](https://github.com/hrvach/deskhop/blob/8df007ac135ef2f6b0c47d135d2fb8316a0d1c47/src/tusb_config.h#L90-L104) is ignored.

Errors I get:
```
[ 32%] Building C object CMakeFiles/board_A.dir/src/usb_descriptors.c.obj
In file included from /home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/tusb.h:75,
                 from /home/jan/workspace/deskhop/src/main.h:29,
                 from /home/jan/workspace/deskhop/src/usb_descriptors.c:27:
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:214:67: error: 'CDC_COMM_SUBCLASS_ABSTRACT_CONTROL_MODEL' undeclared here (not in a function)
  214 |   8, TUSB_DESC_INTERFACE_ASSOCIATION, _itfnum, 2, TUSB_CLASS_CDC, CDC_COMM_SUBCLASS_ABSTRACT_CONTROL_MODEL, CDC_COMM_PROTOCOL_NONE, 0,\
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/src/usb_descriptors.c:209:5: note: in expansion of macro 'TUD_CDC_DESCRIPTOR'
  209 |     TUD_CDC_DESCRIPTOR(
      |     ^~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:214:109: error: 'CDC_COMM_PROTOCOL_NONE' undeclared here (not in a function); did you mean 'HID_ITF_PROTOCOL_NONE'?
  214 |   8, TUSB_DESC_INTERFACE_ASSOCIATION, _itfnum, 2, TUSB_CLASS_CDC, CDC_COMM_SUBCLASS_ABSTRACT_CONTROL_MODEL, CDC_COMM_PROTOCOL_NONE, 0,\
      |                                                                                                             ^~~~~~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/src/usb_descriptors.c:209:5: note: in expansion of macro 'TUD_CDC_DESCRIPTOR'
  209 |     TUD_CDC_DESCRIPTOR(
      |     ^~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:218:30: error: 'CDC_FUNC_DESC_HEADER' undeclared here (not in a function)
  218 |   5, TUSB_DESC_CS_INTERFACE, CDC_FUNC_DESC_HEADER, U16_TO_U8S_LE(0x0120),\
      |                              ^~~~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/src/usb_descriptors.c:209:5: note: in expansion of macro 'TUD_CDC_DESCRIPTOR'
  209 |     TUD_CDC_DESCRIPTOR(
      |     ^~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:220:30: error: 'CDC_FUNC_DESC_CALL_MANAGEMENT' undeclared here (not in a function)
  220 |   5, TUSB_DESC_CS_INTERFACE, CDC_FUNC_DESC_CALL_MANAGEMENT, 0, (uint8_t)((_itfnum) + 1),\
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/src/usb_descriptors.c:209:5: note: in expansion of macro 'TUD_CDC_DESCRIPTOR'
  209 |     TUD_CDC_DESCRIPTOR(
      |     ^~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:222:30: error: 'CDC_FUNC_DESC_ABSTRACT_CONTROL_MANAGEMENT' undeclared here (not in a function)
  222 |   4, TUSB_DESC_CS_INTERFACE, CDC_FUNC_DESC_ABSTRACT_CONTROL_MANAGEMENT, 2,\
      |                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/src/usb_descriptors.c:209:5: note: in expansion of macro 'TUD_CDC_DESCRIPTOR'
  209 |     TUD_CDC_DESCRIPTOR(
      |     ^~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:224:30: error: 'CDC_FUNC_DESC_UNION' undeclared here (not in a function)
  224 |   5, TUSB_DESC_CS_INTERFACE, CDC_FUNC_DESC_UNION, _itfnum, (uint8_t)((_itfnum) + 1),\
      |                              ^~~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/src/usb_descriptors.c:209:5: note: in expansion of macro 'TUD_CDC_DESCRIPTOR'
  209 |     TUD_CDC_DESCRIPTOR(
      |     ^~~~~~~~~~~~~~~~~~
In file included from /home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/tusb.h:37:
/home/jan/workspace/deskhop/src/usb_descriptors.c:210:84: error: 'CFG_TUD_CDC_EP_BUFSIZE' undeclared here (not in a function); did you mean 'CFG_TUD_HID_EP_BUFSIZE'?
  210 |         ITF_NUM_CDC, STRID_DEBUG, EPNUM_CDC_NOTIF, 8, EPNUM_CDC_OUT, EPNUM_CDC_IN, CFG_TUD_CDC_EP_BUFSIZE),
      |                                                                                    ^~~~~~~~~~~~~~~~~~~~~~
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/common/tusb_common.h:43:44: note: in definition of macro 'TU_U16_LOW'
   43 | #define TU_U16_LOW(_u16)      ((uint8_t) ((_u16)       & 0x00ff))
      |                                            ^~~~
/home/jan/workspace/deskhop/pico-sdk/lib/tinyusb/src/device/usbd.h:230:50: note: in expansion of macro 'U16_TO_U8S_LE'
  230 |   7, TUSB_DESC_ENDPOINT, _epout, TUSB_XFER_BULK, U16_TO_U8S_LE(_epsize), 0,\
      |                                                  ^~~~~~~~~~~~~
/home/jan/workspace/deskhop/src/usb_descriptors.c:209:5: note: in expansion of macro 'TUD_CDC_DESCRIPTOR'
  209 |     TUD_CDC_DESCRIPTOR(
      |     ^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/board_A.dir/build.make:76: CMakeFiles/board_A.dir/src/usb_descriptors.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1588: CMakeFiles/board_A.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

Am I holding it wrong? :upside_down_face: 

What can I expect to see on the serial device? I would be interested to see the output of PIO-USB like in https://github.com/sekigon-gonnoc/Pico-PIO-USB/tree/main/examples/capture_hid_report Would that be feasible?

jalmeroth, 2024-11-19:
Can this be closed in favour of the [wiki](https://github.com/hrvach/deskhop/wiki/Build---Debug-hints)?

closed 2024-11-19 by hrvach