hrvach/deskhop #23: Different mouse speed on the two PCs

SyedAmerGilani, 2024-01-11:
I just finished building a Deskhop, great Project!

My Problem is that the mouse speed is different on both PCs.
The left PC is a Linux Machine in the speed is ok. The right PC is a Windows Machine and the mouse cursor speed is there very slow.
Curiously changing the cursor speed in Windows changes nothing.
I saw that you can configure the speed factor in user_config.h, could this be made per screen?

hrvach, 2024-01-11:
Thanks for the feedback, it's exciting to see another build! Could you please share what mouse you have? There should be per-screen speed configuration in the upcoming version(s), stay tuned!

SyedAmerGilani, 2024-01-11:
It is a Kingston Orbit Trackball.
I also tried a ProtoArc ME03 Trackball, with this the speed is very very slow in both screens.

Could my Problem be that my linux pc has 3 Screens and because of this a very big horizontal resolution?

hrvach, 2024-01-11:
> It is a Kingston Orbit Trackball. I also tried a ProtoArc ME03 Trackball, with this the speed is very very slow in both screens.
> 
> Could my Problem be that my linux pc has 3 Screens and because of this a very big horizontal resolution?

Yes, under Linux the coordinates 0-32767 are mapped to the entire virtual desktop, so horizontal movements are huge and vertical probably slower. 

Can you perhaps dump the hid descriptor for your device?

bash$ hexdump -C /sys/bus/hid/devices/*/report_descriptor

I have a suspicion it might be a 12-bit device... there is a bug with these. I'm waiting for a new mouse to arrive and I'll try to test/fix when I can re-create (I'm not sure what the issue is yet).

SyedAmerGilani, 2024-01-14:
Here is the descriptor for the mouse only.
[25A7:FA11.txt](https://github.com/hrvach/deskhop/files/13932262/25A7.FA11.txt)

hrvach, 2024-01-14:
Latest firmware should have it better as far as mouse speed goes and I'm working on having this separately configurable per screen and axis which should solve your issue. It's not a super-fast process (requires some testing and validation) ... 

If you want a quick solution, you can add an ugly hack to the end of get_mouse_offset in mouse.c ... something like

```
if(global_state.active_output == ACTIVE_OUTPUT_A && direction == DIRECTION_X)
   offset = offset / 3;
```

... where 3 would be the number of displays you have under computer A.

hrvach, 2024-01-16:
You should be able to reconfigure this in latest firmware in user_config.h ... not perfect but should work for now. Let me know if it's OK now.

closed 2024-01-16 by hrvach