hrvach/deskhop #25: Got some spare PCB's

Gusher123, 2024-01-12:
I ordered some PCB's (and a case) from JLCPCB but don't need 5 Deskhops's. So if you're in The Netherlands and want a PCB, get in touch!

maximumrozema, 2024-01-12:
I'm interested, but I'm in Canada. :-(

markus-maier, 2024-01-13:
hey!
i've ordered some to **berlin/germany** a few days ago. I have 4 left-overs and also able/willing to provide the 3d-printed cases what it costs in material/electricity (together with the pcb should be aroundish 1,5-2€). easiest would be per mail at zahndianajones@gmail.com

dkuku, 2024-01-14:
Could you ELI5 how to order the boards. 
I never did this before and I see there are multiple gerber files and some are duplicated (B/F) which is probably Back/Front ??

markus-maier, 2024-01-14:
> Could you ELI5 how to order the boards. I never did this before and I see there are multiple gerber files and some are duplicated (B/F) which is probably Back/Front ??

I just made a zip from the gerber directory inside /pcb and uploaded it as is to jlcpcb - that's really it.

dkuku, 2024-01-14:
seems easy, I will try it.

maximumrozema, 2024-01-16:
On second thought, after reading the front page, I will wait until after the planned improvements are made.

hrvach, 2024-01-18:
This is a great initiative, pinning this thread to the top.

jamezwhocares, 2024-01-27:
@markus-maier @Gusher123 Do you still have spare PCB? But I need shipping to LV (pref.), but could manage shipping to DE or NL too. Tnx!

infosecconsultant, 2024-02-18:
I've got some spare PCB's as well for anyone in Australia.

s51517765, 2024-02-23:
I can supply some PCBs for Japan.
However, I added a switch for my study.
Of course it works even without a switch.
See https://github.com/s51517765/deskhop/tree/feature for details.
Please bear a small amount of shipping costs.

[image]

chri2, 2024-03-02:
Great project - thanks for publishing it!

I ordered today 5 pcbs and all the other parts from mouser. Paid around 95€. I expect I can share one or two PCBs and the parts needed if I do not break things along the way ;-) .

Fallingonion, 2024-03-08:
I have a couple of spare PCBs and some extra components available - a TI 7221 isolator and two capacitors, I ordered some spares in case I messed up but amazingly didn't :)
Ping me if interested - although if you're not in a rush they cost less than $4 from JLPCB, delivered :)

chri2, 2024-03-08:
Got at least one spare set of parts, maybe two (still need to assemble one deskhop and will wait whether I break something - my first one went smooth and works well, yeahh :) ).

https://flohmarkt.ween.de/~ChriChri/f93118dd-4679-456f-8804-ce5602f5447c

I'm located in the center of Germany. Bought the parts at mouser and the pcb at jlcpcb and paid in total 18.67€ / set.

frostworx, 2024-03-18:
hey @chri2 
I'd be glad if I could buy one of your sets, but unfortunately I can't register on the url you provided _("Sorry, you can not register with this instance at the moment")_
Would be great if you could help.

_(bookmarked the url anyway, looks very interesting)_

chri2, 2024-03-21:
> hey @chri2 I'd be glad if I could buy one of your sets, but unfortunately I can't register on the url you provided _("Sorry, you can not register with this instance at the moment")_ Would be great if you could help.
> 
> _(bookmarked the url anyway, looks very interesting)_

Yes, registration is disabled. You could only reach me there via a fediverse account (like mastodon) owned by you.

If you do not have an account to the fediverse (yet), please write me an email to the address mentioned in the imprint of the side.

frostworx, 2024-03-21:
> Yes, registration is disabled. You could only reach me there via a fediverse account (like mastodon) owned by you.
>
> If you do not have an account to the fediverse (yet), please write me an email to [...]

Hey @chri2 
thanks for the reply :)

I already contacted you via a fediverse account - I hope the message arrived :)
_(I'm just right now about to leave into a long weekend, so can't reply before next monday)_

dfordequan, 2024-04-01:
> I ordered some PCB's (and a case) from JLCPCB but don't need 5 Deskhops's. So if you're in The Netherlands and want a PCB, get in touch!

Hi @Gusher123, I am in the Netherlands and I need a PCB, I would like to buy one if you still have anything left. Thanks!

Gusher123, 2024-04-01:
No problem, still got 4 left. Can mail 1 or 2 and send you a Tikki. How can I get in touch?

dfordequan, 2024-04-02:
> No problem, still got 4 left. Can mail 1 or 2 and send you a Tikkie. How can we get in touch?

@Gusher123 Thank! I would like to have 2. Maybe you could send me an email: dfoedequan@outlook.com

Gusher123, 2024-04-02:
You got mail

Gusher123, 2024-04-02:
Adres niet gevonden
Je bericht is niet bezorgd aan dfoedequan@outlook.com omdat het adres niet kan worden gevonden of geen e-mail kan ontvangen.

dfordequan, 2024-04-03:
> Adres niet gevonden Je bericht is niet bezorgd aan [dfoedequan@outlook.com](mailto:dfoedequan@outlook.com) omdat het adres niet kan worden gevonden of geen e-mail kan ontvangen.

@Gusher123 Sorry my bad. [dfordequan@outlook.com](mailto:dfordequan@outlook.com)

crablab, 2024-04-10:
Hey! 

I have spare PCBs and some assembled with Picos in the UK. Can ship them wherever if someone wants one.

fazzer4x, 2024-04-21:
@chri2 do you still have a set available?
 Or how about you @Gusher123 ? Looking for shipping to Germany

chri2, 2024-04-21:
> @chri2 do you still have a set available?

One is still available: https://flohmarkt.ween.de/~ChriChri/f93118dd-4679-456f-8804-ce5602f5447c

fazzer4x, 2024-04-21:
> > @chri2 do you still have a set available?
> 
> One is still available: https://flohmarkt.ween.de/~ChriChri/f93118dd-4679-456f-8804-ce5602f5447c

Alright. Sent pm via mastodon. Ty

cornfeedhobo, 2024-06-07:
> I ordered today 5 pcbs and all the other parts from mouser.

@chri2 Was this for v1.1? If so, can you share your mouser parts list with exact part numbers?

chri2, 2024-06-08:
> > I ordered today 5 pcbs and all the other parts from mouser.
> 
> @chri2 Was this for v1.1? If so, can you share your mouser parts list with exact part numbers?

No, those were v1. But I'm already late getting v1.1 ordered. Maybe I'll get around to it this weekend.

hrvach, 2024-06-08:
> > > I ordered today 5 pcbs and all the other parts from mouser.
> 
> > 
> 
> > @chri2 Was this for v1.1? If so, can you share your mouser parts list with exact part numbers?
> 
> 
> 
> No, those were v1. But I'm already late getting v1.1 ordered. Maybe I'll get around to it this weekend.
> 
> 
> 
> (promised two deskhops to colleagues :) )

Elecrow put it on their website, you should be able to order an assbled one

maximumrozema, 2024-06-08:
I ordered 5 v1.1 pcbs earlier this week from jlcpcb. I will probably have two I don't need.

alankeny, 2024-06-22:
I ordered two fully assembled Deskhops from Elecrow on June 8th.  I'm in the central US, and I received them on June 20th.  I'm very happy with their work and with this project.  Thanks!

maximumrozema, 2024-09-26:
I'm in Canada. I have two extra sets of parts. That includes pcb, electronic components, picos and 3d printed case.  Let me know if you're interested.

chris-010, 2025-02-02:
Hi, I'm Chris from Austria/Europe and also have some DeskHop's V1.1 here at spare.

- 3pcs fully assembled and tested
- 4pcs just the pcb and with or without components, as you want

If you are interested, just send me a private message.

Here an example how the PCBs looks like:
[image]

chris-010, 2025-11-30:
Too bad that PM is not possible on  Github, therefore leave your mail here in a post, I'll try to answer you asap, then you can edit the post and remove mail adress.

ChuckChance, 2025-11-30:
Thanks, you can reach me at charles_public38@pm.me

TP369, 2026-10-15:
@chris-010  Hi Chris, are any of your fully assembled and tested DeskHops still available? I'm in Germany and would like to buy one. (Ideally with a case. 😄) Lieben Gruß und vielen Dank!

chris-010, 2026-10-18:
Hi! Leave your email address here and I'll get in touch soon. You can edit or remove your email address at any time after our first contact.

lucoceano, 2026-11-09:
@chris-010 Hi, I'm interested in the fully assembled if you still have it available, can you reach me at deskhop.nemeses014@passmail.com