hrvach/deskhop #26: Key repeating until Esc is pressed

Gusher123, 2024-01-13:
I got a specific setup that has an issue. I use an [IBM model M SSK (PS/2)](https://sharktastica.co.uk/wiki?id=modelmssk) through a [cheap active PS/2 to USB converter](https://media.s-bol.com/RWMjAkK4nv0/645x1200.jpg) sold under the Bandbridge brand. I only use the keyboard port on the PS/2 to USB converter.

When I connect the PS/2 converter to the 'Keyboard' port on the Deskhop and my USB mouse to the 'Mouse' port on the Deskhop and A and B to my Laptop and Desktoop, I get a "stuck key" (repeating a previously pressed key until I press Escape) when I switch between laptop and desktop. When I switch again between laptop and desktop after hitting Escape, the repeating continues.

Initially I thought that I could work around this issue by plugging the mouse in the keyboard port and vice-versa, but after a while the same thing happened in that setup.

My guess is that the PS/2 to USB converter is to blame.

hrvach, 2024-01-13:
It might be due to ps2 to usb converter, can you please try the latest release fw that has a bugfix for a similar issue (when output is switched before a key is released, it used to keep outputting that to the previously active screen).

Same behaviour? Does it repeat on the previous screen you switched FROM or the one you switched TO?

Gusher123, 2024-01-13:
I used the recompiled binaries from the 0.3 release from 2 days ago. 

I also tried with one of my [old](https://oldcomputer.info/pc/apmice/ibm11.jpg) [mouses](https://oldcomputer.info/pc/apmice/ibm21.jpg) connected to the PS/2 to USB converter; the mice are not recognized by the Deskhop (the mice are recognised by my Windows machines when the PS/2 to USB is directly plugged into the Desktop or Laptop).

To answer your question; the repeating is occuring on the active screen (where the mouse pointer is) and transfers with the mouse pointer to the other screen.

Gusher123, 2024-01-13:
When the key is repeating and I disconnect the PS/2 converter from the 'keyboard' port on the Dekshub, the repeating stops.

hrvach, 2024-01-13:
> When the key is repeating and I disconnect the PS/2 converter from the 'keyboard' port on the Dekshop, the repeating stops.
> 
> And I discovered that the A and B port get switched over when I switch over the mouse and keyboard. Is that standard behavior?

It's kind of not meant to be swapped (at least until I rewrite some sections to make this a "plug anything anywhere and it does the right thing" kind of deal). 

I've tried my trusty old Dell AT102W through a DeLock PS2-to-USB converter and it works just fine. Had no mouse to try though. 

It might help if you could do raw usb hid report dump to see what the converter is sending. Does the repeating always happen or occasionally? 

Btw congrats on the nice SSK, those are hard to find!

Gusher123, 2024-01-13:
This is the hid report when I connect the PS/2 converter directly to my unix box

`002:002:001:DESCRIPTOR         1705163878.394500
 05 01 09 02 A1 01 85 01 05 09 19 01 29 05 15 00
 25 01 95 05 75 01 81 02 95 01 75 03 81 03 05 01
 09 01 A1 00 09 30 09 31 15 81 25 7F 75 08 95 02
 81 06 C0 09 38 95 01 81 06 09 3C 15 00 25 01 75
 01 95 01 B1 22 95 07 B1 01 C0 05 0C 09 01 A1 01
 85 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10
 81 00 C0 05 01 09 80 A1 01 85 03 19 81 29 83 25
 01 75 01 95 03 81 02 95 05 81 01 C0

002:002:000:DESCRIPTOR         1705163878.396479
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01
 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06
 75 08 15 00 26 E7 00 05 07 19 00 29 E7 81 00 C0

002:003:000:DESCRIPTOR         1705163878.397662
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 95 05 75 01 81 02 95 03 81 01 05 01
 16 01 F8 26 FF 07 09 30 09 31 75 0C 95 02 81 06
 15 81 25 7F 09 38 75 08 95 01 81 06 05 0C 0A 38
 02 95 01 81 06 C0 05 01 0A 68 01 95 01 81 02 C0`

hrvach, 2024-01-13:
This all looks OK to me. Could you also do reports when you press down keys (and release them)? Try holding a key too for a few secs and then release (usbhid-dump --entity=stream), but I don't know if I'll be able to figure out what's wrong from just that - I'd have to know what the box is doing too. Let's try and see.

Gusher123, 2024-01-13:
This is the output of the Deskhop in a stucken key situation. This time the letter "d" stuck. In normal operation, when I press and hold a key, it just repeats 3 times.

[image]

hrvach, 2024-01-13:
Yep, it shows report id 1, no modifiers, and letter D pressed ... the missing piece of data is - what reports did the box read from the ps2-to-hid converter that made it keep sending D. For the keyboard, it just relays whatever it gets unaltered (unless you pressed a hotkey).

During switching, the box restores keyboard LEDs that were active for that output. Maybe the SetReport message confuses the ps2 converter? Probably won't help, but just for the sake of it try to comment out the call to tuh_hid_set_report in led.c and try?

Gusher123, 2024-01-13:
Will first try with another Model M to see if it's de keyboard or the converter.

hrvach, 2024-01-15:
> Will first try with another Model M to see if it's de keyboard or the converter.

Please try [this debug fw](https://github.com/hrvach/deskhop/files/13938718/board_A.zip) on board A, then open the corresponding serial port that gets created and see what is sent when this stuck key scenario happens. This might provide some info to shed light on the issue.

Gusher123, 2024-01-18:
> Please try [this debug fw](https://github.com/hrvach/deskhop/files/13938718/board_A.zip) on board A, then open the corresponding serial port that gets created and see what is sent when this stuck key scenario happens. This might provide some info to shed light on the issue.

I will, but very busy in the comming weeks. Will take some time to report back.

hrvach, 2024-03-23:
No recent updates, closing. Please reopen if still an issue.

closed 2024-03-23 by hrvach