hrvach/deskhop #30: Rotary button on Glorious GMMK Pro keyboard does not work

raldnor, 2024-01-14:
When attaching a Glorious GMMK keyboard to the deskhop the rotary encoder does not work. The  keyboard is configured with Via firmware (https://www.caniusevia.com) which is built upon QMK firmware. 
I have mapped the encoder to KC_VOLD (volume down), KC_VOLU (volume up) and KC_MUTE (mute by pressing the encoder). Could it be that media keys are somehow not getting through?

hrvach, 2024-01-14:
Can you try, just for test, mapping one of the "ordinary" keys to KEY_MEDIA_VOLUMEUP (KC_VOLU) to try if that makes any difference?

Try plugging the keyboard directly to a Linux machine and usbhid-dump --entity=all and then try turning and pressing the knob. 

None of this was tested with any extended buttons (I have a plain ole keyboard) so functionality can be hit and miss. Will try to improve it :)

raldnor, 2024-01-15:
Remapping the key does not make a difference, I mapped mute to F1, and when I attach the keyboard to the deskhop no input is registered at all when pressing the key. Without the deskhop it works.

The output of usbhid-dump return this for the volume down, volume up and mute keys:
001:014:002:STREAM             1705308590.880546
 04 EA 00

001:014:002:STREAM             1705308590.890533
 04 00 00

001:014:002:STREAM             1705308592.524535
 04 E2 00

Full output of the usbhid-dump:
`
001:015:002:DESCRIPTOR         1705309101.461518
 05 01 09 02 A1 01 85 02 09 01 A1 00 05 09 19 01
 29 08 15 00 25 01 95 08 75 01 81 02 05 01 09 30
 09 31 15 81 25 7F 95 02 75 08 81 06 09 38 15 81
 25 7F 95 01 75 08 81 06 05 0C 0A 38 02 15 81 25
 7F 95 01 75 08 81 06 C0 C0 05 01 09 80 A1 01 85
 03 19 01 2A B7 00 15 01 26 B7 00 95 01 75 10 81
 00 C0 05 0C 09 01 A1 01 85 04 19 01 2A A0 02 15
 01 26 A0 02 95 01 75 10 81 00 C0 05 01 09 06 A1
 01 85 06 05 07 19 E0 29 E7 15 00 25 01 95 08 75
 01 81 02 05 07 19 00 29 EF 15 00 25 01 95 F0 75
 01 81 02 05 08 19 01 29 05 95 05 75 01 91 02 95
 01 75 03 91 01 C0

001:015:001:DESCRIPTOR         1705309101.464502
 06 60 FF 09 61 A1 01 09 62 15 00 26 FF 00 95 20
 75 08 81 02 09 63 15 00 26 FF 00 95 20 75 08 91
 02 C0

001:015:000:DESCRIPTOR         1705309101.467502
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 01 05 07 19 00
 29 FF 15 00 26 FF 00 95 06 75 08 81 00 05 08 19
 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01 C0

Starting dumping interrupt transfer stream
with 1 minute timeout.

001:015:002:STREAM             1705309111.318722
 04 EA 00

001:015:002:STREAM             1705309111.328705
 04 00 00

001:015:002:STREAM             1705309114.048696
 04 E9 00

001:015:002:STREAM             1705309114.058680
 04 00 00

001:015:002:STREAM             1705309115.555688
 04 EA 00

001:015:002:STREAM             1705309115.565674
 04 00 00

001:015:002:STREAM             1705309115.685674
 04 EA 00

001:015:002:STREAM             1705309115.695671
 04 00 00

001:015:002:STREAM             1705309116.295674
 04 E9 00

001:015:002:STREAM             1705309116.305676
 04 00 00

001:015:002:STREAM             1705309116.376669
 04 E9 00

001:015:002:STREAM             1705309116.386667
 04 00 00

001:015:002:STREAM             1705309117.520673
 04 E2 00

001:015:002:STREAM             1705309117.799676
 04 00 00

001:015:002:STREAM             1705309119.022670
 04 E2 00

001:015:002:STREAM             1705309119.231669
 04 00 00

`

Please let me know if you need anything else.

hrvach, 2024-01-15:
Try [board_A.zip](https://github.com/hrvach/deskhop/files/13938718/board_A.zip) and then open the new serial port that appears with a serial terminal, press the keys and paste output here.

raldnor, 2024-01-15:
I tried the firmware, but when using the rotary button no output is generated in the serial console (I used 'screen /dev/ttyACM0 115200'). Other keys do give output, e.g. 
```
Kbd: 00 00 04 00 00 00 00 00 

                             Kbd: 00 00 00 00 00 00 00 00 

                                                          Kbd: 20 00 00 00 00 00 00 00 

                                                                                       Kbd: 20 00 0b 00 00 00 00 00
```

raldnor, 2024-01-15:
When I use 'evtest' I can select the following devices related to my keyboard:
```
/dev/input/event14:	Glorious GMMK Pro ANSI
/dev/input/event15:	Glorious GMMK Pro ANSI Mouse
/dev/input/event16:	Glorious GMMK Pro ANSI System Control
/dev/input/event17:	Glorious GMMK Pro ANSI Consumer Control
```
When I select device 14 I can see keypress events, but not from the rotary button. However when I select device 17 ('consumer control') rotary button events are registered:
```
Select the device event number [0-29]: 17
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x320f product 0x5044 version 0x111
Input device name: "Glorious GMMK Pro ANSI Consumer Control"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 28 (KEY_ENTER)
    Event code 74 (KEY_KPMINUS)
    Event code 78 (KEY_KPPLUS)
    Event code 103 (KEY_UP)
    Event code 105 (KEY_LEFT)
    Event code 106 (KEY_RIGHT)
    Event code 108 (KEY_DOWN)
    Event code 110 (KEY_INSERT)
    Event code 111 (KEY_DELETE)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 116 (KEY_POWER)
    Event code 119 (KEY_PAUSE)
    Event code 120 (KEY_SCALE)
    Event code 128 (KEY_STOP)
    Event code 130 (KEY_PROPS)
    Event code 131 (KEY_UNDO)
    Event code 133 (KEY_COPY)
    Event code 134 (KEY_OPEN)
    Event code 135 (KEY_PASTE)
    Event code 136 (KEY_FIND)
    Event code 137 (KEY_CUT)
    Event code 138 (KEY_HELP)
    Event code 139 (KEY_MENU)
    Event code 140 (KEY_CALC)
    Event code 142 (KEY_SLEEP)
    Event code 144 (KEY_FILE)
    Event code 150 (KEY_WWW)
    Event code 152 (KEY_SCREENLOCK)
    Event code 155 (KEY_MAIL)
    Event code 156 (KEY_BOOKMARKS)
    Event code 158 (KEY_BACK)
    Event code 159 (KEY_FORWARD)
    Event code 161 (KEY_EJECTCD)
    Event code 163 (KEY_NEXTSONG)
    Event code 164 (KEY_PLAYPAUSE)
    Event code 165 (KEY_PREVIOUSSONG)
    Event code 166 (KEY_STOPCD)
    Event code 167 (KEY_RECORD)
    Event code 168 (KEY_REWIND)
    Event code 169 (KEY_PHONE)
    Event code 171 (KEY_CONFIG)
    Event code 172 (KEY_HOMEPAGE)
    Event code 173 (KEY_REFRESH)
    Event code 174 (KEY_EXIT)
    Event code 176 (KEY_EDIT)
    Event code 177 (KEY_SCROLLUP)
    Event code 178 (KEY_SCROLLDOWN)
    Event code 181 (KEY_NEW)
    Event code 182 (KEY_REDO)
    Event code 206 (KEY_CLOSE)
    Event code 207 (KEY_PLAY)
[... 118 lines cut ...]
Event: time 1705327201.395704, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1705327201.395704, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1705327201.395704, -------------- SYN_REPORT ------------
Event: time 1705327201.405703, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1705327201.405703, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1705327201.405703, -------------- SYN_REPORT ------------
Event: time 1705327201.473708, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1705327201.473708, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1705327201.473708, -------------- SYN_REPORT ------------
Event: time 1705327201.483704, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1705327201.483704, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1705327201.483704, -------------- SYN_REPORT ------------
Event: time 1705327202.137719, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1705327202.137719, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1705327202.137719, -------------- SYN_REPORT ------------
Event: time 1705327202.147714, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1705327202.147714, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1705327202.147714, -------------- SYN_REPORT ------------
Event: time 1705327202.234710, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1705327202.234710, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1705327202.234710, -------------- SYN_REPORT ------------
Event: time 1705327202.244712, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1705327202.244712, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1705327202.244712, -------------- SYN_REPORT ------------
```

raldnor, 2024-01-15:
With the Keychron it's exactly the same, also QMK firmware, multiple devices:
```
/dev/input/event14:	Keychron Keychron Q1 Pro
/dev/input/event15:	Keychron Keychron Q1 Pro Mouse
/dev/input/event16:	Keychron Keychron Q1 Pro System Control
/dev/input/event17:	Keychron Keychron Q1 Pro Consumer Control
```
where ordinary keypresses are registered via event14 and 'media'-keypresses via event17. I think the Mouse device (event 15 in this case) is created because QMK firmware can emulate a mouse by assigning mouse functions to specific keys (e.g. on another keyboard layer).

This is the output from the Keychron Consumer Control device:
```
Select the device event number [0-29]: 17
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x3434 product 0x610 version 0x111
Input device name: "Keychron Keychron Q1 Pro Consumer Control"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 28 (KEY_ENTER)
    Event code 74 (KEY_KPMINUS)
    Event code 78 (KEY_KPPLUS)
    Event code 103 (KEY_UP)
    Event code 105 (KEY_LEFT)
    Event code 106 (KEY_RIGHT)
    Event code 108 (KEY_DOWN)
    Event code 110 (KEY_INSERT)
    Event code 111 (KEY_DELETE)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 116 (KEY_POWER)
    Event code 119 (KEY_PAUSE)
    Event code 120 (KEY_SCALE)
    Event code 128 (KEY_STOP)
    Event code 130 (KEY_PROPS)
    Event code 131 (KEY_UNDO)
    Event code 133 (KEY_COPY)
    Event code 134 (KEY_OPEN)
    Event code 135 (KEY_PASTE)
    Event code 136 (KEY_FIND)
    Event code 137 (KEY_CUT)
    Event code 138 (KEY_HELP)
    Event code 139 (KEY_MENU)
    Event code 140 (KEY_CALC)
    Event code 142 (KEY_SLEEP)
    Event code 144 (KEY_FILE)
    Event code 150 (KEY_WWW)
    Event code 152 (KEY_SCREENLOCK)
    Event code 155 (KEY_MAIL)
    Event code 156 (KEY_BOOKMARKS)
    Event code 158 (KEY_BACK)
    Event code 159 (KEY_FORWARD)
    Event code 161 (KEY_EJECTCD)
    Event code 163 (KEY_NEXTSONG)
    Event code 164 (KEY_PLAYPAUSE)
    Event code 165 (KEY_PREVIOUSSONG)
    Event code 166 (KEY_STOPCD)
    Event code 167 (KEY_RECORD)
    Event code 168 (KEY_REWIND)
    Event code 169 (KEY_PHONE)
    Event code 171 (KEY_CONFIG)
    Event code 172 (KEY_HOMEPAGE)
    Event code 173 (KEY_REFRESH)
    Event code 174 (KEY_EXIT)
    Event code 176 (KEY_EDIT)
    Event code 177 (KEY_SCROLLUP)
    Event code 178 (KEY_SCROLLDOWN)
    Event code 181 (KEY_NEW)
    Event code 182 (KEY_REDO)
[... 108 lines cut ...]
Event: time 1705327668.825345, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1705327668.825345, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1705327668.825345, -------------- SYN_REPORT ------------
Event: time 1705327668.827341, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1705327668.827341, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1705327668.827341, -------------- SYN_REPORT ------------
Event: time 1705327670.233339, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1705327670.233339, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1705327670.233339, -------------- SYN_REPORT ------------
Event: time 1705327670.235334, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1705327670.235334, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1705327670.235334, -------------- SYN_REPORT ------------
Event: time 1705327671.825339, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1705327671.825339, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1705327671.825339, -------------- SYN_REPORT ------------
Event: time 1705327672.033342, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1705327672.033342, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1705327672.033342, -------------- SYN_REPORT ------------
Event: time 1705327672.536336, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1705327672.536336, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1705327672.536336, -------------- SYN_REPORT ------------
Event: time 1705327672.717337, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1705327672.717337, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1705327672.717337, -------------- SYN_REPORT ------------
```

hrvach, 2024-01-15:
Yep, that's the culprit - no keys registered as keyboard and it only wants to send them as consumer control with another report ID... something I don't know how to universally support yet. You could theoretically bind the rotary wheel as ordinary keypresses in QMK if it could do it, but doing the rotary wheel absolute position is tricky.

You could probably extend the report descriptor for consumer control and pass these events through.

raldnor, 2024-01-15:
So is this case for any  keyboard with multimedia keys? For example in the supported events list there's also an event named 'KEY_BRIGHTNESS_MIN' which I think is related to screen brightness.

hrvach, 2024-01-15:
Not necessarily. If the keyboard sends it as scan codes per USB 1.11 spec, it should work. For example, some of these would be ok:
 
    "0xe8":"KEY_MEDIA_PLAYPAUSE",
    "0xe9":"KEY_MEDIA_STOPCD",
    "0xea":"KEY_MEDIA_PREVIOUSSONG",
    "0xeb":"KEY_MEDIA_NEXTSONG",
    "0xec":"KEY_MEDIA_EJECTCD",
    "0xed":"KEY_MEDIA_VOLUMEUP",
    "0xee":"KEY_MEDIA_VOLUMEDOWN",
    "0xef":"KEY_MEDIA_MUTE",
    "0xf0":"KEY_MEDIA_WWW",
    "0xf1":"KEY_MEDIA_BACK",
    "0xf2":"KEY_MEDIA_FORWARD",
    "0xf3":"KEY_MEDIA_STOP",
    "0xf4":"KEY_MEDIA_FIND",
    "0xf5":"KEY_MEDIA_SCROLLUP",
    "0xf6":"KEY_MEDIA_SCROLLDOWN",
    "0xf7":"KEY_MEDIA_EDIT",
    "0xf8":"KEY_MEDIA_SLEEP",
    "0xf9":"KEY_MEDIA_COFFEE",
    "0xfa":"KEY_MEDIA_REFRESH",
    "0xfb":"KEY_MEDIA_CALC",

And yes, I'm also confused with the COFFEE button and tempted to try it. :)

raldnor, 2024-01-15:
This might be a  very stupid question since I am not very familiar with this topic, but isn't it that report id's point to a single endpoint? So could it be possible to just replicate all the reports?

hrvach, 2024-01-15:
> This might be a very stupid question since I am not very familiar with this topic, but isn't it that report id's point to a single endpoint? So could it be possible to just replicate all the reports?

All questions are good, and I'm by no means an expert around this ... I was just hacking around to fix my own problem :)

Passing reports through should not be too hard, but you'd also need to re-use all (or a subset) of that keyboard's descriptors which can vary greatly, and that's where problems start. I'm first aiming to cover all the basics and then aim for more advanced stuff.

raldnor, 2024-01-15:
I understand. I'm looking into the code and see that in usb.c you filter on HID_ITF_PROTOCOL_KEYBOARD, and when true execute process_keyboard_report(). Now I have been searching if the consumer control device is defined somewhere in the pico-sdk, but that does not seem to be the case. Otherwise it might have worked by adding another switch statement with the report id for consumer control. With a quick search I did found some code for an arduino that seems to use consumer control, but I cannot figure out how to apply it into this code: https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_composite/hid_composite.ino

SyedAmerGilani, 2024-01-17:
In my Keyboardio Model 100(not QMK) i have the problem that the multimedia keys don't work (next, pause, etc).

evtest also reports for this keyboard:
```
/dev/input/event10:	Keyboardio Model 100
/dev/input/event11:	Keyboardio Model 100 Consumer Control
/dev/input/event12:	Keyboardio Model 100 Keyboard
/dev/input/event13:	Keyboardio Model 100 Mouse
/dev/input/event14:	Keyboardio Model 100
```
the multimedia key are on event11

hrvach, 2024-01-18:
See, that's not a keyboard! :) Now seriously, what might be a solution to this is to yank the keyboard descriptor for consumer control, offer that to the PC along with our initial descriptor and pass the reports. Not sure if you'd need to switch to hid report protocol for the keyboard to actually send those or it would not care and just send them.

My poor ole keyboard is just a keyboard, so I don't have a way to test. :-|

OrpheeGT, 2024-01-18:
I'm curious to see how my Corsair Strafe RGB keyboard will be handled once I built the setup. (especially the RGB leds)

Currently I'm running running https://github.com/ckb-next/ckb-next on my Ubuntu, thanks to their work too for it.

hrvach, 2024-01-18:
> I'm curious to see how my Corsair Strafe RGB keyboard will be handled once I built the setup. (especially the RGB leds)
> 
> Currently I'm running running https://github.com/ckb-next/ckb-next on my Ubuntu, thanks to their work too for it.

It will probably not work beyond the "you can type on it" and running default rgb animations locally, since your keyboard is not a Corsair as far as the PC is concerned and the driver won't see it as such. 

I have some ideas how to improve support for fancy keyboards, but the ability to send commands by a computer driver to control something on the keyboard is a potential security risk when bridging two different systems.

OrpheeGT, 2024-01-18:
I understand the concern, and I only installed the "fancy" driver on my personnal computer, I did not install the corsair driver on the corporate Windows computer.

hrvach, 2024-01-18:
Don't get me wrong, I really dig what's possible today with keyboards and open-source! Efforts like the QMK firmware are amazing and simply did not exist before ...  I'd probably get one of those cool keyboards if I didn't love my current one so much :)

SyedAmerGilani, 2024-01-19:
> I have some ideas how to improve support for fancy keyboards, but the ability to send commands by a computer driver to control something on the keyboard is a potential security risk when bridging two different systems.

i would love to support you implementing those ideas. i write C/C++ in my day job so even could help you implement. But i don´t know much about the usb stuff and would not know where to begin.
The security point is valid, maybe that could be something that is default disabled and could be activated by those that accept the risk.

hrvach, 2024-05-24:
Try the latest release? Also, there is a beta release with improved support, I'd also appreciate feedback on that one. 

https://github.com/hrvach/deskhop/releases/tag/v0.60

Grooovec, 2024-05-24:
> Try the latest release? Also, there is a beta release with improved support, I'd also appreciate feedback on that one.
> 
> https://github.com/hrvach/deskhop/releases/tag/v0.60

The multimedia keys on my keyboard that previously did not work, do work with this version!
thank you!

SyedAmerGilani, 2024-05-24:
> Try the latest release? Also, there is a beta release with improved support, I'd also appreciate feedback on that one.
> 
> https://github.com/hrvach/deskhop/releases/tag/v0.60

The multimedia keys on my keyboard that previously did not work, do work with this version!
Thank You!

hrvach, 2024-05-26:
cc: @raldnor  apologies to bug you, I'd appreciate some [help with testing a potential fix if you have a moment?](https://github.com/hrvach/deskhop/discussions/94)

jalmeroth, 2024-11-19:
Can we close this issue?

hrvach, 2024-11-19:
Let's call this one done, please reopen if needed.

closed 2024-11-19 by hrvach