hrvach/deskhop #32: Cant enter keys for which a deskhop modifier is needed.

SyedAmerGilani, 2024-01-15:
To enter @ i have to press right alt + q on my keyboard.
But that does not work with deskhop. I suppose because pressing right alt goes in the mouse_zoom_hotkey_handler and further keys are ignored.

hrvach, 2024-01-15:
Good point. You can recompile without the altgr shortcut, and also please suggest what would be a better choice here - I'm open to ideas (this was a button I never use).

SyedAmerGilani, 2024-01-15:
I have disabled the key as you suggested, that works for me at the moment.
It is hard to find a single non modifier key that is not used generally. maybe scroll lock?

hrvach, 2024-01-16:
Many modern keyboards don't have that anymore. I'll try changing it to something else in the future release, or let the altgr through to the OS (also not a bad option).

closed 2024-01-16 by hrvach