hrvach/deskhop #33: Error when building source

cskwrd, 2024-01-15:
I get the following error when building from source:
```
In file included from /home/dev/src/usb_descriptors.c:26:
/home/dev/src/main.h:88:20: error: expected identifier or '(' before ':' token
   88 | enum packet_type_e : uint8_t {
      |                    ^
make[2]: *** [CMakeFiles/board_A.dir/build.make:76: CMakeFiles/board_A.dir/src/usb_descriptors.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1596: CMakeFiles/board_A.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

I have not made any changes to the source. My guess is a dependency isn't configured correctly.
For what its worth I'm working inside this docker container: https://github.com/lukstep/raspberry-pi-pico-docker-sdk

hrvach, 2024-01-15:
Yep, I think enum typing is not a regular C thing, maybe an older GCC complains about it. Try removing the semicolon and type.

cskwrd, 2024-01-15:
That certainly allowed me to work past the issue, thank you! Are the build dependencies listed anywhere?

hrvach, 2024-01-16:
The guideline is "get the latest and greatest" until I test with specific versions and compile a table. :)

vaidhy, 2024-01-16:
I think this is a gcc 12 vs gcc 13 feature. GCC 12 does not support types for enum.

hrvach, 2024-01-16:
Will be fixed in the next release.

cskwrd, 2024-01-16:
Noted! Thanks for the help.

closed 2024-01-16 by cskwrd