hrvach/deskhop #36: USB HUB: Only one device is recognized

maxim-kukushkin, 2024-01-17:
First of all, thank you very much for this project! The idea, documentation and the quality of models/schemas are amazing! It was a pleasure to build this device, everything went smooth

The update from yesterday indeed enabled partial support of USB HUBs. However, there's an issue I'm facing with mine:

I use Ugreen USB switcher (https://www.amazon.co.uk/dp/B01N6GD9JO?ref_=cm_sw_r_cp_ud_dp_8NEX4KNSA6PH3T8T8QWG), I've connected my keyboard and mouse into 2 USB ports (2 other ports are empty). However, only one device is recognized by Deskhop.

When I switch outputs on Ugreen, or if I simply unplug and plug it in - only keyboard is recognized and works. The mouse is not detected.

In case I try to unplug the devices from the HUB and plug one by one - only the first device is recognized. For example, if I plug the mouse first and then the keyboard, only the mouse will work. It doesn't depend if the USB HUB is connected to A or B.

Is there any other debug info I can collect (I use Linux on both computers, in theory a Windows one is available)

hrvach, 2024-01-17:
Thanks for the report! Good catch, I *might* know what the issue is, could you please test this firmware for me and report back?

[beta_firmware.zip](https://github.com/hrvach/deskhop/files/13968607/beta_firmware.zip)

maxim-kukushkin, 2024-01-17:
It's almost working :)

It correctly identified both keyboard and mouse connected through the HUB and worked perfectly. However, once I disconnected the HUB's cable and connected again, nothing was detected until Raspberry's restart

hrvach, 2024-01-17:
This seems like a TinyUSB issue, deadlocking when two devices try to enumerate in a very short timespan (or something similar). I've tried to dogscience it, but I don't know what I'm doing so it might blow up. 

Any improvement with this? [beta_firmware2.zip](https://github.com/hrvach/deskhop/files/13969966/beta_firmware2.zip)

maxim-kukushkin, 2024-01-18:
hmmm... After changing to the beta2 firmware it completely stopped recognizing mouse via the HUB.
But after rolling back to the beta1 firmware I wasn't able to reproduce the behavior even from my previous comment... I'll try to do more experiments tomorrow

hrvach, 2024-01-18:
I have managed to recreate it, seems like it fails with:

[1:] [1:] USBH Defer Attach until current enumeration complete

and ends up in a deadlock. First 1 or 2 re-plugs work for me, then it hangs. With just one device plugged in the hub it works as well... This looks internal to TinyUSB, more debugging needed.