hrvach/deskhop #39: Suggestion: Add holes to support clone Raspberry Pi Pico pinouts

Ardakilic, 2024-01-19:
Hello,

There are some Raspberry Pi Pico clones, which are cheaper and have usb-c connectors, such as these:

[image]

Pinouts match 1:1 on these clone pico boards. 

However, since the current PCB design is for smd soldering only, it's too hard to solder these pico to the board.

If the soldering pads chould support both smd and through hole soldering, something like the center part of this PCB, it'd be way easier so solder these.

[image]

I'd appreciate if this is considered. Sorry I'm not a board designer, otherwise this would have been a pull request.

Thanks in advance!

hrvach, 2024-01-20:
Can do, that's for clearly documenting it like this. The rev1 board is already in the works, I'll try to squeeze in this addition.

OrpheeGT, 2024-01-31:
> Can do, that's for clearly documenting it like this. The rev1 board is already in the works, I'll try to squeeze in this addition.

Do you suggest to wait before ordering a new batch if some friends request for this great toy ?

Ardakilic, 2024-01-31:
@OrpheeGT if you're looking for USB-C like I do in the end, I got similar but different clones for the time being. Found them on a Aliexpress seller. They worked nicely:

[image]

[image]

Tested yesterday with Win-MAC and MAC-MAC. Worked without issues.

However, I'd prefer these black clones. They have boot select built in and overall fells a better quality product.

hrvach, 2024-01-31:
The planned changes have already been done, but I'm waiting for the boards/parts to confirm everything is OK and functional as expected. Maybe I'm a bit overly cautious, but that goes with the job :)

I'm adding the holes as an option so it's up to the user to decide - original Pico does lack some features like a reset button, but honestly I'm still sticking with the original because it's pretty cheap, stocked everywhere and also giving some symbolic support for the Raspberry Pi Foundation :)

Micro USB is OK, but I'm keeping it plugged in all the time so I don't know how it behaves after 5000 cycles :)

hrvach, 2024-03-23:
Holes added in PCB v1.1.

closed 2024-03-23 by hrvach

Ardakilic, 2024-03-23:
Awesome news, thank you! 🚀