hrvach/deskhop #40: Custom image issues

vaidhy, 2024-01-22:
I am able to cross-compile and build on my debian 12 (gcc 12.2) after installing the following:
sudo apt install cmake gcc-arm-none-eabi libnewlib-arm-none-eabi libstdc++-arm-none-eabi-newlib 

However, the keyboard on my custom images do not work. They work with the pre-built binaries. I changed the CAPS_LOCK to F12 in user_config.h.

Am I compiling with the right cross compiler? If there is are more instructions on compiling the code with either a linux or mac machine, it would help a lot.

hrvach, 2024-01-22:
Try doing a git pull in your pico-sdk and pico-sdk/lib/tinyusb folders. Does that change anything?

arm-none-eabi-gcc should be OK.

vaidhy, 2024-01-22:
Never mind. I found the issue. I was using pico_w instead of pico :(

closed 2024-01-22 by vaidhy