hrvach/deskhop #42: deskhop - active usb hub not working

GiT-BeDa, 2024-01-24:
First off all. Thanks for your great work! I followed your project for the last month. Because of your great documentary i got my deskhop board assembled and it is a pleasure to watch the software evolve.

My setup is a cherry G80-3000N RGB keyboard (0.79W) and a Corsair M65 RGB Ultra Gaming mouse (~0.54W). If i connect these periperals to my deskhop board it seems not one of these work out of the box. So i got the idea in my mind that the problem is the power consumption of these devices, i ordered a active powered USB-Hub (CSL - USB 3.0 Hub Aktiv - 10 GBit/s from Primewire) and connect my mouse and keyboard via these powered HUB but none of my periperals even light up nor do anything. 

If i try it with a standard keyboard and mouse direct connected to Board A and B it works like charme. If i try to connect these working keyboard and mouse via the HUB it stops working. If i connect the HUB directly to the PC everything works. Is there a way to debug or solve these behavior for me? Do you have any suggestions, what i can do to make it work?

I have seen that it is possible to switch beetween desktops using a configurable key, are there any plans to make it possible to switch via lets say middle mouse double click? I would love to only switch beetween outputs on keypress or double middle mouse click, because of fullscreen applications and games.

hrvach, 2024-01-24:
Perhaps the easy way to test is to try connecting board A to a PC and wait for the LED to come on, then plug in the keyboard and check if the LED goes down (along with the port) and if your USB device disappears from your PC.

Hubs are tricky to support and I have just one that works; I have to get 2-3 more that have issues and try to debug.

Try checking what kind of power requirements your peripherals seem to report (Linux, Google around for other OS-es):

bash# lsusb -v | grep -E "^Bus|[0-9]+mA"

GiT-BeDa, 2024-01-24:
It seems all devices reporting 500mA:
> DeskHop direct connected: Bus 002 Device 004: ID 2e8a:107c Hrvoje Cavrak DeskHop Switch MaxPower 500mA
> Keyboard  direct connected: Bus 002 Device 003: ID 046a:00de Cherry GmbH G80-3000N FL RGB Keyboard MaxPower 500mA
> Mouse  direct connected: Bus 001 Device 002: ID 1b1c:1b9e Corsair CORSAIR M65 RGB ULTRA Gaming Mouse MaxPower 500mA

Ok, i tested a little bit around and removed the HUB from the setup:

I connected only board A to my PC. If i connect the keyboard it lights up and it works. I can turn off the LED via Caps, then i remove the keyboard and connect the mouse to board A. It also works and the LED on the pico turns off if i reach the left side of my screen.

Then i connect my second PC with board B, the keyboard on A works as it should. I can switch beetween the computers via Caps and use it as soon i connect my mouse with board B, the mouse doesn't light up and at the moment i try to swap to board B with Caps it seems like the pico B reboots (the LED on the pico B turns on at Caps an switches back to A).

Then i swaped mouse and keyboard (2 PC connected, mouse at A, keyboard at B) now the mouse works and i can switch with the mouse. The keyboard does not light up, as soon i switch to the other pc the pico B seems to reboot.

Is it possible that board B has a problem to provide enough power? As i said with a standard keyboard and mouse it works flawlessly.

It would a pleasure for me to spend a problematic HUB to you, to give a little bit support back to your project. If you would provide a address to me.

GiT-BeDa, 2024-11-14:
Again, thank you for your great work! I have successfully built a v2 board (with the capacitor in place), and it works like a charm!

closed 2024-11-14 by GiT-BeDa