hrvach/deskhop #44: How to utilize keyboard and mouse of Laptop as input instead of that from Usb Port A and B?

lemonriver, 2024-01-29:
I have a laptop and a desktop device.  And do not want to buy any extra keyboard and mouse. :(

hrvach, 2024-01-29:
Unfortunately there is no way to physically do this, since your laptop keyboard and touchpad are directly wired to your laptop. For this case, perhaps a better solution might be some of the software solutions suggested in the README?

lemonriver, 2024-01-29:
Thanks hrvach for your suggestion.  But maybe it's super cool if only use the deskhop as device connector, and input from device A or device B or deskhop could be selected as the real input source.  Of course, we need to develop some software to hijack input from device A or B, just similar to that input from deskhop.