hrvach/deskhop #45: Corsair RGB Strafe Keyboard and Corsair Scimite Elite mouse not working

OrpheeGT, 2024-01-29:
Hello,

I successfuly built the board today

I plugged my keyboard and mouse it did not work.

Keyboard LED was blinking (no computer detected)
Mouse was only moving to the right of the screen only when I was clicking on left or right button

I tried the test_fw mentionned in wiki but mouse did not work at all with this one, no move at all.

I plugged another mouse, Logitech G500s and a spare Mac keyboard, and they both instantly worked fine.

So I guess FW if well flashed and solders are OK

Please find my lsusb result from my corsair keyboard/mouse : 

```
Bus 009 Device 036: ID 1b1c:1b8b Corsair CORSAIR SCIMITAR RGB ELITE Gaming Mouse
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b8b 
  bcdDevice            3.21
  iManufacturer           1 Corsair
  iProduct                2 CORSAIR SCIMITAR RGB ELITE Gaming Mouse
  iSerial                 3 1203701FAF5D1C045DD7433BF5001C05
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     172
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

Bus 009 Device 037: ID 1b1c:1b20 Corsair STRAFE RGB Gaming Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b20 STRAFE RGB Gaming Keyboard
  bcdDevice            3.08
  iManufacturer           1 Corsair
  iProduct                2 Corsair STRAFE RGB Gaming Keyboard
  iSerial                 3 13008030AEAA1004550316D6F5001940
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     132
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

hrvach, 2024-01-29:
Please do:

usbhid-dump --entity=descriptor 

to dump the HID descriptors for these (when plugged directly into the computer) and usbhid-dump --entity=stream and then try to type and move the mouse and paste the output somewhere please.

The mouse problem sounds like the descriptor might have not been parsed correctly. The keyboard device specifies 500 mA of current, not sure how much it uses but perhaps try to disable the RGB lights temporarily and see if that changes anything?

As far as screen reversal goes, I just swap the cables :D have to make that a software-configurable option...

OrpheeGT, 2024-01-29:
usbhid-dump : 

# usbhid-dump --model=1b1c:1b20
009:037:001:DESCRIPTOR         1706533974.582343
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 02 81 02 09 04 91 02 C0

009:037:000:DESCRIPTOR         1706533974.587347
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 97 15 00 25 01
 75 01 95 98 81 02 05 08 19 01 29 03 25 01 75 01
 95 03 91 02 75 05 95 01 91 01 C0 05 0C 09 01 A1
 01 85 02 05 0C 19 00 2A FF 0F 15 00 26 FF 0F 75
 10 95 02 81 00 C0 06 C0 FF 09 02 A1 01 85 03 15
 00 26 FF 00 95 3F 75 08 09 02 81 02 C0 06 C2 FF
 09 03 A1 01 85 0E 15 00 26 FF 00 95 3F 75 08 09
 03 81 02 C0

# usbhid-dump --model=1b1c:1b8b
009:036:001:DESCRIPTOR         1706534023.516541
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 02 81 02 09 04 91 02 C0

009:036:000:DESCRIPTOR         1706534023.518819
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 80 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0 05 0C 09 01
 A1 01 85 02 05 0C 19 00 2A 3C 02 15 00 26 FF 0F
 75 10 95 02 81 00 C0 06 C1 FF 09 02 A1 01 85 03
 15 00 26 FF 00 09 02 95 3F 75 08 81 02 C0 06 C2
 FF 09 03 A1 01 85 0E 15 00 26 FF 00 95 3F 75 08
 09 03 81 02 C0 05 01 09 06 A1 01 85 10 05 07 19
 E0 29 E7 15 00 25 01 75 01 95 08 81 02 19 00 29
 97 15 00 25 01 75 01 95 98 81 02 C0

OrpheeGT, 2024-01-29:
```
# usbhid-dump --entity=descriptor
011:002:000:DESCRIPTOR         1706534125.988645
 06 00 FF 09 01 A1 01 19 01 29 40 15 01 25 40 75
 08 95 40 81 00 19 01 29 40 91 00 C0

002:002:000:DESCRIPTOR         1706534125.988832
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 03
 15 00 25 01 95 03 75 01 81 02 95 01 75 05 81 01
 05 01 09 30 09 31 15 00 26 FF 7F 35 00 46 FF 7F
 75 10 95 02 81 02 05 01 09 38 15 81 25 7F 35 00
 45 00 75 08 95 01 81 06 C0 C0

009:008:003:DESCRIPTOR         1706534125.995497
 05 0C 09 01 A1 01 85 01 09 E9 09 EA 75 01 95 02
 15 00 25 01 81 02 75 06 95 01 81 03 75 18 81 03
 C0 06 43 FF 0A 02 02 A1 01 85 11 95 13 75 08 26
 FF 00 09 02 81 00 09 02 91 00 C0 06 00 FF 09 01
 A1 01 09 0A A1 02 85 FF 95 01 75 20 09 0B 17 01
 00 00 80 27 FF FF FF 7F 81 02 C0 09 22 A1 02 85
 90 09 33 75 08 15 00 26 FF 00 81 02 09 34 81 02
 05 01 09 3B 75 10 27 FF FF 00 00 81 02 06 00 FF
 09 0A A1 02 05 0A 19 01 29 0A 95 0A 75 08 26 FF
 00 81 02 C0 C0 06 00 FF 09 22 A1 02 85 90 09 33
 95 01 91 02 09 34 91 02 05 01 09 3B 75 10 27 FF
 FF 00 00 91 02 06 00 FF 09 0A A1 02 05 0A 19 01
 29 0A 95 0A 75 08 26 FF 00 91 02 C0 C0 06 00 FF
 09 22 A1 02 85 C4 09 33 95 01 81 02 09 34 81 02
 05 01 09 3B 75 10 27 FF FF 00 00 81 02 06 00 FF
 09 0A A1 02 05 0A 19 01 29 3B 95 3B 75 08 26 FF
 00 81 02 C0 C0 06 00 FF 09 22 A1 02 85 C4 09 33
 95 01 91 02 09 34 91 02 05 01 09 3B 75 10 27 FF
 FF 00 00 91 02 06 00 FF 09 0A A1 02 05 0A 19 01
 29 3B 95 3B 75 08 26 FF 00 91 02 C0 C0 06 00 FF
 09 0C A1 02 85 FF 05 01 09 3B 95 01 B1 02 09 3A
 A1 02 05 0A 19 01 29 3E 95 3E B1 02 C0 C0 06 00
 FF 09 0C A1 02 85 E2 05 01 09 3B 95 01 81 02 09
 3A A1 02 05 0A 19 01 29 3E 95 3E 81 02 C0 C0 C0

009:036:001:DESCRIPTOR         1706534125.996897
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 02 81 02 09 04 91 02 C0

009:036:000:DESCRIPTOR         1706534125.999171
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 20 15 00 25 01 95 20 75 01 81 02 05 01 09 30
 09 31 16 01 80 26 FF 7F 75 10 95 02 81 06 09 38
 15 81 25 7F 75 08 95 01 81 06 C0 C0 05 0C 09 01
 A1 01 85 02 05 0C 19 00 2A 3C 02 15 00 26 FF 0F
 75 10 95 02 81 00 C0 06 C1 FF 09 02 A1 01 85 03
 15 00 26 FF 00 09 02 95 3F 75 08 81 02 C0 06 C2
 FF 09 03 A1 01 85 0E 15 00 26 FF 00 95 3F 75 08
 09 03 81 02 C0 05 01 09 06 A1 01 85 10 05 07 19
 E0 29 E7 15 00 25 01 75 01 95 08 81 02 19 00 29
 97 15 00 25 01 75 01 95 98 81 02 C0

009:037:001:DESCRIPTOR         1706534126.002389
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 02 81 02 09 04 91 02 C0

009:037:000:DESCRIPTOR         1706534126.007463
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 97 15 00 25 01
 75 01 95 98 81 02 05 08 19 01 29 03 25 01 75 01
 95 03 91 02 75 05 95 01 91 01 C0 05 0C 09 01 A1
 01 85 02 05 0C 19 00 2A FF 0F 15 00 26 FF 0F 75
 10 95 02 81 00 C0 06 C0 FF 09 02 A1 01 85 03 15
 00 26 FF 00 95 3F 75 08 09 02 81 02 C0 06 C2 FF
 09 03 A1 01 85 0E 15 00 26 FF 00 95 3F 75 08 09
 03 81 02 C0
[... 732 lines cut ...]

009:036:000:STREAM             1706534266.649307
 01 02 00 00 00 00 00 FF FF 00

009:036:000:STREAM             1706534266.650267
 01 02 00 00 00 FF FF 00 00 00

009:036:000:STREAM             1706534266.652300
 01 02 00 00 00 FF FF 00 00 00

009:036:000:STREAM             1706534266.655270
 01 02 00 00 00 00 00 FF FF 00

009:036:000:STREAM             1706534266.656254
 01 02 00 00 00 FF FF 00 00 00

009:036:000:STREAM             1706534266.698283
 01 00 00 00 00 00 00 00 00 00

009:036:000:STREAM             1706534266.702343
 01 00 00 00 00 01 00 00 00 00

009:036:000:STREAM             1706534266.708275
 01 00 00 00 00 01 00 00 00 00
```

OrpheeGT, 2024-01-29:
The keyboard has a 2nd USB connector for power, this USB is plugged on a powered USB hub, so It should not be a problem ?

switching off lights does not help.

hrvach, 2024-01-29:
How are you plugging this? one in port A, one in port B on the box or both through a hub?

OrpheeGT, 2024-01-29:
The USB hub is only used to power the 2nd USB (power usb) of the keyboard.

The main USB keyboard is directly plugged on USB A

Mouse directly plugged to USB B

mini usb are respectively plugged direcly on computers.

hrvach, 2024-01-29:
Thanks, first thing I notice mouse is reporting 32 buttons and 16 was set as hard limit (because I thought nothing would ever top that). Will try taking a look after work...

OrpheeGT, 2024-01-29:
Yep, there a some buttons on it :D

[image]

OrpheeGT, 2024-01-29:
Ok, I discovered something.

The Keyboard as a "BIOS" mode ...

When you long press F1 + Windows lock button (5 seconds) it switch to "BIOS" mode as called on the web.

Then it is correcty working, but the "insert" led blinks constently (a bit annoying).

In this mode, the leds are stuck to hardware default mode, and I can't even disable/switch off leds at all.

I neved needed this "BIOS" mode until now, with or without Corsair drivers (they are not installed on corporate laptop).

Here is the lsusb output in this BIOS mode :

```
Bus 009 Device 037: ID 1b1c:1b20 Corsair STRAFE RGB Gaming Keyboard
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b20 STRAFE RGB Gaming Keyboard
  bcdDevice            3.08
  iManufacturer           1 Corsair
  iProduct                2 Corsair STRAFE RGB Gaming Keyboard
  iSerial                 3 13008030AEAA1004550316D6F5001940
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0042
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     132
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      29
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```
# usbhid-dump --model=1b1c:1b20
009:009:000:DESCRIPTOR         1706543447.883544
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 75 08 95 01 81 01 05 07 19 00
 2A FF 00 15 00 26 FF 00 75 08 95 06 81 00 05 08
 19 01 29 03 25 01 75 01 95 03 91 02 75 05 95 01
 91 01 C0
```

I don't know why I'm not able to run descriptor command anymore (even with a reboot) : 

# usbhid-dump --entity=descriptor
009:006:001:Failed to claim: Resource busy
009:006:001:Failed to release: Entity not found
009:006:000:Failed to release: Entity not found
009:009:000:Failed to release: Entity not found
009:005:001:Failed to release: Entity not found
009:005:000:Failed to release: Entity not found

hrvach, 2024-01-29:
Just a wild guess, does [this](https://github.com/hrvach/deskhop/files/14087007/test_fw_descriptor.zip) do anything for the mouse?
 
As for the keyboard, it seems to refuse the BOOT mode unless manually switched with the keyboard shortcut. Full hid report protocol for the keyboard is on the to-do list, might get done alongside descriptor proxying at some point...

OrpheeGT, 2024-01-29:
Same as latest official FW.

when I move the mouse, nothing appens unless I also click on a (right or left) button at the same time.

And the only thing it does is it moves from left to right no matter the way I move it.

hrvach, 2024-01-29:
Hmmm, it's like the mouse refuses to switch modes. This makes any difference? 
[test_force_hid.zip](https://github.com/hrvach/deskhop/files/14087867/test_force_hid.zip)

OrpheeGT, 2024-01-29:
Same behaviour... whereas it works OK with Logitech G500s.

hrvach, 2024-01-29:
I have just one idea left for debugging in the blind

[Uploading test_force_hid2.zip…]()

OrpheeGT, 2024-01-29:
No change, same behaviour.

OrpheeGT, 2024-01-29:
I'm willing to do any debugging you need, just tell me what you want me to do.

OrpheeGT, 2024-01-29:
Debug pin are soldered and ready to use, if you guide me with the needed hardware to plug it (that I might already have)

hrvach, 2024-01-29:
Write this [debug firmware](https://github.com/hrvach/deskhop/files/14089468/debug.zip) which will create additional serial port for output, then open your serial port of choice (screen -L /dev/ttyACM0 or similar works), plug in the mouse, wait a few seconds, click buttons and move it a bit... upload the output here as a zipfile.

OrpheeGT, 2024-01-29:
```
# cat screenlog.0 
id 0
  Bind EP 03 to driver id 0
  Bind EP 83 to driver id 0
Full Speed
[1:0] Open EP0 with Size = 8
Get 8 byte of Device Descriptor
[1:0] Get Descriptor: 80 06 00 01 00 00 08 00 
[1:0] Control data:
  0000:  12 01 00 02 00 00 00 40                          |.......@|

Set Address = 1
[1:0] Set Address: 00 05 01 00 00 00 00 00 

[1:1] Open EP0 with Size = 64
Get Device Descriptor
[1:1] Get Descriptor: 80 06 00 01 00 00 12 00 
[1:1] Control data:
  0000:  12 01 00 02 00 00 00 40 1C 1B 8B 1B 21 03 01 02  |.......@....!...|
  0010:  03 01                                            |..|

Get Configuration[0] Descriptor (9 bytes)
[1:1] Get Descriptor: 80 06 00 02 00 00 09 00 
[1:1] Control data:
  0000:  09 02 42 00 02 01 00 A0 96                       |..B......|

Get Configuration[0] Descriptor
[1:1] Get Descriptor: 80 06 00 02 00 00 42 00 
[1:1] Control data:
  0000:  09 02 42 00 02 01 00 A0 96 09 04 00 00 01 03 01  |..B.............|
  0010:  02 00 09 21 11 01 00 01 22 AC 00 07 05 81 03 40  |...!...."......@|
  0020:  00 01 09 04 01 00 02 03 00 00 00 09 21 11 01 00  |............!...|
  0030:  01 22 1D 00 07 05 82 03 40 00 01 07 05 02 03 40  |."......@......@|
  0040:  00 01                                            |..|

Parsing Configuration descriptor (wTotalLength = 66)
[1] HID opening Interface 0
  Open EP 81 with Size = 64
  HID opened
  Bind EP 81 to driver id 0
[1] HID opening Inter  Open EP 82 with Size = 64
  Open EP 02 with Size = 64
  HID   Bind EP 02 to driver id 0
Set Configuration = 1
[1:1] Set Co09 01 00 00 00 00 00 

Device configured
HID set config: itf = 0
HID Set Idle 
[1:1] Class Request: 21 0A 00 00 00 00 00 00 

HID Set Protocol = 0
[1:1] Class Request: 21 0B 00 00 00 00 00 00 

HID Get Report Descriptor
[1:1] Get Descriptor: 81 06 00 22 00 00 AC 00 
[1:1] Control data:
  0000:  05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05  |..............).|
  0010:  15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01  |..%...u.....u...|
  0020:  05 01 09 30 09 31 09 38 15 81 25 7F 75 08 95 03  |...0.1.8..%.u...|
  0030:  81 06 95 03 75 08 81 01 C0 C0                    |....u.....|

initial mouse protocol: 0
HID Set Protocol = 1
[1:1] Class Request: 21 0B 01 00 00 00 00 00 
buttons offset: 0
buttons size: 5
x offset: 8
x size: 8
y offset: 16
y size: 8
wheel offset: 24
wheel size: 10
uses_report_id: 0, report_id: 0
HID set config:HID Set Idle 
rr: 
Rx:0, y:0, w:0, b:0

set_protocol_complete_cb protocol = 1
set protocol cb: 1, uses_report_id: 0, report_id: 0
rr: 01 00 00 00 00 01 00 01 00 00 
Rx:0, y:0, w:0, b:1
rr: 01 00 00 00 00 01 00 00 00 00 
Rx:0, y:0, w:0, b:1
```

With 2 mice connected, G500s on computer, and Corsair on deskhop, as soon as I plug the Corsair to deskhop, the left click is stuck clicked on the G500s, I can't left unclick anymore, I have to unplug the micro USB from board B to get back left click

OrpheeGT, 2024-01-29:
Here are the debug logs from the keyboard USB on board A :

I was in BIOS mode, I ran the screen command, unplugged the keyboard, switched it to normal mode (with other board help)

Plugged it in to USB A

then once no more logs on screen, switched back to BIOS mode to get back working keyboard

```
# cat screenlog.0
[1:0:0] USBH DEVICE REMOVED

Device unplugged address = 1

  HIDh close addr = 1 index = 0

[1:] USBH DEVICE ATTACH

Full Speed

[1:0] Open EP0 with Size = 8

Get 8 byte of Device Descriptor

[1:0] Get Descriptor: 80 06 00 01 00 00 08 00 

[1:0] Control data:

  0000:  12 01 00 02 00 00 00 40                          |.......@|

Set Address = 1

[1:0] Set Address: 00 05 01 00 00 00 00 00 

[1:1] Open EP0 with Size = 64

Get Device Descriptor

[1:1] Get Descriptor: 80 06 00 01 00 00 12 00 

[1:1] Control data:

  0000:  12 01 00 02 00 00 00 40 1C 1B 20 1B 08 03 01 02  |.......@.. .....|

  0010:  03 01                                            |..|

Get Configuration[0] Descriptor (9 bytes)

[1:1] Get Descriptor: 80 06 00 02 00 00 09 00 

[1:1] Control data:

  0000:  09 02 42 00 02 01 00 A0 FA                       |..B......|

Get Configuration[0] Descriptor

[1:1] Get Descriptor: 80 06 00 02 00 00 42 00 

[1:1] Control data:

  0000:  09 02 42 00 02 01 00 A0 FA 09 04 00 00 01 03 01  |..B.............|

  0010:  01 00 09 21 11 01 00 01 22 84 00 07 05 81 03 40  |...!...."......@|

  0020:  00 01 09 04 01 00 02 03 00 00 00 09 21 11 01 00  |............!...|

  0030:  01 22 1D 00 07 05 82 03 40 00 01 07 05 02 03 40  |."......@......@|

  0040:  00 01                                            |..|

[... 241 lines cut ...]

HID Set Report complete

HID Set Report: id = 0, type = 2, len = 1

[1:1] Class Request: 21 09 00 02 00 00 01 00 

[1:1] Control data:

  0000:  07                                               |.|

HID Set Report complete

HID Set Report: id = 0, type = 2, len = 1

[1:1] Class Request: 21 09 00 02 00 00 01 00 

HID Set Report: id = 0, type = 2, len = 1

[1:1] Control data:

  0000:  01                                               |.|

HID Set Report complete
```

OrpheeGT, 2024-01-29:
Adding log from mouse with movements and clicks

[screenlog.0.mouse.txt](https://github.com/hrvach/deskhop/files/14089850/screenlog.0.mouse.txt)

hrvach, 2024-01-29:
Ok, I still don't know why such behavior but I can tell you what's going on. 

The mouse descriptor you sent earlier from the computer is:

```
Usage Page (Desktop),               ; Generic desktop controls (01h)
Usage (Mouse),                      ; Mouse (02h, application collection)
Collection (Application),
    Report ID (1),
    Usage (Pointer),                ; Pointer (01h, physical collection)
    Collection (Physical),
        Usage Page (Button),        ; Button (09h)
        Usage Minimum (01h),
        Usage Maximum (20h),
        Logical Minimum (0),
        Logical Maximum (1),
        Report Count (32),
        Report Size (1),
        Input (Variable),
        Usage Page (Desktop),       ; Generic desktop controls (01h)
        Usage (X),                  ; X (30h, dynamic value)
        Usage (Y),                  ; Y (31h, dynamic value)
        Logical Minimum (-32767),
        Logical Maximum (32767),
        Report Size (16),
        Report Count (2),
        Input (Variable, Relative),
        Usage (Wheel),              ; Wheel (38h, dynamic value)
        Logical Minimum (-127),
        Logical Maximum (127),
        Report Size (8),
        Report Count (1),
        Input (Variable, Relative),
    End Collection,
End Collection,
```
... etc (it's a bit longer, it's even a keyboard)

What the box gets is:
```
[1:1] Get Descriptor: 81 06 00 22 00 00 AC 00
[1:1] Control data:
  0000:  05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05  |..............).|
  0010:  15 00 25 01 95 05 75 01 81 02 95 01 75 03 81 01  |..%...u.....u...|
  0020:  05 01 09 30 09 31 09 38 15 81 25 7F 75 08 95 03  |...0.1.8..%.u...|
  0030:  81 06 95 03 75 08 81 01 C0 C0                    |....u.....|
```

which translates to:

```
Usage Page (Desktop),               ; Generic desktop controls (01h)
Usage (Mouse),                      ; Mouse (02h, application collection)
Collection (Application),
    Usage (Pointer),                ; Pointer (01h, physical collection)
    Collection (Physical),
        Usage Page (Button),        ; Button (09h)
        Usage Minimum (01h),
        Usage Maximum (05h),
        Logical Minimum (0),
        Logical Maximum (1),
        Report Count (5),
        Report Size (1),
        Input (Variable),
        Report Count (1),
        Report Size (3),
        Input (Constant),
        Usage Page (Desktop),       ; Generic desktop controls (01h)
        Usage (X),                  ; X (30h, dynamic value)
        Usage (Y),                  ; Y (31h, dynamic value)
        Usage (Wheel),              ; Wheel (38h, dynamic value)
        Logical Minimum (-127),
        Logical Maximum (127),
        Report Size (8),
        Report Count (3),
        Input (Variable, Relative),
        Report Count (3),
        Report Size (8),
        Input (Constant),
    End Collection,
End Collection
```

but the reports sent are based on the descriptor you sent earlier from the PC

01 00 00 00 00 FF FF 00 00 00
(report id 01, 4 bytes buttons, 2 bytes X, 2 bytes Y and wheel)

I think what happens here is this - mouse sends some sort of simplified boot protocol report, then gets switched to hid report but descriptor is not re-read and data gets misinterpreted.

OrpheeGT, 2024-01-29:
Okay... any ideas ?

what about keyboard output logs ? :)

hrvach, 2024-01-29:
Let's try to get one working first :) It might be reasonable to add a check - if provided report length doesn't match what the descriptor specifies it should be, re-read the descriptor. Now the tricky part is, how to cleanly do this with tinyusb :)

hrvach, 2024-01-30:
Can you try [yet another one](https://github.com/hrvach/deskhop/files/14096490/test_b_default.zip) that probably won't work with this stubborn rodent? :) Important thing - make sure to plug mouse in B.

OrpheeGT, 2024-01-30:
It works !

hrvach, 2024-01-30:
Awesome! I think I have an idea what's happening but I'll need to test-drive this change for a few days before committing it. :)

OrpheeGT, 2024-01-30:
Now the keyboard :p

OrpheeGT, 2024-01-31:
I'm having an issue I did not have until now, at least I did not catch it earlier...

On my Ubuntu Cinnamon, running Nemo as File Manager, after some time/hours, org.Nemo process hangs... and I can't open it anymore, I must reboot the OS.

Reading from internet, it is usually to be related to some USB mount issue...

So I unplugged Deskhop and I will check in some hours how Nemo is working or not...

If it works fine, I will put back original 0.42 FW and test it with standard Logitech G500s...

I'll keep you updated, it may be totally unrelated to Deskhop.

hrvach, 2024-01-31:
I assume the usb mount issue for the Nemo file manager is mostly related to storage. If you updated the firmware, rp2040 bootloader will exit abruptly and disappear from the system which might in theory get the file manager to get stuck on trying to do something with it. You can try attaching to it with strace -p <pid> to see what it's doing when it hangs, it can point in the right direction.

OrpheeGT, 2024-01-31:
I'm sorry but I'm not sure to follow you...
which firmware ?

I should run strace -p on Nemo's pid ?

hrvach, 2024-01-31:
If you were updating the pico board firmware by pressing the button and connecting it, that made the rp2040 bootloader create a usb storage device, you drag .uf2 to it and then the device abruptly disappears from the system. Maybe something like that triggered it... check by running "mount" if you still have any devices mounted.

OrpheeGT, 2024-01-31:
Well I don't think it is this then.

I did not play with FW since yesterday, and I already had to reboot the OS...

The only thing I did is enabling anti screensaver mode (I love it, don't ask why :p)

Currently running strace, but I don't know when it happens, if it happens while process is running, or when I try to re-open File manager after a period of time I closed it.

I have to find reproducible steps to trigger it. Currently I'm swimming in the sea...

OrpheeGT, 2024-02-01:
Well, plugged since yesterday, no issue yet... I can't reproduce it...

Maybe it was actually related to FW update / storage boot... wait & see... thanks !

If you need anything for my keyboard issue, I'm available :D

hrvach, 2024-02-01:
That's a more complicated one - it refuses to negotiate working in boot mode and full descriptor parsing, nkro support etc. need to be added. Should happen but will likely take some time...

OrpheeGT, 2024-02-01:
I understand, I'm in "BIOS" mode currently.

May I ask how to enable Caps Lock on computer ? (not switching from one to another but real caps lock)

I may suggest "Pause/break" Key instead of Caps lock ?

[image]

I never used this key... :)

hrvach, 2024-02-01:
There is src/user_config.h where you can define

#define HOTKEY_TOGGLE HID_KEY_CAPS_LOCK

and you can find the keys here: https://github.com/hathach/tinyusb/blob/master/src/class/hid/hid.h

The problem is, many keyboards are smaller than TKL nowadays and don't have that key anymore - choosing a good hotkey is a nightmare. I've thought perhaps using left shift+caps lock if you actually need caps lock.  

Btw, nice graphic! :)

OrpheeGT, 2024-02-01:
> I've thought perhaps using left shift+caps lock if you actually need caps lock.

It would do the job :)

OrpheeGT, 2024-04-09:
Hello :)

Is there any progress with keyboard compatibility ?
I tried latest 0.51 FW, and I still must use keyboard BIOS mode to make it works.

Thanks :)

hrvach, 2024-04-09:
You have NKRO if I remember correctly, some of the problematic keyboards got fixed in this release but NKRO is still pending :) Sorry about that.

OrpheeGT, 2024-04-09:
I don't know what is NKRO but I thrust you on that :D
Don't be sorry, you're doing your best <3

hrvach, 2024-04-09:
> I don't know what is NKRO but I thrust you on that :D Don't be sorry, you're doing your best <3

NKRO is n-key rollover, basically keyboards before that could only transmit up to 6 keys pressed at once, and apparently that wasn't enough, so nkro sends a bit array of all keys so you can press as many as you want. I never understood how 6 keys at once would be a limitation, maybe some stenography software or gaming shortcuts.

I need to rewrite the hid parser to something more universal in order to support keyboard descriptors as well, and not break anything :)

OrpheeGT, 2024-04-09:
Ah OK
So you are probably right, it is a mechanical gaming keyboard... advertised to to be able to send as many key wanted/needed like all new mechanical gaming keyboard it seems.

Looking right now at keyboard specs from corsair website :

Key Rollover : Full Key (NKRO) with 100% Anti-Ghosting

OrpheeGT, 2024-05-24:
Hello,

 @hrvach I tried 0.60 release, my mouse went crazy with it. whereas it works with 0.51 :)

I couln't try NKRO support.

As I lose mouse support, it seems the default CAPS shortcut to switch between computers does not work anymore on 0.60

hrvach, 2024-05-24:
Thanks for trying it out, I must have made a mistake of some sort. Apologies!

OrpheeGT, 2024-05-24:
> Thanks for trying it out, I must have made a mistake of some sort. Apologies!

as a reminder my mouse is a Corsair Scimitar Elite you had to work on and fix it... as we discussed above in this issue already.

hrvach, 2024-05-26:
cc: @OrpheeGT  apologies to bug you, I'd appreciate some [help with testing a potential fix if you have a moment?](https://github.com/hrvach/deskhop/discussions/94)

OrpheeGT, 2024-05-26:
@hrvach Hi !
Just tried it, and the mouse is facing the same bug as it had in the beginning.
No reaction unless I click on a mouse button, then mouse just move to the right as long as I keep clicked on the button.

Keyboard does not seems to react in NKRO mode. still needs to be in BIOS mode for it to work.

hrvach, 2024-05-26:
> @hrvach Hi ! Just tried it, and the mouse is facing the same bug as it had in the beginning. No reaction unless I click on a mouse button, then mouse just move to the right as long as I keep clicked [...]

Managed to get my keyboard working in NKRO, but I'm probably still missing something (obvious). Really appreciate your help with testing, back to the drawing board... :)

OrpheeGT, 2024-05-26:
You did fix the mouse issue here :

https://github.com/hrvach/deskhop/issues/45#issuecomment-1916559083

OrpheeGT, 2024-05-26:
And in case you forgot, my mouse is quite tricky with is numpad integrated keyboard : 

[image]

hrvach, 2024-05-26:
What I'm trying to do is remove the need for that fix in the first place (and failing obviously). :-)

OrpheeGT, 2024-06-25:
@hrvach 

Here is in NKRO mode :

```
root@UbuntuOS:~# usbhid-dump --entity=descriptor --model=1b1c:1b20
009:036:001:DESCRIPTOR         1719301291.647799
 06 C2 FF 09 04 A1 01 15 00 26 FF 00 95 40 75 08
 09 02 B1 02 09 02 81 02 09 04 91 02 C0

009:036:000:DESCRIPTOR         1719301291.650179
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 97 15 00 25 01
 75 01 95 98 81 02 05 08 19 01 29 03 25 01 75 01
 95 03 91 02 75 05 95 01 91 01 C0 05 0C 09 01 A1
 01 85 02 05 0C 19 00 2A FF 0F 15 00 26 FF 0F 75
 10 95 02 81 00 C0 06 C0 FF 09 02 A1 01 85 03 15
 00 26 FF 00 95 3F 75 08 09 02 81 02 C0 06 C2 FF
 09 03 A1 01 85 0E 15 00 26 FF 00 95 3F 75 08 09
 03 81 02 C0

root@UbuntuOS:~# usbhid-dump --entity=stream --model=1b1c:1b20
Starting dumping interrupt transfer stream
with 1 minute timeout.

009:036:000:STREAM             1719301347.991040
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.052228
 01 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.126112
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.160060
 01 00 00 08 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.239050
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.486133
 01 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.550046
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.787117
 01 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301349.868069
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301350.002032
 01 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301350.071031
 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00 00

009:036:000:STREAM             1719301350.112055
[... 489 lines cut ...]

009:040:000:STREAM             1719301593.814883
 40 00 00 00 00 00 00 00

009:040:000:STREAM             1719301593.862758
 00 00 00 00 00 00 00 00

009:040:000:STREAM             1719301594.030765
 00 00 08 00 00 00 00 00

009:040:000:STREAM             1719301594.102691
 00 00 00 00 00 00 00 00

009:040:000:STREAM             1719301594.518707
 40 00 00 00 00 00 00 00

009:040:000:STREAM             1719301594.694733
 40 00 08 00 00 00 00 00

009:040:000:STREAM             1719301594.774750
 40 00 00 00 00 00 00 00

009:040:000:STREAM             1719301594.822742
 00 00 00 00 00 00 00 00
```

OrpheeGT, 2024-06-29:
Hi ! 

As I wrote in the beta release, I was just really dumb and I did not use the right FW...

It works perfectly fine with latest 0.61 FW, even the integrated numeric keyboard on my Corsair Scimitar Elite Mouse works almost fine : 
[image]

matching values / keys : 1 2 3 4 5 6 7 8 9

not matching keys : 
10 = 0
11 = -
12 = +

But this is not really a big issue, I never used it as is, but as macro keys under windows with corsair software.
As with DeskHop corsair software does not work at all (it make sense as the product id is not corsair anymore) the macro key/driver is inactive.

I think we can consider this issue as resolved.

On my Corsair Keyboard even Fn media keys works fine ;)