hrvach/deskhop #47: Logitech MX Mechanical

jalmeroth, 2024-02-15:
Hi there,

thanks for this awesome project! :rocket: 

Unfortunately, I have problems running my `Logitech MX Mechanical` keyboard with latest firmware release. So I followed the [troubleshooting guide](https://github.com/hrvach/deskhop/wiki/Troubleshooting) and it works using the provided test firmware.

So following the guide, here are the descriptors from `usbhid-dump` while the USB dongle is directly connected to a Linux maschine:

> 003:030:002:DESCRIPTOR         1707999030.135855
>  06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
>  FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
>  A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
>  00 09 02 91 00 C0
> 
> 003:030:001:DESCRIPTOR         1707999030.136754
>  05 01 09 02 A1 01 85 02 09 01 A1 00 95 10 75 01
>  15 00 25 01 05 09 19 01 29 10 81 02 95 02 75 10
>  16 01 80 26 FF 7F 05 01 09 30 09 31 81 06 95 01
>  75 08 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38
>  02 81 06 C0 C0 05 0C 09 01 A1 01 85 03 95 02 75
>  10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
>  09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
>  09 81 09 83 81 00 75 01 15 00 25 01 09 9B 81 06
>  75 05 81 03 C0
> 
> 003:030:000:DESCRIPTOR         1707999030.137775
>  05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
>  75 01 95 08 81 02 95 05 05 08 19 01 29 05 91 02
>  95 01 75 03 91 03 95 70 75 01 05 07 19 04 29 73
>  81 02 95 05 19 87 29 8B 81 02 95 03 19 90 29 92
>  81 02 C0

Do you need anything else? Do you need the descriptor of `deskhop` while the connector is connected to it?

Thanks again,
Jan

hrvach, 2024-02-15:
Thanks for providing the USB HID descriptors! I feel like playing doctor when I say this, but can you please describe the symptoms you're having?

I need some basic information - is it just a keyboard, a keyboard + mouse on a single dongle? Which port is it plugged into? Thanks!

jalmeroth, 2024-02-15:
> Thanks for providing the USB HID descriptors! I feel like playing doctor when I say this, but can you please describe the symptoms you're having?
> 
> I need some basic information - is it just a keyboard, a keyboard + mouse on a single dongle? Which port is it plugged into? Thanks!

Whoa. What a fast reply, thank you!

The issues are that I can't switch computers using Caps Lock. And when I open a terminal and try type character `s` it results in an `a`. Backspace will becaome `[18~` and Caps lock is `3`.

jalmeroth, 2024-02-15:
Also: If I connect a different cabled keyboard to `deskhop` everything works as expected.

hrvach, 2024-02-15:
I believe what's happening is this - your keyboard is also a mouse, and when you plug in the dongle, the mouse gets configured to hid report protocol which also reconfigures the keyboard with it. Until I fix this in a cleaner way, can you try this:

[test-logitech.zip](https://github.com/hrvach/deskhop/files/14297146/test-logitech.zip)

jalmeroth, 2024-02-15:
This seems to work :+1: Very cool, thanks once more!

hrvach, 2024-02-15:
Thank you for confirming - I have to stop assuming a keyboard is just a keyboard :)

jalmeroth, 2024-02-15:
Well, I think it's the dongle`s fault. It can be used universally and therefore all dongles register themselfs trice, without respecting what device get's connected to it.

hrvach, 2024-02-15:
That wouldn't normally be a problem, but I think it can't do keyboard in one protocol and mouse in another simultaneously. I have to implement full hid report for keyboards finally and the support should improve for such edge cases.

jalmeroth, 2024-04-02:
@hrvach would you mind sharing your magic code in a patch or something? It seems the latest release is broken for me again. Thank you!

hrvach, 2024-04-02:
Build with enforce_ports set to 1 in user config file. Sorry for not documenting more explicitly. Let me know if that works...

jalmeroth, 2024-04-02:
Awesome, this works as advertised! Thank you for all the good stuff you do here :clap:

jalmeroth, 2024-04-08:
As my media keys are not working, I dug a bit into the debug mode and found something (hopefully) interesting:
```
[1:1] Class Request: 21 0A 00 00 02 00 00 00
[1:1] Control STALLED, xferred_bytes = 0
21 0A 00 00 02 00 00 00
```
At some point during device enumeration a request gets stalled. My assumption is, that this is the consumer control device and therefore DH doesn't even recongnize the events for this device. Please find the full [screenlog.txt](https://github.com/hrvach/deskhop/files/14911218/screenlog.txt) of re-attaching the Logitech dongle.

Let me know, if I can provide any other information to this.

hrvach, 2024-04-09:
The thing is, consumer control device is not a keyboard, and as such it's neither processed nor advertised to the output computer. I'm not sure if consumer control is universal or Logitech-specific and would only function with a device using their vendor id. I don't have anything with media keys to further investigate, I'll see if anyone I know has one to lend.

jalmeroth, 2024-04-09:
I have a spare Logitech Dongle, which I could send you, but unfortunately no spare device. :sweat_smile: 

But I had a look at the [hid-remapper](https://github.com/jfedor2/hid-remapper) project and I can use media keys with it. Here are their descriptors:

```
003:092:001:DESCRIPTOR         1712658819.398891
 06 00 FF 09 20 A1 01 09 20 85 64 75 08 95 20 B1
 02 C0 09 21 A1 01 09 21 85 65 75 08 95 3F 81 02
 C0

003:092:000:DESCRIPTOR         1712658819.399808
 05 01 09 02 A1 01 05 01 09 02 A1 02 85 01 09 01
 A1 00 05 09 19 01 29 08 95 08 75 01 25 01 81 02
 05 01 09 30 09 31 95 02 75 10 16 00 80 26 FF 7F
 81 06 A1 02 85 63 09 48 95 01 75 02 15 00 25 01
 35 01 45 78 B1 02 85 01 09 38 35 00 45 00 16 00
 80 26 FF 7F 75 10 81 06 C0 A1 02 85 63 09 48 75
 02 15 00 25 01 35 01 45 78 B1 02 35 00 45 00 75
 04 B1 03 85 01 05 0C 16 00 80 26 FF 7F 75 10 0A
 38 02 81 06 C0 C0 C0 C0 05 01 09 06 A1 01 85 02
 05 07 19 E0 29 E7 15 00 25 01 75 01 95 08 81 02
 19 04 29 73 95 70 81 02 19 87 29 8B 95 05 81 02
 09 90 09 91 95 02 81 02 95 01 81 03 85 62 05 08
 95 05 19 01 29 05 91 02 95 01 75 03 91 03 C0 05
 0C 09 01 A1 01 85 03 15 00 25 01 09 B5 09 B6 09
 B7 09 CD 09 E2 09 E9 09 EA 75 01 95 07 81 02 05
 0B 09 2F 95 01 81 02 C0
```

I also used successfully used this example from https://github.com/sekigon-gonnoc/Pico-PIO-USB/tree/main/examples/host_hid_to_device_cdc

So they must do something differently.

hrvach, 2024-04-09:
This is the descriptor:

Collection (Application),
    Report ID (3),
    Logical Minimum (0),
    Logical Maximum (1),
    Usage (Scan Next Track),                    ; Scan next track (B5h, one-shot control)
    Usage (Scan Previous Track),                ; Scan previous track (B6h, one-shot control)
    Usage (Stop),                               ; Stop (B7h, one-shot control)
    Usage (Play Pause),                         ; Play/pause (CDh, one-shot control)
    Usage (Mute),                               ; Mute (E2h, on/off control)
    Usage (Volume Inc),                         ; Volume increment (E9h, re-trigger control)
    Usage (Volume Dec),                         ; Volume decrement (EAh, re-trigger control)
    Report Size (1),
    Report Count (7),
    Input (Variable),
    Usage Page (Telephony),                     ; Telephony (0Bh)
    Usage (2Fh),
    Report Count (1),
    Input (Variable),
End Collection

Can you usbhid-dump your keyboard while pressing the media keys and share?

jalmeroth, 2024-04-09:
Also, what do you think about building an example implementation of a virtual device (in code) where we can put certain descriptors to simulate problematic devices?

jalmeroth, 2024-04-09:
> Can you usbhid-dump your keyboard while pressing the media keys and share?

Not so easy without pressing enter to start :wink: I used a loop:
> while true; usbhid-dump; echo "======"; sleep 1; end > usbhid-dump.log

```
003:095:002:DESCRIPTOR         1712660829.769524
 06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
 00 09 02 91 00 C0

003:095:001:DESCRIPTOR         1712660829.771578
 05 01 09 02 A1 01 85 02 09 01 A1 00 95 10 75 01
 15 00 25 01 05 09 19 01 29 10 81 02 95 02 75 10
 16 01 80 26 FF 7F 05 01 09 30 09 31 81 06 95 01
 75 08 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38
 02 81 06 C0 C0 05 0C 09 01 A1 01 85 03 95 02 75
 10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
 09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
 09 81 09 83 81 00 75 01 15 00 25 01 09 9B 81 06
 75 05 81 03 C0

003:095:000:DESCRIPTOR         1712660829.773283
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 05 05 08 19 01 29 05 91 02
 95 01 75 03 91 03 95 70 75 01 05 07 19 04 29 73
 81 02 95 05 19 87 29 8B 81 02 95 03 19 90 29 92
 81 02 C0
```

jalmeroth, 2024-04-09:
And here's what hid-remapper logs
[image]

hrvach, 2024-04-09:
Try usbhid-dump -e stream, interested in what the actual report looks like. I'll try getting some hardware and it will be much easier to debug and work on it then.

jalmeroth, 2024-04-09:
Here we go with Play/Pause

```
003:098:001:STREAM             1712662934.264778
 03 CD 00 00 00

003:098:001:STREAM             1712662934.354794
 03 00 00 00 00
```

jalmeroth, 2024-04-09:
And here's the output from the [capture_hid_report](https://github.com/sekigon-gonnoc/Pico-PIO-USB/blob/main/examples/capture_hid_report/capture_hid_report.c) example, when pressing `playpause`:

```
Root 0 connected
Device 0 Connected
control in[complete]
Enumerating 046d:c548, class:0, address:1
control out[complete]
control in[complete]
control in[complete]
Manufacture:Logitech
control in[complete]
control in[complete]
Product:USB Receiver
control in[complete]
control in[complete]
control out[complete]
inum:0, altsetting:0, numep:1, iclass:3, isubclass:1, iprotcol:1, iface:0
	bcdHID:1.11, country:0, desc num:1, desc_type:34, desc_size:67
control out[complete]
control in[complete]
		Report descriptor:05 01 09 06 a1 01 05 07 19 e0 29 e7 15 00 25 01 75 01 95 08 81 02 95 05 05 08 19 01 29 05 91 02 95 01 75 03 91 03 95 70 75 01 05 07 19 04 29 73 81 02 95 05 19 87 29 8b 81 02 95 03 19 90 29 92 81 02 c0
			epaddr:0x81, attr:3, size:64, interval:1
inum:1, altsetting:0, numep:1, iclass:3, isubclass:1, iprotcol:2, iface:0
	bcdHID:1.11, country:0, desc num:1, desc_type:34, desc_size:133
control out[complete]
control in[complete]
		Report descriptor:05 01 09 02 a1 01 85 02 09 01 a1 00 95 10 75 01 15 00 25 01 05 09 19 01 29 10 81 02 95 02 75 10 16 01 80 26 ff 7f 05 01 09 30 09 31 81 06 95 01 75 08 15 81 25 7f 09 38 81 06 95 01 05 0c 0a 38 02 81 06 c0 c0 05 0c 09 01 a1 01 85 03 95 02 75 10 15 01 26 ff 02 19 01 2a ff 02 81 00 c0 05 01 09 80 a1 01 85 04 95 01 75 02 15 01 25 03 09 82 09 81 09 83 81 00 75 01 15 00 25 01 09 9b 81 06 75 05 81 03 c0
			epaddr:0x82, attr:3, size:64, interval:1
inum:2, altsetting:0, numep:1, iclass:3, isubclass:0, iprotcol:0, iface:0
	bcdHID:1.11, country:0, desc num:1, desc_type:34, desc_size:54
control out[complete]
control in[complete]
		Report descriptor:06 00 ff 09 01 a1 01 85 10 95 06 75 08 15 00 26 ff 00 09 01 81 00 09 01 91 00 c0 06 00 ff 09 02 a1 01 85 11 95 13 75 08 15 00 26 ff 00 09 02 81 00 09 02 91 00 c0
			epaddr:0x83, attr:3, size:64, interval:1
046d:c548 EP 0x81:	00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
046d:c548 EP 0x82:	03 cd 00 00 00
046d:c548 EP 0x83:	11 01 04 00 01 01 01 00 00 00 00 00 00 00 00 00 00 00 00 00
046d:c548 EP 0x82:	03 00 00 00 00
046d:c548 EP 0x83:	11 01 07 00 2d 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

jalmeroth, 2024-04-22:
@hrvach after a lot of debugging, I finally start to understand what's going on with this device… :sweat_smile: 

With `ENFORCE_PORTS` set to `0` the dongle gets enumerated like this:

- Interface 0 is identified as keyboard with boot protocol (default)
- Interface 1 is identified as mouse with boot protocol (default), but we activate report protocol for more info

Unfortunately, setting the report protocol on interface 1 affects the reporting of interface 0 too and it will start to reply in report mode. :facepalm: Ergo: the keyboard report can no longer be parsed as boot report and has a strange behavior.

With `ENFORCE` set to `1` the enumeration skips the protocol switch, therefore the keyboard continues to work.

jalmeroth, 2024-05-02:
As requested [here](https://github.com/hrvach/deskhop/issues/61#issuecomment-2090180791), here are some media keys:
```
003:008:002:DESCRIPTOR         1714660961.247438
 06 00 FF 09 01 A1 01 85 10 95 06 75 08 15 00 26
 FF 00 09 01 81 00 09 01 91 00 C0 06 00 FF 09 02
 A1 01 85 11 95 13 75 08 15 00 26 FF 00 09 02 81
 00 09 02 91 00 C0

003:008:001:DESCRIPTOR         1714660961.248937
 05 01 09 02 A1 01 85 02 09 01 A1 00 95 10 75 01
 15 00 25 01 05 09 19 01 29 10 81 02 95 02 75 10
 16 01 80 26 FF 7F 05 01 09 30 09 31 81 06 95 01
 75 08 15 81 25 7F 09 38 81 06 95 01 05 0C 0A 38
 02 81 06 C0 C0 05 0C 09 01 A1 01 85 03 95 02 75
 10 15 01 26 FF 02 19 01 2A FF 02 81 00 C0 05 01
 09 80 A1 01 85 04 95 01 75 02 15 01 25 03 09 82
 09 81 09 83 81 00 75 01 15 00 25 01 09 9B 81 06
 75 05 81 03 C0

003:008:000:DESCRIPTOR         1714660961.249938
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 05 05 08 19 01 29 05 91 02
 95 01 75 03 91 03 95 70 75 01 05 07 19 04 29 73
 81 02 95 05 19 87 29 8B 81 02 95 03 19 90 29 92
 81 02 C0

Starting dumping interrupt transfer stream
with 1 minute timeout.

003:008:001:STREAM             1714660963.896400
 03 CD 00 00 00

003:008:002:STREAM             1714660963.904250
 11 01 04 00 01 01 01 00 00 00 00 00 00 00 00 00
 00 00 00 00

003:008:001:STREAM             1714660963.911260
 03 00 00 00 00

003:008:002:STREAM             1714660963.919314
 11 01 07 00 19 02 00 00 00 00 00 00 00 00 00 00
 00 00 00 00

003:008:002:STREAM             1714660963.934299
 11 01 0D 00 01 00 00 00 00 00 00 00 00 00 00 00
 00 00 00 00

003:008:001:STREAM             1714660964.676273
 03 B6 00 00 00

003:008:001:STREAM             1714660964.781362
 03 00 00 00 00

003:008:001:STREAM             1714660965.358250
 03 B5 00 00 00

003:008:001:STREAM             1714660965.449341
 03 00 00 00 00

003:008:001:STREAM             1714660965.839295
 03 E2 00 00 00

003:008:001:STREAM             1714660965.936178
 03 00 00 00 00

003:008:001:STREAM             1714660966.544330
 03 EA 00 00 00

003:008:001:STREAM             1714660966.656303
 03 00 00 00 00

003:008:001:STREAM             1714660966.964174
 03 E9 00 00 00

003:008:001:STREAM             1714660967.054258
 03 00 00 00 00

003:008:002:STREAM             1714660968.876282
 11 01 0B 00 08 03 04 00 00 00 00 00 00 00 00 00
 00 00 00 00

003:008:002:STREAM             1714660969.446274
 11 01 0B 00 08 02 04 00 00 00 00 00 00 00 00 00
 00 00 00 00

003:008:001:STREAM             1714660969.761277
 03 6F 00 00 00

003:008:001:STREAM             1714660969.866226
 03 00 00 00 00

003:008:001:STREAM             1714660970.226179
 03 70 00 00 00

003:008:001:STREAM             1714660970.324180
 03 00 00 00 00
```

jalmeroth, 2024-08-15:
@hrvach I implemented a MVP that works pretty well for discovering the correct device type from HID descriptor with my setup: <https://gist.github.com/jalmeroth/691aa8f6a4d0aecd3ba7be9bf3aa5cfb>.

I think it is universal applicable, but I'm uncertain of the current state in this project. There were a lot of changes that didn't make it into this repo yet, so I'm wondering, if I should try to integrate this change or if it's already solved.

jalmeroth, 2024-08-17:
@hrvach  I just want to mention that I put the complete MVP code here now: https://github.com/jalmeroth/deskhopl
I hope I didn't mess up with the license/credits. If so please let me know. Thanks!

hrvach, 2024-08-18:
> @hrvach I just want to mention that I put the complete MVP code here now: https://github.com/jalmeroth/deskhopl I hope I didn't mess up with the license/credits. If so please let me know. Thanks!

@jalmeroth thank you so much for taking the time and effort to work on this, I was super busy at work and then patching some known issues (v0.63), I'll take a look at your changes and try to incorporate them (with your permission). Thank you!

jalmeroth, 2024-08-18:
> @jalmeroth thank you so much for taking the time and effort to work on this, I was super busy at work and then patching some known issues (v0.63), I'll take a look at your changes and try to [...]

I'm glad if I can help in support this project! Let me know, if you need someone to test or if you have any questions. :smile:

jalmeroth, 2024-11-16:
I just checked the latest release `v0.66`:
- device discovery works (the keyboard isn't considered a mouse anymore)
- media keys are working fine too

Thank you, @hrvach! :clap:

closed 2024-11-16 by jalmeroth