hrvach/deskhop #49: Support for gaming?

STRIKERnz, 2024-02-21:
Will this ever work with games? I know this is not the main demographic for this but with my set up I wanted to use it on my main PC and a second PC that I use for media/study. I've been using Synergy but found deskhop. I put one together but once I tried to play any games I found you can't move your mouse more then a few degrees making playing anything with Keyboard and mouse not really practical.

hrvach, 2024-02-22:
It would work with something like implementing a full mouse passthrough mode. I'm trying to look into improving support for keyboards/mice currently and perhaps multiple screens on win/mac currently, let me add this to the planned features list.

ozwin2, 2024-03-19:
hey, sorry this is likely the wrong channel for this (dont know how to navigate to the planned features list), but was this something that was being looked at? kinda confused about "can't move your mouse more than a few degrees", what exactly does that mean? if gaming through the deskhop would it work fine by locking (Control+L) to the gaming system, so then the mouse wont jump across to the other system, or is this more that when playing Fullscreen your mouse disappears so that you mouse doesn't ever get to the edge to move across to the other system from the gaming system, unless the game is paused when the mouse pointer becomes visible again and so the mouse can move from the gaming system to the other system?

STRIKERnz, 2024-03-19:
So, both full screen and non full screen games were doing this. Even with the mouse locked to the gaming system, It would only allow a small range of movement within that systems 1920x1080 area/box. I'm assuming because most games hide the mouse cursor but you still have free rain to move the cursor more then to the edges of a 1080 box. You end up with a very narrow area to move around with.

STRIKERnz, 2024-03-19:
So, both full screen and non full screen games were doing this. Even with the mouse locked to the gaming system, It would only allow a small range of movement within that system's 1920x1080 area/box. It wasn't moving between screens but rather just a small area of range of movement. I'm assuming because most games hide the mouse cursor but still have free rain to move the cursor more then to the edges of a 1080 box. You end up with a very narrow area to move around with.

ozwin2, 2024-03-19:
ahh i understand, so i presume this is still the case? i guess in simple terms an if statement of: if full screen do not move cursor between systems else allow movement. obviously don't know how hard that would be to implement in code.

STRIKERnz, 2024-03-19:
here is an example [video](https://youtu.be/l4mHAwZkc1) you can only move the mouse within a box.

ozwin2, 2024-03-20:
Oh man, that's unusable like that haha!

MakeNModify, 2024-07-17:
I did not test it but would imagine the main problem is for First (and Third) Person games, where you can "rotate" into one direction by constantly moving your mouse into one direction. That works for relative movement but for absolute movements you'll hit a wall. I would propose to make a hotkey to lock the deskhop to the current output and switch to relative movement and switch back on another push.
(Am currently waiting for parts and look into it when I've build it)

hrvach, 2024-08-04:
I've been working on 
[deskhop-gaming-test.zip](https://github.com/user-attachments/files/16487631/deskhop-gaming-test.zip)
because apparently a lot of people would find this feature useful. This is just a test release, not a daily driver but I'd really appreciate some gamer feedback from you folks. (@STRIKERnz @Earthling-Z3R0 @MakeNModify and others).

Left shift + right shift + G will lock the mouse to the current output and switch it to standard relative mouse. Pressing this again will revert to the "normal" deskhop mode.

Please be so kind to focus on scenario a) where machine you're gaming on is connected to the same PICO board where the mouse plugs in and scenario b) machine you're gaming on is connected to the pico where the keyboard plugs in.

Would this be acceptable for gaming? Thanks!

superbreadi, 2024-08-05:
Hi it works perfectly for me! Also my other issue with widescreensupport https://github.com/hrvach/deskhop/issues/131#issue-2448824151 is also fixed in gaming mode

STRIKERnz, 2024-08-07:
> I've been working on [deskhop-gaming-test.zip](https://github.com/user-attachments/files/16487631/deskhop-gaming-test.zip) because apparently a lot of people would find this feature useful. This is [...]

Works for me fine in both scenarios.

nathank07, 2024-08-13:
> I've been working on [deskhop-gaming-test.zip](https://github.com/user-attachments/files/16487631/deskhop-gaming-test.zip) because apparently a lot of people would find this feature useful. This is [...]

Works for me on MacOS/Linux and fixes my issues with X. Would love to see this added as a feature in the future!

hrvach, 2024-08-14:
It will definitely be added as a feature, probably next release (as soon as I find a few free hours).

hrvach, 2024-08-24:
This was released in 0.63, closing as complete.

closed 2024-08-24 by hrvach