hrvach/deskhop #50: Dry Studio Black Diamond 75 keyboard not working

infosecconsultant, 2024-02-23:
I thought I'd try with a few other keyboard I have. Unfortunately the Dry Studio Black Diamond 75 keyboard does not register with the deskhop at all. 
I'm _guessing_ it's got issues with the power draw but I'm not sure. I don't currently have a way to measure that but I've ordered a little inline USB power monitor/tester. 
I turned off the LEDs etc but still no luck. 
If there's any advice on debugging please let me know and I'll give it a shot.

hrvach, 2024-02-23:
Try dumping the usb hid report first, maybe it's also detected as a mouse ... I had to google this keyboard out of curiosity and must admit I have never seen anything like this :) it's pretty interesting...

infosecconsultant, 2024-02-26:
I'll try to do that when I'm back tomorrow. 
This output? 

```lsusb -v
usbhid-dump
```

The keyboard supports Bluetooth... Is there any chance the raspi could pair a bluetooth keyboard? Or have a bluetooth dongle attached to it?

hrvach, 2024-02-27:
Unless a BT dongle presents as a usb keyboard, I doubt that would work, but I was working on pico W BT support, mostly in the other direction (use BT as client so you need only one pico). 

Trying to do support for non-boot keyboards as well, that should sort out most issues with a few stubborn keyboard models.

hrvach, 2024-04-01:
Any change with the latest fw?

infosecconsultant, 2024-04-02:
Hey hrvach, I've got some other firmware issues with that keyboard unfortunately. I'll have to try again once the RMA'd unit comes though in a couple of weeks. Sorry :(

infosecconsultant, 2024-04-19:
Hey @hrvach 

Still no luck with the keyboard. Not registering any keypresses at all and I've updated to the most recent release of firmware. 

The following output is from a linux machine plugged into the 'A' port of the deskhop.

usbdhid-dump

```
001:005:001:DESCRIPTOR         1713489329.878272
 05 01 09 02 A1 01 85 01 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 95 05 75 01 81 02 95 01 75 03
 81 01 05 01 09 30 09 31 15 81 25 7F 95 02 75 08
 81 06 09 38 15 81 25 7F 95 01 75 08 81 06 05 0C
 0A 38 02 15 81 25 7F 95 01 75 08 81 06 C0 C0

001:005:000:DESCRIPTOR         1713489329.879163
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 95 08 75 01 81 02 95 01 75 08 81 01 05 08
 19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01
 05 07 19 00 2A FF 00 15 00 26 FF 00 95 06 75 08
 81 00 C0 05 01 09 02 A1 01 85 02 09 01 A1 00 05
 09 19 01 29 05 15 00 25 01 95 05 75 01 81 02 95
 01 75 03 81 01 05 01 09 30 09 31 15 00 26 FF 7F
 75 10 95 02 81 02 09 38 15 81 25 7F 95 01 75 08
 81 06 95 01 75 08 81 01 C0 C0
```

And lsusb -v output

```

Bus 001 Device 005: ID 2e8a:107c Hrvoje Cavrak DeskHop Switch
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x2e8a Hrvoje Cavrak
  idProduct          0x107c DeskHop Switch
  bcdDevice            1.00
  iManufacturer           1 Hrvoje Cavrak
  iProduct                2 DeskHop Switch
  iSerial                 3 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x003b
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 [unknown]
      bInterfaceProtocol      0 
      iInterface              2 DeskHop Switch
        HID Device Descriptor:
[... 268 lines cut ...]
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0001
  Self Powered
```

I'm not sure why the device is an 'Apple' device... Bit odd. I'm testing on a mac and a debian based HP at the moment.

Please let me know if you need anything further from my end.

hrvach, 2024-04-21:
Output (Constant, Variable),
    Report Count (224),
    Report Size (1),

Your keyboard is NKRO, that's not supported yet.. it should work in boot mode though, but I suspect it's also a mouse. Can you try rebuilding with ENFORCE_PORTS set to 1 in user_config.h ?

hrvach, 2024-05-24:
@infosecconsultant is there a change with the latest experimental BETA (https://github.com/hrvach/deskhop/releases/tag/v0.60) ?