hrvach/deskhop #54: Feature request: 'jiggle' pointing device on non-active computer

kpfleming, 2024-02-27:
Many of us are forced to use inactivity timers which lock sessions on our computers (security!). There are various tools out there which claim to be able to effectively stop this behavior by 'jiggling' the pointer so it appears to be active, but those must be installed on the controlled computer so that can be a challenge itself.

Deskhop is perfectly positioned to solve this problem for us, since it is outside the controlled computer but in the path between the actual pointing device and that computer.

hrvach, 2024-02-27:
This functionality already exists - check the README describing the screensaver. Just to be clear - this is disabled by default and was not designed to work around any security policies. It's more of a fun novelty that's entertaining to watch, made after a request from my kid :-)

closed 2024-02-27 by hrvach

kpfleming, 2024-02-27:
Ha! I completely overlooked that since I had a different name for it in my head :-) Thanks.