hrvach/deskhop #56: Dual screen on macOS doesn't work

jalmeroth, 2024-03-02:
Hi,

I have two laptops and one external monitor. If I connect the monitor to the Linux-laptop, everything works as expected. I can move to all three screens: Linux-Laptop (DeskHop B) -> external monitor -> Mac  (DeskHop A).

Unfortunately, if I connect the monitor to the macOS-laptop (using the same USB-C cable as with the other laptop), I can only control one screen on the Mac. It just doesn't allow me to use the second screen. Altough I can use the built-in trackpad to move between both screens. Either it's stuck to the primary screen, or if I use the trackpad to move the mouse to the secondary screen, it just allows this one. So it's: Mac (DeskHop B) or external monitor -> Linux  (DeskHop A) but not Mac (DeskHop B) -> external monitor -> Linux  (DeskHop A).

The Mac is running Sonoma 14.3.1. What information can I provide to narrow this problem?

Fallingonion, 2024-03-08:
I would also be interested in a fix for this issue. My setup is a macbook with an extra monitor and a PC. Let;s call the laptop's built in monitor screen A, the extra Mac monitor screen B, and the PC monitor screen C.  Reading from left to right the layout is A - B - C. 
Switching between the Mac and the PC works, but it depends where the mouse is on the Mac. If the cursor is on B, then I can move from B to C and back again no problem, but cannot move the mouse from B to A. However I can use the trackpad on the Mac to move the cursor to A. 
If I then use the external mouse and try to move from A to B, it jumps from A to C and back again. Hopefully that makes sense and I believe it matches the behaviour mentioned in the other thread.
It's not fatal for me as I do most work on B and C, and it's easy enough to use the trackpad should I need to adjust anything... butr it would  be great to be able to seamlessly move across all three :)

hrvach, 2024-03-08:
Multiple mac screens workaround should be easier to implement than multiple windows screens, will try once I'm back from my work trip.

hrvach, 2024-03-22:
Ok, the updated codebase is out. You need to rebuild and define OUTPUT_A_OS/OUTPUT_B_OS to specify where your MACOS is, and define the corresponding screen_count in defaults.c 

Will be moved to config file and supported by a keyboard shortcut for easy config.

Leaving this open until confirmed working OK by several folks.

Fallingonion, 2024-03-23:
That's great, thanks, I'll give it a try and report back.
"You need to rebuild and define OUTPUT_A_OS/OUTPUT_B_OS to specify where your MACOS is, and define the corresponding screen_count in defaults.c" - sorry for the stupid question but can you clarify exactly what changes I would need to make for the A-B-C layout mentioned above? I'm very new to all this and would hate to make a mistake :)

hrvach, 2024-03-23:
B needs to be your main mac screen, A needs to be the extended one (that's defined under MacOS I believe).

Then in defaults.c for output A you would define screen_count = 2 instead of 1 and in user_defaults.h OUTPUT_A_OS as MACOS.

It's still not super user-friendly, as it's a work in progress, but should get there. I'll also try writing better instructions.

jalmeroth, 2024-03-24:
> Ok, the updated codebase is out. You need to rebuild and define OUTPUT_A_OS/OUTPUT_B_OS to specify where your MACOS is, and define the corresponding screen_count in defaults.c
>
> Will be moved to [...]

My test setup:
A1 (left): Linux
B2 (center): external Monitor
B1 (right): macOS Laptop

First notice: my keyboard, as reported in https://github.com/hrvach/deskhop/issues/47 does not work with this build.

Other then that, it kinda work… :tada:  Sometimes, when you freshly connect deskhop and I try move from A1, I actually end up on B1 first (should be B2). Moving the mouse to the right edge of B1 seems to discover that the external monitor is on the left, not the right. After that, I can move the mouse from A1 to B2 to B1 as expected. :+1: 

Is there anything I can provide? Maybe [logs from WireShark](https://wiki.wireshark.org/CaptureSetup/USB) or something?

Fallingonion, 2024-03-24:
Hm, I'm getting the following error when trying to build it:

C:\deskhop\deskhop\src\mouse.c: In function 'switch_desktop':
C:\deskhop\deskhop\src\mouse.c:139:13: error: a label can only be part of a statement and a declaration is not a statement
  139 |             mouse_report_t move_relative_one
      |             ^~~~~~~~~~~~~~
mingw32-make.exe[2]: *** [CMakeFiles\board_A.dir\build.make:173: CMakeFiles/board_A.dir/src/mouse.c.obj] Error 1
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:1587: CMakeFiles/board_A.dir/all] Error 2
mingw32-make.exe: *** [Makefile:90: all] Error 2

I thought it might be because I'm building on Windows but I tried on an Ubuntu machine and got the same error. Am I doing something wrong?

hrvach, 2024-03-24:
> Hm, I'm getting the following error when trying to build it:
>
> C:\deskhop\deskhop\src\mouse.c: In function 'switch_desktop': C:\deskhop\deskhop\src\mouse.c:139:13: error: a label can only be part of [...]

Can you please pull the latest version and try again? Initializing a struct within a switch/case doesn't seem to make some compiler versions happy. :-)

hrvach, 2024-03-24:
> > Ok, the updated codebase is out. You need to rebuild and define OUTPUT_A_OS/OUTPUT_B_OS to specify where your MACOS is, and define the corresponding screen_count in defaults.c
> > Will be moved to config file and supported by a keyboard shortcut for easy config.
> > Leaving this open until confirmed working OK by several folks.
> 
> My test setup: A1 (left): Linux B2 (center): external Monitor B1 (right): macOS Laptop
> 
> First notice: my keyboard, as reported in #47 does not work with this build.
> 
> Other then that, it kinda work… 🎉 Sometimes, when you freshly connect deskhop and I try move from A1, I actually end up on B1 first (should be B2). Moving the mouse to the right edge of B1 seems to discover that the external monitor is on the left, not the right. After that, I can move the mouse from A1 to B2 to B1 as expected. 👍
> 
> Is there anything I can provide? Maybe [logs from WireShark](https://wiki.wireshark.org/CaptureSetup/USB) or something?

@jalmeroth do these do anything for your keyboard?

[test-fw-enforce-ports.zip](https://github.com/hrvach/deskhop/files/14736847/test-fw-enforce-ports.zip)

jalmeroth, 2024-03-24:
> @jalmeroth do these do anything for your keyboard?
>
> [test-fw-enforce-ports.zip](https://github.com/hrvach/deskhop/files/14736847/test-fw-enforce-ports.zip)

Yes, keyboard works with this.

But I can't move the mouse from A1 to B2 (from left [A] to right [B2]). I have to go to the left to land on B2 but can't reach B1 at all. :shrug:

hrvach, 2024-03-24:
> > > > Ok, the updated codebase is out. You need to rebuild and define OUTPUT_A_OS/OUTPUT_B_OS to specify where your MACOS is, and define the corresponding screen_count in defaults.c
> 
> > > > Will be moved to config file and supported by a keyboard shortcut for easy config.
> 
> > > > Leaving this open until confirmed working OK by several folks.
> 
> > > 
> 
> > > 
> 
> > > My test setup: A1 (left): Linux B2 (center): external Monitor B1 (right): macOS Laptop
> 
> > > First notice: my keyboard, as reported in #47 does not work with this build.
> 
> > > Other then that, it kinda work… 🎉 Sometimes, when you freshly connect deskhop and I try move from A1, I actually end up on B1 first (should be B2). Moving the mouse to the right edge of B1 seems to discover that the external monitor is on the left, not the right. After that, I can move the mouse from A1 to B2 to B1 as expected. 👍
> 
> > > Is there anything I can provide? Maybe [logs from WireShark](https://wiki.wireshark.org/CaptureSetup/USB) or something?
> 
> > 
> 
> > @jalmeroth do these do anything for your keyboard?
> 
> > 
> 
> > [test-fw-enforce-ports.zip](https://github.com/hrvach/deskhop/files/14736847/test-fw-enforce-ports.zip)
> 
> 
> 
> @hrvach Yes, keyboard works with this.
> 
> 
> 
> But I can't move the mouse from A1 to B2 (from left [A] to right [B2]). I have to go to the left to land on B2 but can't reach B1 at all. :shrug: 

@jalmeroth Try parking cursor on mac output and pressing right shift + backspace + 2 and then right shift + backspace + 9. Any difference?

jalmeroth, 2024-03-24:
@hrvach yes, now I can move to B1 as well. But still the jump is on the wrong edge

jalmeroth, 2024-03-24:
@hrvach oh, and is moving windows on macOS (B1 <-> B2) supposed to work already?

Fallingonion, 2024-03-25:
> @Fallingonion can you please pull the latest version and try again? Initializing a struct within a switch/case doesn't seem to make some compiler versions happy. :-)

OK, so that now builds correctly and works perfectly for my setup!  Amazing, thanks :)

Fallingonion, 2024-04-01:
One side comment - on the Mac, I can't click+drag a window from one screen to the other using the mouse that goes through deskhop. The built-in trackpad works as you would expect. No idea if there's anything that can be done about that :)

hrvach, 2024-04-01:
Hmmm, maybe ... I didn't think of the mouse buttons, the relative "helper" mouse should mimic the button presses from the absolute one, then it just *might* work

hrvach, 2024-04-01:
@Fallingonion can you try this and let me know? 
[test_drag_macos.zip](https://github.com/hrvach/deskhop/files/14828499/test_drag_macos.zip)

Fallingonion, 2024-04-02:
Hi - didn't work unfortunately: I was unable to move the mouse to screen A (the laptop's built-in screen.) Movement between B and C (Mac extra screen and Windows) works as expected...

hrvach, 2024-04-02:
Did you configure that output as two screens (Right Shift + Backspace + 2), MacOS (Right Shift + Backspace + 9) ?

Fallingonion, 2024-04-02:
> Did you configure that output as two screens (Right Shift + Backspace + 2), MacOS (Right Shift + Backspace + 9) ?

Oops, no, I hadn't. I have now done that and can move the mouse to all three screens, but I still can't click+drag between the two Mac screens.
Thanks for all your help with this :)

hrvach, 2024-04-02:
I think I know what else might be needed, will try looking into it...

crablab, 2024-04-11:
> but I still can't click+drag between the two Mac screens.

++ Same. 

I also notice that sometimes when moving from the second MacOS screen to the primary, the cursor gets stuck on the second screen 🤔 It's almost like it got stuck in a second layer, as moving it to the far left edge doesn't move the cursor to the other machine, but back into the secondary MacOS screen.

hrvach, 2024-04-11:
It probably needs some tweaking to reliably move the "helper" mouse for the MacOS workaround. For the click-and-drag, I have an idea how it might be done, but still not 100% it will work... will try though! :)

Ardakilic, 2024-04-15:
Hello, I didn't want to create a new issue so I wanted to add a comment here. I have a single external monitor, and my mac's main display and external monitor are placed vertically (On top, my external monitor (through a converter since there's no usb-c direct connection), and on the bottom my mac's main integrated display). I am using Deskhop like a KVM switch, and at this stage the second pc is shut down, and on the single machine the mouse does not pass from the bottom main display to external on the same OS.

[image]

My mouse is Steelseries Rival 3, and I'm utilizing 2.4 ghz usb dongle. It works, just does not go to the top external display.

hrvach, 2024-04-15:
> Hello, I didn't want to create a new issue since I believe my issue is related, so I wanted to add a comment here. I have a single external monitor, and my mac's main display and external monitor are placed vertically (On top, my external monitor (through a converter since there's no usb-c direct connection), and on the bottom my mac's main integrated display). I am using Deskhop like a KVM switch, and at this stage the second pc is shut down, and on the single machine the mouse does not pass from the bottom main display to external on the same OS.
> My mouse is Steelseries Rival 3, and I'm utilizing 2.4 ghz usb dongle. It works, just does not go to the top external display.

You need to configure the output is a mac and there are two monitors there. Top/bottom still isn't implemented, so you need to configure displays as left-right.

akorobko, 2024-06-16:
Thanks for the cool project. Just built it and it is great so far. I confirm as well that dragging files between Mac screens does not work :(

hrvach, 2024-06-16:
That is a known issue, I have a few ideas how to potentially address it - probably will rely on relative mouse while dragging.

akorobko, 2024-06-16:
Thanks. I tried to fix it myself yesterday. And while now I can drag something to another screen (I moved check for buttons being pressed lower; and passing buttons mask to the relative move report). But as soon as I move something to another screen, the button does not depress anymore and remains sticky. Will keep trying other things and maybe do PR if I figure this out.

hrvach, 2024-06-16:
I thought I had it initially, but it turned out trickier than originally thought. Normally, you use absolute movement and relative only to switch screens, but shifting the pressed buttons between the two reports caused the dragged thing to be dropped. I wonder if I could define a relative helper mouse report **without any** buttons, that might be worth trying.

jalmeroth, 2024-11-27:
> I thought I had it initially, but it turned out trickier than originally thought. Normally, you use absolute movement and relative only to switch screens, but shifting the pressed buttons between [...]

@hrvach I did a quick test on this and I think it's a pretty neat idea.

Here's my test descriptor:
```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x08,        //     Usage Maximum (0x08)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x95, 0x08,        //     Report Count (8)
0x75, 0x01,        //     Report Size (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
0x75, 0x10,        //     Report Size (16)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x95, 0x01,        //     Report Count (1)
0x75, 0x08,        //     Report Size (8)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0C,        //     Usage Page (Consumer)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x95, 0x01,        //     Report Count (1)
0x75, 0x08,        //     Report Size (8)
0x0A, 0x38, 0x02,  //     Usage (AC Pan)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //     Report Count (1)
0x75, 0x08,        //     Report Size (8)
0x81, 0x01,        //     Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x03,        //   Report ID (3)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x95, 0x02,        //     Report Count (2)
0x75, 0x10,        //     Report Size (16)
0x16, 0x01, 0x80,  //     Logical Minimum (-32767)
0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection

// 112 bytes
```

Report ID (2) is our normal ABS-mouse
Report ID (3) is a Rel-mouse w/o buttons or wheels

Then I send some reports:

Report ID (2): Click on a window `020100000000000000`
[image]

While not releasing the button, I continue with Report ID (3): move mouse `03ffff0000`
[image]

The window starts moving…

Once we are finish moving the cursor, release the buttons. Boom, magic! :sparkle:

hrvach, 2024-11-27:
Awesome! The second idea was digitizer pen (check for descriptor at the bottom of usb_descriptors.h) for Windows. This would work across all desktops, and I remember it did work but I had some issues with mouse buttons and clicking... any thoughts?

jalmeroth, 2024-11-27:
Do you mean like this?

```
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x02,        // Usage (Mouse)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x09, 0x01,        //   Usage (Pointer)
0xA1, 0x00,        //   Collection (Physical)
0x05, 0x09,        //     Usage Page (Button)
0x19, 0x01,        //     Usage Minimum (0x01)
0x29, 0x08,        //     Usage Maximum (0x08)
0x15, 0x00,        //     Logical Minimum (0)
0x25, 0x01,        //     Logical Maximum (1)
0x95, 0x08,        //     Report Count (8)
0x75, 0x01,        //     Report Size (1)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
0x75, 0x10,        //     Report Size (16)
0x95, 0x02,        //     Report Count (2)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x09, 0x38,        //     Usage (Wheel)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x95, 0x01,        //     Report Count (1)
0x75, 0x08,        //     Report Size (8)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x0C,        //     Usage Page (Consumer)
0x15, 0x81,        //     Logical Minimum (-127)
0x25, 0x7F,        //     Logical Maximum (127)
0x95, 0x01,        //     Report Count (1)
0x75, 0x08,        //     Report Size (8)
0x0A, 0x38, 0x02,  //     Usage (AC Pan)
0x81, 0x06,        //     Input (Data,Var,Rel,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //     Report Count (1)
0x75, 0x08,        //     Report Size (8)
0x81, 0x01,        //     Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection
0x05, 0x0D,        // Usage Page (Digitizer)
0x09, 0x01,        // Usage (Digitizer)
0xA1, 0x01,        // Collection (Application)
0x85, 0x03,        //   Report ID (3)
0x09, 0x01,        //   Usage (Digitizer)
0xA1, 0x00,        //   Collection (Physical)
0x95, 0x02,        //     Report Count (2)
0x75, 0x10,        //     Report Size (16)
0x15, 0x00,        //     Logical Minimum (0)
0x26, 0xFF, 0x7F,  //     Logical Maximum (32767)
0x05, 0x01,        //     Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //     Usage (X)
0x09, 0x31,        //     Usage (Y)
0x81, 0x02,        //     Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              //   End Collection
0xC0,              // End Collection
```
Does this have to be Abs or Rel? In the `usb_descriptors.h` it's Abs, but I think it is better to move when it is Rel, no?

jalmeroth, 2024-11-27:
Ok, now you lost me. :sweat_smile: Is the windows issue documented somewhere? Unfortunately, I don't have access to a windows machine (with multi monitor setup) to test this…

hrvach, 2024-11-27:
So tl;dr, there are 3 cases - Linux just works, macOS has a thing where absolute coords apply to the current screen so a "helper mouse" could be used to nudge it between the two (this thread). Windows is the third case, since KB5003637 where Microsoft broke things and effectively made absolute mode apply to the primary screen output only (if you use a relative mouse to move to screen 2, any input from the absolute one returns you to screen 1). The current workaround is basically switch to using relative mode on screens 2 and beyond in Windows. It's not perfect but better than nothing.

However, there is a potential workaround with the digitizer pen for moving around, as that specific combo of collection/usagepage/usage will apply to a virtual desktop stretching through all of the screens (just like on Linux).  I was able to move the cursor perfectly accurate on multi-screen windows, but couldn't figure out buttons properly (and I don't use Windows either, haha). If you have any ideas, I'm all ears :)

eltariel, 2024-11-29:
Interesting wrinkle I discovered with MacOS: if you're using two mismatched or vertically offset screens, moving the mouse off the part that doesn't line up with the other screen will cause it to wrap back to the same screen instead 😵‍💫

hrvach, 2024-12-03:
> Interesting wrinkle I discovered with MacOS: if you're using two mismatched or vertically offset screens, moving the mouse off the part that doesn't line up with the other screen will cause it to wrap back to the same screen instead 😵‍💫

Very good debugging skills, kudos! I think I know why this might happen - the OS is resolution-aware and prevents switching desktops if you hit the edge and there is no screen on the left side. The box can't know this, and it does the usual "move relative left and then jump to the absolute right coord at the same height". That's why I think it shifts you to the same screen and tallies it internally as being on the leftmost screen 2.

Relative mode with internal position calculation to figure out when to switch, but this is a bit inaccurate over time, as different mouse acceleration curves cause a mismatch between the internal position and the computer one. Other option would be configuring the seconds screen "bump" and prevent switching there. What do you think?

eltariel, 2024-12-04:
> Very good debugging skills, kudos! I think I know why this might happen - the OS is resolution-aware and prevents switching desktops if you hit the edge and there is no screen on the left side. The [...]

This is pretty much the same conclusion I reached.

> Relative mode with internal position calculation to figure out when to switch, but this is a bit inaccurate over time, as different mouse acceleration curves cause a mismatch between the internal [...]

I'd rather not use relative mode with position tracking because of the inaccuracy issue. Being able to designate parts of a screen edge as "don't switch here" would work, but we might also want to move to an appropriate y-offset on the target screen to avoid jumping around vertically.

Another option is to do an absolute move to a specific vertical position before using the relative move to swap screens. For my setup it should be fine to just use the halfway point but we'd probably want to be able to use a different setting for each direction eventually. I've started implementing this myself because I like to keep the whole edge active rather than dealing with dead regions, but haven't had enough time to get very far with it yet...