hrvach/deskhop #57: 8bitDo Retro Mechanical Keyboard not working.

seii-m, 2024-03-06:
I plugged my keyboard it did not work.
As a reference, I paste the results of the following command.

# usbhid-dump --model=2dc8:5201
001:005:002:DESCRIPTOR         1709695945.563311
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 95 01 75 08 81 01 95 05
 75 01 05 08 19 01 29 05 91 02 95 01 75 03 91 01
 95 06 75 08 15 00 26 F1 00 05 07 19 00 2A F1 00
 81 00 C0 05 0C 09 01 A1 01 85 02 75 10 95 01 15
 01 26 8C 03 19 01 2A 8C 03 81 00 C0 05 01 09 80
 A1 01 85 06 09 82 09 81 09 83 15 00 25 01 19 01
 29 03 75 01 95 03 81 02 95 05 81 01 C0 05 01 09
 06 A1 01 85 0C 05 07 19 E0 29 E7 15 00 25 01 75
 01 95 08 81 02 15 00 25 01 19 00 29 77 75 01 95
 78 81 02 C0 05 01 09 06 A1 01 85 0A 05 07 19 E0
 29 E7 15 00 25 01 75 01 95 08 81 02 15 00 25 01
 19 00 29 77 75 01 95 78 81 02 C0 05 0C 09 01 A1
 01 85 11 15 00 25 01 75 01 95 03 81 01 75 01 95
 01 05 0C 09 B8 81 02 06 FF 00 09 03 81 02 75 01
 95 03 81 01 C0

001:005:001:DESCRIPTOR         1709695945.570816
 05 8C 09 01 A1 01 85 B1 09 02 15 00 26 FF 00 75
 08 95 20 81 02 85 B2 09 03 95 20 75 08 15 00 26
 FF 00 91 02 85 54 15 00 26 FF 00 09 07 75 08 95
 20 81 02 85 51 09 08 95 20 75 08 15 00 26 FF 00
 91 02 85 52 09 09 95 20 75 08 15 00 26 FF 00 91
 02 C0

001:005:000:DESCRIPTOR         1709695945.576274
 05 01 09 02 A1 01 85 03 09 01 A1 00 05 09 19 01
 29 05 15 00 25 01 75 01 95 05 81 02 75 03 95 01
 81 01 05 01 09 30 09 31 16 01 80 26 FF 7F 75 10
 95 02 81 06 09 38 15 81 25 7F 75 08 95 01 81 06
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 C0
 C0

hrvach, 2024-03-06:
I think this might be a similar issue to https://github.com/hrvach/deskhop/issues/47#issuecomment-1946046032

Please be so kind to try that build and report back if anything changes.

seii-m, 2024-03-07:
Thank you for your response.

After updating the firmware and connecting the keyboard, it was recognized, but the keys were jumbled.

hrvach, 2024-03-07:
It seems this keyboard does NKRO only and will not do USB HID Boot mode (which it should according to the standard).

https://www.reddit.com/r/8bitdo/comments/187njzt/8bitdo_retro_keyboard_and_bitlockerbios/

I'll be adding support for this, however work is a bit intense these days so it won't be the fastest.

hrvach, 2024-04-01:
Any change with the latest release?

seii-m, 2024-04-03:
I'm sorry for the delayed response. 
The latest release isn't working.

mglushko, 2026-08-19:
Root cause for this one, with a fix and a reproduction on hardware.

`get_keyboard()` in keyboard.c returns before it can reach a second slot:

```c
if (iface->num_keyboards == 1 || !iface->uses_report_id)
    return &iface->keyboards[PRIMARY_KEYBOARD];
```

Interface 2 of this keyboard declares three keyboard collections, on report IDs 1, 12
and 10. The first one parsed claims keyboards[0] and sets num_keyboards to 1, and from
then on every collection on the interface gets handed that same slot. num_keyboards can
never go above 1 once report IDs are in use, so MAX_KEYBOARDS is dead code.

The two NKRO blocks then write over the 6KRO collection's key array on the shared entry,
and because they map one usage per bit across 120 bits they set is_nkro on it too. So a
report ID 1 report gets decoded as though its bytes were bitmap bits. Keycode 0x04 sits
at payload offset 2, which is bit 2 of the bitmap's second byte, so it arrives as usage
10.

The keyboard is not dropping the odd key. It is typing different ones.

I do not own one, so I built a spare RP2040 into a stand in that presents the same 245
bytes and types on two of the three collections. Straight into a PC it types

    abcdef,./

and through deskhop it types

    gmovw3,./

The fix is to split the lookup. get_keyboard() becomes lookup only, a new
get_or_add_keyboard() claims a slot for an unseen report ID at parse time, and the
descriptor handler calls that one instead. 84 bytes of flash, no change in RAM. With it
in place the same board on the same port types abcdef,./ again, over 48 lines.

Fix: https://github.com/mglushko/deskhop-extended/pull/2
Emulator: https://github.com/mglushko/deskhop-hidtests/tree/main/emu
Build to try, if you still have the keyboard:
https://github.com/mglushko/deskhop-extended/releases/tag/v1.05

#211 and #295 look like the same fault.