hrvach/deskhop #58: extending deskhop by an open hardware usb-c hub

chri2, 2024-03-06:
I'm waiting for my PCBs and parts and hopefully will soon be able to test the **deskhop**. What a nice project, thanks a lot!

My use case seems to be a bit different from what I read here:

### mobile device desktop stand

I'm using a Librem5 mobile and a tablet running Lineage. Both offer convergence: I can use a usb-c hub and connect mouse, keyboard and even another display if the hub offers the right hdmi/displayport pass out.

This is all nice, but it doesn't cover my major use case:

Working on my desktop and wanting to just answer some chat message or look up some contact on my mobile that sits beside it.

I already tried waypipe, but this has a lot of downsides and pitfalls and needs network connectivity between the two devices.

But now I found **deskhop** that gives me the hope to be able to put together a setup like this:

* deskhop to share my keyboard and mouse between desktop and mobile
* an extension to the case to have a stand for mobile devices on top of the deskhop case
* a **usb-c hub** included into the extended deskhop case
  * usb-c pd power in to optionally charge the mobile device, power one deskhop rpi, power connected usb devices
  * a usb hub port to connect to the deskhop for mouse/keyboard
  * optional:
    * another usb hub port to connect anything like a memory stick
    * **usb-c port** to connect devices like a Nitrokey usb-c, a yubikey, a modern memory stick

I could put a usb-c hub like [this](https://shop.puri.sm/shop/usb-c-hub/) inside the case, but for its use case I wouldn't need ethernet nor hdmi and it is missing the optional **usb-c port** to connect my Nitrokey.

I started looking around for an open hardware design of a usb-c hub that could integrate with the deskhop into the device I'd like to put together, but didn't find any.

The projects I found are:
* https://hackaday.com/2020/02/02/a-tiny-usb-hub-for-all-your-hardware-modding-needs/
* https://www.olimex.com/Products/USB-Modules/USB-NeoHub/open-source-hardware
* https://www.microtype.io/open-source-usb-hub-design-with-kicad/
* commercial non-open product example: https://icybox.de/product/hubs/IB-HUB1426-CPD

I'd be thankful for any suggestions, opinions and help.

hrvach, 2024-03-07:
I'd advise to get an external hub with features you need for the mobile (USB-PD, video output, usb 3.0 ports etc) and connect the box output to one of its usb2 ports, then test if everything works to your satisfaction. If you need a single box solution, perhaps extending/redesigning the case to incorporate a hub of your liking might be the easiest thing?

chri2, 2024-03-07:
Yes, that is the way I'll take. I'll report back whether this works.

But in the end it would be nice to have the usb-c hub part also in open hardware and maybe in pcb format that could find its place in the same casing.

chri2, 2024-03-09:
I tried now with two usb-c hubs: A Hoyoki CB-C71 and an Icy Box IB-HUB1426-CPD. Both worked nice and the mobile phone I used as a second device got charged. The usb-c port on the icy box one works for e.g. a NitroKey I regularly need.

The icy box usb-hub - even though quiet expensive and not open hardware - would fit nicely with the project. Its longest side is ~9cm. If I'd make the deskhop case a bit bigger to hold the hub on top of it, it would fit nicely.

Still: I'd rather prefer to add an open hardware board for the usb-hub.

**If anybody comes along this issue and knows an open hardware usb-c board with the above specs (or those of the icy box one) I'd appreciate very much reading about it!**