hrvach/deskhop #59: Feature request: Mouse accelleration

github-k8n, 2024-03-07:
Deskhop is working great, however the fact that there is no accelleration built in is a bit of a bummer. (due to it being an absolute pointing device, i can't set it in the OS itself.

Maybe some simple lookup table or array in the code could be used to determine the mouse movement.
Calculate the (vector) speed and then look up the appropriate multiplication factor..
10 -> 1
20 -> 1.5
30 -> 2

That way everyone could set up their own speed curve.

(Accelleration helps a lot when just flicking the cursor into another screen section but still being able to fine control the movement.)

github-k8n, 2024-03-07:
Just to clarify, I am NOT talking about switching between computers, that is working great as-is (i set it to 100).
It is about the acceleration of the mouse cursor within the screen.
That's one of the things you just don't notice until it isn't there anymore..

hrvach, 2024-03-07:
This should not be hard to implement, I just never noticed this as something I miss (but then again I was using a 2$ mouse and thought that was fine too 😀 )

Fallingonion, 2024-03-08:
I'd like to second this request - I've just built my first deskhop and it's *awesome*, but mouse acceleration would be great!

hrvach, 2024-04-01:
Implemented. Please check and let me know if you have a better acceleration curve to suggest.

closed 2024-04-01 by hrvach

github-k8n, 2024-04-01:
Thank you! Works like a charm!
(Now i just have to get used again to mouse accelleration after working without it for a while since using deskhop :P )