hrvach/deskhop #60: mouse speed x/y

chri2, 2024-03-07:
Just finished my first deskhop - **so cool**!

I use a Linux Desktop with dual screen on board A and a Linux Phone Desktop on board B.

I have the following in my `user_config.h`:

```
/* Output A values */
#define MOUSE_SPEED_A_FACTOR_X 1
#define MOUSE_SPEED_A_FACTOR_Y 128

/* Output B values */
#define MOUSE_SPEED_B_FACTOR_X 128
#define MOUSE_SPEED_B_FACTOR_Y 1
```

And still on the dual screen desktop on board A the horizontal movement is really fast and the vertical movement too slow.

On the Linux phone horizontal movement still could be a bit faster and vertical movement is just fine.

Did I miss(interpret) something or is it just that the size of the phone (board b) screen using 720x1440 and the desktop (board a) screens 1920x1080 + 2560x1440 side by side is to different?

chri2, 2024-03-08:
Just finished my 2nd deskhop and found that this one programmed with the same uf2 files behaves more like expected. Wit the above mentioned settings it is nearly impossible to move the mouse in speed=1 direction and nearly impossible to stop at the right spot with the mouse in the speed=128 direction.

What could be the reason for the fault of the first board I soldered? The digital insulator? A problem at the usb-a connectors?

hrvach, 2024-03-08:
I think specific resolutions shouldn't matter; absolute movements work in an "abstract" coordinate space of (0,0) to (32767, 32767) which then maps to a physical one depending on the actual screen resolution. How this is done is left to the operating system and I haven't noticed a difference between vertical and horizontal movement speeds with default options (but then again, I think a 2$ mouse is fine so my opinion should not be trusted).

First thing - try resetting flash config using Right Shift + F12 + D

speed = 128 would probably be waaaay too fast, try picking the ratio of X and Y speeds that matches the ratio of the screen coordinates if you need different speeds, so if the default is let's say 15 in Y direction and your width is 3 times your height, you could pick 5 in X direction. For the phone maybe try 8 in Y direction and 16 in X direction?

If leds flash back and forth when you plug in the mouse and keyboard, isolator is good.

I haven't seen a similar issue before so I'm still thinking about what might cause it. Let me know.

chri2, 2024-03-09:
> First thing - try resetting flash config using Right Shift + F12 + D

This solved the issue immediately.

The only difference I'm aware of in the setup procedure of the two deskhop has been that I flashed the first one (the one with the problem with the mouse speeds) first with the pre-build image and than later compiled an image and flashed that one.

The second one I flashed after building directly with an image I had build (the one with the speed=1 and speed=128 settings) and it worked as expected and described above. Changing the settings in `user_config.h` to something else than worked after re-compiling and flashing.

chri2, 2024-03-09:
P.D. - the hint to reset the configuration when encountering problems could end up in the troubleshooting wiki page.

closed 2024-03-09 by chri2