hrvach/deskhop #61: media keys not working

chri2, 2024-03-08:
I'm using a Cherry Strait 3.0 keyboard. It seems that the media keys for `<play/pause>`, `<volume down>`, `<mute>`, `<volume up>` are represented by there own input device.

dmesg when connecting:
```
[406511.327571] usb 1-7: new low-speed USB device number 73 using xhci_hcd
[406511.493065] usb 1-7: New USB device found, idVendor=046a, idProduct=0180, bcdDevice= 1.03
[406511.493082] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[406511.493090] usb 1-7: Product: wired keyboard
[406511.493096] usb 1-7: Manufacturer:  
[406511.505538] input:   wired keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:046A:0180.0030/input/input134
[406511.564462] hid-generic 0003:046A:0180.0030: input,hidraw3: USB HID v1.11 Keyboard [  wired keyboard] on usb-0000:00:14.0-7/input0
[406511.572461] input:   wired keyboard Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:046A:0180.0031/input/input135
[406511.631995] input:   wired keyboard System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:046A:0180.0031/input/input136
[406511.632572] hid-generic 0003:046A:0180.0031: input,hidraw5: USB HID v1.11 Device [  wired keyboard] on usb-0000:00:14.0-7/input1
```

When trying `evtest` I get a selection of three devices for the keyboard:

```
/dev/input/event11:	  wired keyboard
/dev/input/event12:	  wired keyboard Consumer Control
/dev/input/event13:	  wired keyboard System Control
```

The _normal_ keyboard on `event11` is what works using deskhop. Events for media keys can be found on `event12`:

```
Input device ID: bus 0x3 vendor 0x46a product 0x180 version 0x111
Input device name: "  wired keyboard Consumer Control"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
[...]
Event: time 1709889093.911391, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00cd
Event: time 1709889093.911391, type 1 (EV_KEY), code 164 (KEY_PLAYPAUSE), value 1
Event: time 1709889093.911391, -------------- SYN_REPORT ------------
Event: time 1709889094.023375, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00cd
Event: time 1709889094.023375, type 1 (EV_KEY), code 164 (KEY_PLAYPAUSE), value 0
Event: time 1709889094.023375, -------------- SYN_REPORT ------------
Event: time 1709889095.679357, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1709889095.679357, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1709889095.679357, -------------- SYN_REPORT ------------
Event: time 1709889095.783314, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1709889095.783314, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1709889095.783314, -------------- SYN_REPORT ------------
Event: time 1709889096.431247, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1709889096.431247, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1709889096.431247, -------------- SYN_REPORT ------------
Event: time 1709889096.559251, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1709889096.559251, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1709889096.559251, -------------- SYN_REPORT ------------
Event: time 1709889097.039276, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1709889097.039276, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1709889097.039276, -------------- SYN_REPORT ------------
Event: time 1709889097.103318, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1709889097.103318, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1709889097.103318, -------------- SYN_REPORT ------------
```

These are not relayed when the keyboard is connected through deskhop.

I didn't find out what the third `event13` device is used for - maybe it's just there even though no keys are connected.

hrvach, 2024-03-08:
Try dumping hid descriptors first, I'm guessing this is appearing as something else other than keyboard... I don't have anything to test on while trying to provide support for this... I could try making a blind attempt at fixing it but that's not easy without testing.

Fallingonion, 2024-03-08:
I have the same issue with my Logitech G815 - the volume control doesn't work tthrough the deskhop.
I'm on windows so have attached the output from hidsharp.test.exe - if this isn't useful let me know and I'll find a linux machine and use usbhid-dump.
[hidsharp_logitech_g815.txt](https://github.com/hrvach/deskhop/files/14541748/hidsharp_logitech_g815.txt)

chri2, 2024-03-09:
I hope this will help:

```
me@linux:~/deskhop/usbhid-dump$ lsusb | grep CHERR
Bus 001 Device 026: ID 046a:0180 CHERRY Strait 3.0
me@linux:~/deskhop/usbhid-dump$ sudo src/usbhid-dump -a 1:26
001:026:001:DESCRIPTOR         1709985170.815236
 05 0C 09 01 A1 01 85 01 19 00 2A FF 03 15 00 26
 FF 03 95 01 75 10 81 00 C0 05 01 09 80 A1 01 85
 02 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95
 05 81 01 C0

001:026:000:DESCRIPTOR         1709985170.822942
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 03 75 01
 05 08 19 01 29 03 91 02 95 05 75 01 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 C0
```

chri2, 2024-03-29:
Does it help? Can I do anything else to help to debug the issue?

hrvach, 2024-03-29:
You have:

Usage Page (Consumer),              ; Consumer (0Ch)
Usage (Consumer Control),           ; Consumer control (01h, application collection)
Collection (Application),
    Report ID (1),
    Usage Minimum (00h),
    Usage Maximum (03FFh),
    Logical Minimum (0),
    Logical Maximum (1023),
    Report Count (1),
    Report Size (16),
    Input,
End Collection,

and I don't have anything similar to test with / develop on. Supporting a heterogeneous device ecosystem with every keyboard having a wildly different set of keys and implementations is quite tricky. I have a few ideas how to potentially address this, but they are either shitty, complicated or both :) 

Any good suggestion and some domain knowledge on how to provide better support for these additional keys is more than welcome!

chri2, 2024-04-01:
I'm not a developer nor do I have insight into HID. I naïvely came up with the thought that those keys in question are detected by Linux (Debian) without the need for any configuration. Could there be learned something from that how to handle the kind of hid used here?

A side question: does anybody know of a keyboard offering keys for mute on/off, volume up/down, play/pause that works with deskhop (and Linux)?

raldnor, 2024-04-10:
Just to add another keyboard as well, this is a Glorious GMMK Pro with Via firmware:
```
$ lsusb|grep lor
Bus 001 Device 009: ID 320f:5044 Glorious GMMK Pro ANSI

```
 $ sudo usbhid-dump -a 1:9 
001:009:002:DESCRIPTOR         1712774528.547981
 05 01 09 02 A1 01 85 02 09 01 A1 00 05 09 19 01
 29 08 15 00 25 01 95 08 75 01 81 02 05 01 09 30
 09 31 15 81 25 7F 95 02 75 08 81 06 09 38 15 81
 25 7F 95 01 75 08 81 06 05 0C 0A 38 02 15 81 25
 7F 95 01 75 08 81 06 C0 C0 05 01 09 80 A1 01 85
 03 19 01 2A B7 00 15 01 26 B7 00 95 01 75 10 81
 00 C0 05 0C 09 01 A1 01 85 04 19 01 2A A0 02 15
 01 26 A0 02 95 01 75 10 81 00 C0 05 01 09 06 A1
 01 85 06 05 07 19 E0 29 E7 15 00 25 01 95 08 75
 01 81 02 05 07 19 00 29 EF 15 00 25 01 95 F0 75
 01 81 02 05 08 19 01 29 05 95 05 75 01 91 02 95
 01 75 03 91 01 C0

001:009:001:DESCRIPTOR         1712774528.550978
 06 60 FF 09 61 A1 01 09 62 15 00 26 FF 00 95 20
 75 08 81 02 09 63 15 00 26 FF 00 95 20 75 08 91
 02 C0

001:009:000:DESCRIPTOR         1712774528.553976
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 95 08 75 01 81 02 95 01 75 08 81 01 05 07 19 00
 29 FF 15 00 26 FF 00 95 06 75 08 81 00 05 08 19
 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01 C0
```

The Glorious creates a lot of devices:
```
/dev/input/event4:	Glorious GMMK Pro ANSI Mouse
/dev/input/event8:	Glorious GMMK Pro ANSI System Control
/dev/input/event10:	Glorious GMMK Pro ANSI Consumer Control
/dev/input/event12:	Glorious GMMK Pro ANSI Keyboard
/dev/input/event13:	Glorious GMMK Pro ANSI
```
Device 10 has the media keys:
```
Input driver version is 1.0.1
Input device ID: bus 0x3 vendor 0x320f product 0x5044 version 0x111
Input device name: "Glorious GMMK Pro ANSI Consumer Control"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 1 (KEY_ESC)
    Event code 28 (KEY_ENTER)
    Event code 74 (KEY_KPMINUS)
    Event code 78 (KEY_KPPLUS)
    Event code 103 (KEY_UP)
    Event code 105 (KEY_LEFT)
    Event code 106 (KEY_RIGHT)
    Event code 108 (KEY_DOWN)
    Event code 110 (KEY_INSERT)
    Event code 111 (KEY_DELETE)
    Event code 113 (KEY_MUTE)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
    Event code 116 (KEY_POWER)
    Event code 119 (KEY_PAUSE)
    Event code 120 (KEY_SCALE)
    Event code 128 (KEY_STOP)
    Event code 130 (KEY_PROPS)
    Event code 131 (KEY_UNDO)
    Event code 133 (KEY_COPY)
    Event code 134 (KEY_OPEN)
[... 139 lines cut ...]
Event: time 1712774996.959313, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1712774996.959313, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 1
Event: time 1712774996.959313, -------------- SYN_REPORT ------------
Event: time 1712774996.969293, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e9
Event: time 1712774996.969293, type 1 (EV_KEY), code 115 (KEY_VOLUMEUP), value 0
Event: time 1712774996.969293, -------------- SYN_REPORT ------------
Event: time 1712774998.178300, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1712774998.178300, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 1
Event: time 1712774998.178300, -------------- SYN_REPORT ------------
Event: time 1712774998.188294, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00ea
Event: time 1712774998.188294, type 1 (EV_KEY), code 114 (KEY_VOLUMEDOWN), value 0
Event: time 1712774998.188294, -------------- SYN_REPORT ------------
Event: time 1712775000.094289, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1712775000.094289, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1712775000.094289, -------------- SYN_REPORT ------------
Event: time 1712775000.299273, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1712775000.299273, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1712775000.299273, -------------- SYN_REPORT ------------
Event: time 1712775001.489266, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1712775001.489266, type 1 (EV_KEY), code 113 (KEY_MUTE), value 1
Event: time 1712775001.489266, -------------- SYN_REPORT ------------
Event: time 1712775001.675263, type 4 (EV_MSC), code 4 (MSC_SCAN), value c00e2
Event: time 1712775001.675263, type 1 (EV_KEY), code 113 (KEY_MUTE), value 0
Event: time 1712775001.675263, -------------- SYN_REPORT ------------
```

hrvach, 2024-04-15:
It's very hard for me to debug remotely, but I've decided to open PP donation fund to purchase 1-2 keyboards/mice with those fancy buttons and rotary knobs, NKRO etc. That would enable me to at least try getting this to work for more people and improve the overall user experience...

raldnor, 2024-04-15:
Cool, I just transferred some funds. Hopefully it will help to fix this issue :). Thanks for all the effort so far, it seems to develop in a mature product.

hrvach, 2024-04-15:
> Cool, I just transferred some funds. Hopefully it will help to fix this issue :). Thanks for all the effort so far, it seems to develop in a mature product.

Thank you, you really overdid it. If anyone has any cheap keyboards with knobs/media buttons to recommend, that would be great.

raldnor, 2024-04-15:
Well, a nice approach could be to start with QMK capable keyboards, but they're not exactly cheap. Perhaps a Keychron, like this one:
https://www.keychron.com/products/keychron-v1-max-qmk-via-wireless-custom-mechanical-keyboard
or:
https://www.keychron.com/products/keychron-v2-qmk-custom-mechanical-keyboard?variant=40196919459929
The wireless with the keychron is not mandatory, you can still use them with an USB cable.

Another option could be to buy a logitech with media keys, but I don't have such a keyboard myself so it would be hard for me to recommend one.

hrvach, 2024-04-15:
I have a keyboard I like and use, so really don't need anything expensive/fancy. Just something that will replicate the not working things so I can try to fix it and improve overall user experience by having the buttons work (worst case some configuration/mapping is needed). Also, rewriting the hid parser in a bit more universal way is something I need to do and that should take care of things like NKRO support. From that point on, most keyboards and mice should work sufficiently well. Then further improve mac/win workarounds and basic usability should be there.

Some things are also ugly and dumb, those need a rework so I can not be ashamed of them anymore, but as long as it kinda works... :)

hrvach, 2024-05-01:
Anyone up for some testing? I have partial support for media keys (no custom array ones yet, mostly generic variable ones should work).

[deskhop-media-keys.zip](https://github.com/hrvach/deskhop/files/15181094/deskhop-media-keys.zip)

Fallingonion, 2024-05-01:
I'll give it a try, thanks!

raldnor, 2024-05-01:
Wow awesome! Just tried it with my keychron (with QMK) and it works great!

hrvach, 2024-05-01:
> Wow awesome! Just tried it with my keychron (with QMK) and it works great!
> 
> I do have another bug though; in QMK I have programmed that the fn-key+ESC shuts down the computer (well, the dialog pops up), that key combination does not work unfortunately.

That might be the system control section, also doable. I finally got my keychron with those knobs and media keys, so now I somewhat understand how they work :)

jalmeroth, 2024-05-02:
> Anyone up for some testing? I have partial support for media keys (no custom array ones yet, mostly generic variable ones should work).
> 
> [deskhop-media-keys.zip](https://github.com/hrvach/deskhop/files/15181094/deskhop-media-keys.zip)

@hrvach would you mind sharing the code in a branch or something? It doesn't work for me, but I would like to take a look what's different on my Logitech. Thanks for considering!

hrvach, 2024-05-02:
> @hrvach would you mind sharing the code in a branch or something? It doesn't work for me, but I would like to take a look what's different on my Logitech. Thanks for considering!

I'll publish it soon after cleaning it up, can you please do usbhid-dump -e all and press a couple of media keys for me?
I need the consumer control descriptor (assuming this one is yours from a previous post)

Usage Page (Consumer),                  ; Consumer (0Ch)
Usage (Consumer Control),               ; Consumer control (01h, application collection)
Collection (Application),
    Report ID (3),
    Report Count (2),
    Report Size (16),
    Logical Minimum (1),
    Logical Maximum (767),
    Usage Minimum (Consumer Control),   ; Consumer control (01h, application collection)
    Usage Maximum (02FFh),
    Input,
End Collection,

hrvach, 2024-05-02:
@jalmeroth any difference with 
[deskhop-media-keys-2.zip](https://github.com/hrvach/deskhop/files/15194747/deskhop-media-keys-2.zip)
?

jalmeroth, 2024-05-03:
> @jalmeroth any difference with [deskhop-media-keys-2.zip](https://github.com/hrvach/deskhop/files/15194747/deskhop-media-keys-2.zip) ?

Standard keys are working but media keys are not send. Is report mode enabled? Do you might wanna create a debug build so we can see what's happening on the deskhop side in the console?

hrvach, 2024-05-03:
> @hrvach standard keys are working but media keys are not send. Is report mode enabled? Do you might wanna create a debug build so we can see what's happening on the deskhop side in the console?

@jalmeroth consumer control is likely not sent in boot mode with that particular model ... :-| should start working once I get nkro support done.

chri2, 2024-05-15:
Just checked using 0.52 with a Cherry STRAIT 3.0 BLACK. It seems that all keys are working now. The keys I found formerly not working were: 

* `circle with a play triangle inside it` → opens in Gnome/Debian the gnome-settings
* `play/pause`
* `-` (above number keypad) → volume down
* `+` (above number keypad) → volume up
* `speaker symbol` → mute/unmute

Thanks for the great work! As far as I'm concerned this can be closed.

hrvach, 2024-05-26:
cc: @chri2  apologies to bug you, I'd appreciate some [help with testing does this still upcoming release still work for your use case?](https://github.com/hrvach/deskhop/discussions/94)

closed 2024-05-26 by hrvach