hrvach/deskhop #68: Issue with Razer Basilisk V3 Programmable Buttons Not Working Across Systems via Deskhop

ASUDCT, 2024-03-19:
Firstly, I would like to express my admiration for the project. The idea of a hardware-level solution for sharing a keyboard and mouse across different computers and operating systems is truly fantastic. However, I've encountered an issue that I hope can be addressed.

I use a Razer Basilisk V3 mouse, which features 10 programmable buttons. These buttons are crucial to my workflow, as I have them set up with various keyboard shortcuts and macros. The mouse allows for the storage of up to 5 onboard profiles, with all configuration details directly written into the mouse's memory. This setup works flawlessly on individual systems running either Windows 11 or Ubuntu 20.04, without the need for any additional driver installations.

The problem arises when I use Deskhop to share my mouse across different computers. Once Deskhop is in use, none of the custom functionalities programmed into the mouse's buttons work on any computer. To diagnose the issue, I used usbhid-dump on Ubuntu 20.04 and noticed that there are no responses registered from any of the programmable buttons—except for the standard left click, right click, middle click, scroll, and mouse movement.

hrvach, 2024-03-20:
Hey! Thanks for the feedback and I'm sorry to hear about your mouse issues. Please be so kind to provide usb hid descriptor (https://github.com/DIGImend/usbhid-dump or install from package if one available).

ASUDCT, 2024-04-07:
> Hey! Thanks for the feedback and I'm sorry to hear about your mouse issues. Please be so kind to provide usb hid descriptor (https://github.com/DIGImend/usbhid-dump or install from package if one available).
I am not very familiar with USB hid descriptors. This is the descriptor that I displayed after using my mouse to implement four functions: "LCtrl+C", "LCtrl+V", "LCtrl", and "LShift" (if I remember correctly)
[image]