hrvach/deskhop #70: Directory 'path/to/pico-sdk' not found error appears oddly.

Walkline80, 2024-03-19:
Here is my operation steps:

The `deskhop` and `esp32` folders are in sibling directories.

```bash
$ cd deskhop
$ PICO_SDK_PATH=/esp32/micropython/lib/pico-sdk/ cmake -S . -B build
```

Of cause the cmake give me an error message:

```bash
CMake Error at pico_sdk_import.cmake:52 (message):
  Directory '/esp32/micropython/lib/pico-sdk' not found
```

And then correct the wrong path:

```bash
$ PICO_SDK_PATH=~/esp32/micropython/lib/pico-sdk/ cmake -S . -B build
```

But, the cmake still give me an error message, exactly the same as the above, even the `~` not appears.

Then, I have to remove the `build` folder inside `deskhop` and execute the command again:

```bash
$ rm -rf build
$ PICO_SDK_PATH=~/esp32/micropython/lib/pico-sdk/ cmake -S . -B build
```

This time the cmake will success, I think it's a cmake script issue?

hrvach, 2024-03-22:
Providing the wrong path in the PICO_SDK_PATH will make CMakeCache.txt cache it, I think you can provide --fresh if you want it to update the variable.

hrvach, 2024-03-24:
pico sdk is now bundled and you no longer need to provide sdk path variable, so considering resolved.

closed 2024-03-24 by hrvach

Walkline80, 2024-03-25:
Thanks for the update, this improves a lot of efficiency and also saves a lot of hard drive space.