hrvach/deskhop #73: Building on Arch Linux

jalmeroth, 2024-03-24:
Hi,

I tried to build the project on Arch Linux and received the following error:
```
[100%] Linking CXX executable board_B.elf
/usr/lib/gcc/arm-none-eabi/13.2.0/../../../../arm-none-eabi/bin/ld: CMakeFiles/board_B.dir/src/setup.c.obj: in function `initial_setup':
setup.c:(.text.initial_setup+0xc0): undefined reference to `tuh_hid_set_default_protocol'
Memory region         Used Size  Region Size  %age Used
           FLASH:       77432 B      2044 KB      3.70%
    FLASH_CONFIG:           0 B         4 KB      0.00%
             RAM:       92308 B       256 KB     35.21%
       SCRATCH_X:          2 KB         4 KB     50.00%
       SCRATCH_Y:           0 B         4 KB      0.00%
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/board_B.dir/build.make:1481: board_B.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1650: CMakeFiles/board_B.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
```

I used https://aur.archlinux.org/packages/pico-sdk to install the Pico SDK.

Unfortunately, it looks like the provided tinyusb version is outdated?

So I checked-out the tinyusb project manually with `git clone https://github.com/hathach/tinyusb` and used these commands to build:

```sh
PICO_BOARD=pico PICO_SDK_PATH=/path/to/pico-sdk PICO_TINYUSB_PATH=/path/to/tinyusb cmake -S . -B build
cmake --build build
```

hrvach, 2024-03-24:
I agree with you, this version mismatch is a bid annoying - I will go with bundling a minimal sdk + tinyusb to provide the needed components to build with no errors, updated the repo. Please check if this resolves your issue.

Cheers!

jalmeroth, 2024-03-24:
Yep, that works! Thank you.

(Of course one still needs the dependency/tooling like `arm-none-eabi-binutils arm-none-eabi-gcc arm-none-eabi-newlib cmake`)

closed 2024-03-24 by jalmeroth