hrvach/deskhop #74: more classic kvm button mode

Dami1-59, 2024-03-24:
Hello everyone and congratulations for this revolutionary project compared to classic KVM which causes disconnections and reconnections

but unfortunately, I cannot use your system because it has 3 screens for simracing and a PC with a 4th screen for navigation

I would like to ask you if you are thinking of creating a switching mode by keyboard shortcut or physical button without taking into account mouse positioning

thank you and good luck

Bonjour à tous et félicitation pour ce projet révolutionnaire comparé au KVM classique qui provoque des déconnexion reconnexion

mais malheureusement, je ne peux pas utiliser votre système car possédant 3 écran pour le simracing et un pc avec un écran 4éme écran pour naviguer 

je voudrais vous demander si vous pensez créer un mode de basculement par raccourci clavier ou bouton physique sans prise en conte du positionnement de souris

merci et bonne continuation

Fallingonion, 2024-03-25:
Deskhop can already do this - the default key to switch is caps lock, but you can define another key if you wish in user_config.h.
(I have it set to use the Pause key.)

Dami1-59, 2024-03-30:
thank you for the answer

I will see about ordering the equipment

closed 2024-03-30 by hrvach

torypages, 2024-05-04:
This is great. The monitor edge detection seems possibly problematic, but I'm very happy with only being able to switch via a keyboard shortcut. And of course that keyboard shortcut would change both the mouse and keyboard.