hrvach/deskhop #79: Custom made keyboard not working

lvntbkdmr, 2024-03-29:
Hey, I have a custom made keyboard with ZMK firmware (normally it works as bluetooth keyboard, but I connected it via cable). When I directly connect the keyboard to the host with cable, it works as expected and here is my hid dump

```
1D50 615E: ZMK Project - Corne
DESCRIPTOR:
  05  01  09  06  a1  01  85  01  05  07  19  e0  29  e7  15  00
  25  01  75  01  95  08  81  02  05  07  75  08  95  01  81  03
  05  07  15  00  26  ff  00  19  00  29  ff  75  08  95  06  81
  00  c0  05  0c  09  01  a1  01  85  02  05  0c  15  00  26  ff
  0f  19  00  2a  ff  0f  75  10  95  06  81  00  c0
  (77 bytes)
```

However, when I connect this to the Deskhop, it does not work. Is there a problem with my hid report? There is a flag which enables USB Boot mode while compiling, I also enabled that flag, but still non functioning.

hrvach, 2024-03-29:
This looks pretty normal, some keyboards seem to ignore boot mode altogether. Does this work for you? [test_fw.zip](https://github.com/hrvach/deskhop/files/14803692/test_fw.zip)

lvntbkdmr, 2024-03-29:
> This looks pretty normal, some keyboards seem to ignore boot mode altogether. Does this work for you? [test_fw.zip](https://github.com/hrvach/deskhop/files/14803692/test_fw.zip)

ah yes, it works as expected. What did you do ? :D

lvntbkdmr, 2024-03-29:
ah ok I got it, can you provide the diff file or patch etc. so I can apply that workaround to my own fork for now? after you implement that change on this repo, I can rebase to that

hrvach, 2024-03-29:
> ah ok I got it, can you provide the diff file or patch etc. so I can apply that workaround to my own fork for now? after you implement that change on this repo, I can rebase to that

Try adding:
```
    if (length == KBD_REPORT_LENGTH + 1)
        keyboard_report = (hid_keyboard_report_t *)(raw_report + 1);
```

to keyboard.c in process_keyboard_report after the first if block. Let me know if it still works OK and I'll merge it with the master.

lvntbkdmr, 2024-03-29:
> Try adding:
>
> ```
>     if (length == KBD_REPORT_LENGTH + 1)
>         keyboard_report = (hid_keyboard_report_t *)(raw_report + 1);
> ```
>
> to keyboard.c in process_keyboard_report after the first if block. [...]

Tried that, yes it works

closed 2024-04-01 by hrvach