hrvach/deskhop #81: possibility to support android.

liuwentai, 2024-03-31:
Impressive project that I was looking for long... I can now use mac and pc side by side with only one keyboard/mouse, but I also want to do the same with android tablet, turned out that keyboard functioning well , however  mouse didn't show up. Switching between android and mac  is OK , just mouse on android is not responding. any advice ?

hrvach, 2024-03-31:
I have honestly never tried Android, but I'm pretty sure it could be made to work. I'll try to see if I still have an old Android phone lying around and give it a go...

waiwai9930, 2024-12-18:
Has it been resolved? My Android device also cannot use a mouse, but my keyboard can only type, and the function keys cannot be used.

hrvach, 2024-12-19:
> Has it been resolved? My Android device also cannot use a mouse, but my keyboard can only type, and the function keys cannot be used.

To an extent. I think you should be able to switch to gaming mode and that should work on Android. I've looked into it originally and it seems to not support absolute mouse in any way, but there was some recent development with alternative input descriptors (touchpad) which I believe would work great with Android.

That should be one of the upcoming changes, hopefully soon.

waiwai9930, 2024-12-21:
Thank you for your reply. I found that my Android device can work when plugged into the A port, but not with the B port output. This is already very helpful for me. There is a dead zone for the mouse on the edge of the Android device. I will configure it in detail when I get home.

waiwai9930, 2024-12-21:
Thank you for your answer. Today, a mistake in the configuration unexpectedly made the Android tablet usable. I changed the configuration to Windows for the Android direction and set the screen to two, and surprisingly, the mouse started working normally. The keyboard also works fine. This is really cool.

00xdF, 2026-09-17:
> Thank you for your answer. Today, a mistake in the configuration unexpectedly made the Android tablet usable. I changed the configuration to Windows for the Android direction and set the screen to [...]

Hi yvfx, have u tried it on iOS

liuwentai, 2026-09-17:
尊敬的客户： 您的报装材料已成功发送，我们将尽快审核，请您耐心等待，期间如有任何疑问，请联系您的专属客户经理14795510147。

yvfx, 2026-09-17:
> Hi yvfx, have u tried it on iOS

Sorry, I haven't tested compatibility on iOS devices.