hrvach/deskhop #82: Screenlock on macOS doesn't work

jalmeroth, 2024-04-02:
Hi there,

I noticed that the new feature to lock both devices (`Right ALT + Right Shift + L`) at the same time didn't work for me on macOS.

Deskhop sends `Ctrl + Alt + Q` but it should be `Ctrl + Cmd + Q`: 
[image]

Has this successfully been tested by others? If not I can prepare a PR for this.

Regards,
Jan

hrvach, 2024-04-02:
Did you configure OS you have connected on each output so the box know which shortcut to send?

jalmeroth, 2024-04-02:
Yes, I did and I successfully used `KEYBOARD_MODIFIER_LEFTGUI` instead of `KEYBOARD_MODIFIER_LEFTALT`.

hrvach, 2024-04-02:
Oh, this might be due to the fact my command and option keys are swapped :)

closed 2024-04-02 by hrvach