hrvach/deskhop #84: HID remapping

tomByrer, 2024-04-03:
Just an idea; offer key & mouse remapping.  Handy for missing/reprogramming [media keys](https://github.com/hrvach/deskhop/issues/61), foreign characters, macros, etc

https://github.com/jfedor2/hid-remapper/

jalmeroth, 2024-04-03:
I had the same thought, when taking screenlock away from @hrvach in my PR yesterday. 🫣

hrvach, 2024-04-03:
This is a good idea. I'm currently trying to focus primarily on getting the basic features working for most people - multiple desktops, weird keyboards, non-working media keys etc. Once that's done, hopefully more fun stuff can be implemented :)

tomByrer, 2024-04-04:
Yes, lower priority, though I wonder if adding this could help fix the "non-working media keys", or perhaps that code could give a clue.
eg: https://github.com/jfedor2/hid-remapper/issues/120#issuecomment-2004787215

hrvach, 2024-04-15:
It definitely could, I'm going to order a keyboard with some media keys to finally have some hardware to test/develop on. The expectation is that this would help me figure it out and provide support for your plays, pauses and volumes :)

chri2, 2024-04-15:
That's great news! I already was about to build a hid-remapper to get the keys back ;-). If you'd really only need it for testing I could send you one of [these](https://www.ebay.de/itm/312560384801). The media keys should be the same as the ones I asked about. Let me know if that'd help.

222Phoenix, 2025-03-14:
HID-remapper works with Deskhop, if anyone has both