hrvach/deskhop #87: Switching at the top/bottom of a screen

crablab, 2024-04-10:
Hey! :wave:  

Thanks so much for open sourcing this - it's an awesome project. 

Playing around and trying it out, it seems the only switches that are possible are going sideways between monitors? 

I wondered if it was possible to add configuration options so you can switch at the top/bottom of a monitor. In my use case, I have a MacOS machine with two monitors - one main, and one to the left (vertical). I want to put a laptop below the main Mac monitor and use Deskhop to switch between them but vertically (ie. on the bottom edge of the main monitor, and top edge of the laptop). I've attached an image of what I mean :) 

[image]

I don't think this is something currently supported? 

Read through some of the related issues (eg. https://github.com/hrvach/deskhop/issues/3, https://github.com/hrvach/deskhop/issues/56) and the recent changes to address more monitor configurations. I don't think this has been addressed elsewhere, but please do redirect me if I've inadvertently missed something. 

Also, if you've got any pointers I'd be happy to try and put together a PR to implement this!

hrvach, 2024-04-11:
This would be possible, but is not currently supported ... adding another "switching axis" makes sense, I'll try to put this on the roadmap and you are, by all means, more than welcome to help! :)

PasGour, 2024-09-19:
Thank you @hrvach for this great project !
I'd be interested in such feature, as I have a setup quite similar to @crablab's one.

bau-sec, 2025-01-05:
Another vote for this one, I have two wide screen monitors positioned one on top of the other. Thanks for this project, very useful.