hrvach/deskhop #91: (roadmap proposal) Text-based configuration management

kpfleming, 2024-04-14:
As noted in #72, it would be useful for users to be able to manage their configuration without having to compile/recompile firmware. This would also allow users to upgrade to new firmware releases directly using builds from the repository (and via LVFS).

I've been working on a plan for this, which includes a number of steps:
1. Implement firmware and configuration upload/download via DFU (#78)
2. Cleanup config_t and ensure that it includes *all* of the user configuration (including KBD_LED_AS_INDICATOR, HOTKEY_TOGGLE, etc. which are currently compiled directly into the code) ([implemented](https://github.com/kpfleming/deskhop/tree/config-cleanup))
3. Add a host-side tool which can import/export configuration using an INI-style text file ([implemented](https://github.com/kpfleming/deskhop/tree/config-ini), [example INI file](https://github.com/kpfleming/deskhop/blob/config-ini/config.ini.example))
4. Add INI upload/download via DFU
5. Use INI export/import during the DFU firmware upgrade process to preserve configuration across firmware upgrades that change the configuration version

I'm using [minIni](https://github.com/compuphase/minIni) as the INI-file library, it's small enough and simple enough to be incorporated into the firmware without a major increase in size.

I waited to open this issue until I had items 1-3 already implemented and tested, to be certain that I actually *could* implement them :-)

All feedback welcome; if this approach is acceptable, I'll open followup PRs as each of the earlier ones are merged, eventually getting to completion at step 5.

hrvach, 2024-04-15:
I have an alternative idea for a friendly config without any external software, but need to try a proof of concept when I'm finally back from work trip - I'm curious if it would be feasible.

Sorry for being slow lately guys, but work is a bit crazy.

kpfleming, 2024-04-15:
> Great work! Excellent documentation in the ini-files as well. Would it be possible to protect the DFU ini upload with something like a password or pin, requiring a new firmware upload to reset such a password/pin?

All things are possible :-) It's a little challenging to use any sort of shared-key mechanism when the device doesn't have a user interface, though.

jalmeroth, 2024-04-15:
>  It's a little challenging to use any sort of shared-key mechanism when the device doesn't have a user interface, though.

What about pressing the BOOTSEL button instead?

kpfleming, 2024-04-15:
> I have an alternative idea for a friendly config without any external software, but need to try a proof of concept when I'm finally back from work trip - I'm curious if it would be feasible.

The approach I'm working on doesn't *require* any external software, but it was much easier to develop/debug the INI-handling code outside of the firmware. In addition, once it is in the firmware if the user attempts to upload an invalid configuration, there's no way for the firmware to report back errors to let them know what was wrong, so I think having an external tool that can be used to validate the configuration will be useful.

hrvach, 2024-06-23:
Latest branch/rc has a web UI, so hopefully configuration should be easier and more user friendly. It's far from ideal, but with some additional tweaking hopefully it will be able to fully replace the awkward keyboard shortcuts.

closed 2024-06-23 by hrvach