hrvach/deskhop #93: 8bitdo Retro Keyboard ultimnate tool

denbrazh, 2024-05-25:
Starts with error then crashed 
Error: 未捕获异常： at System.Number.ThrowOverflowOrFormatException(ParsingStatus status, TypeCode type) at System.Single.Parse(String s) at AdvanceSuper.Hid.HIDManage.readGamePadNewVersion(Int32& color, Int32 pid) at AdvanceSuper.SelectPlatform.readversion() at System.Threading.ThreadHelper.ThreadStart_Context(Object state) at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state) --- End of stack trace from previous location --- at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state) at System.Threading.ThreadHelper.ThreadStart()未捕获异常： at System.Number.ThrowOverflowOrFormatException(ParsingStatus status, TypeCode type) at System.Single.Parse(String s) at AdvanceSuper.Hid.HIDManage.readGamePadNewVersion(Int32& color, Int32 pid) at AdvanceSuper.SelectPlatform.readversion() at System.Threading.ThreadHelper.ThreadStart_Context(Object state) at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state) --- End of stack trace from previous location --- at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state) at System.Threading.ThreadHelper.ThreadStart()未捕获异常： at System.Number.ThrowOverflowOrFormatException(ParsingStatus status, TypeCode type) at System.Single.Parse(String s) at AdvanceSuper.Hid.HIDManage.readGamePadNewVersion(Int32& color, Int32 pid) at AdvanceSuper.SelectPlatform.readversion() at System.Threading.ThreadHelper.ThreadStart_Context(Object state) at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state) --- End of stack trace from previous location --- at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state) at System.Threading.ThreadHelper.ThreadStart()

hrvach, 2024-05-25:
I'm sorry but I don't understand what this report is about - this seems like C# or something...

hrvach, 2024-06-23:
I'll assume this is due to some keyboard vendor tool for configuration, none of these work by design due to security concerns. Apologies for the inconvenience!

closed 2024-06-23 by hrvach