hrvach/deskhop #97: Updated Bill of Materials (BOM)

cornfeedhobo, 2024-06-06:
The current bill of materials is targeted at users that are already comfortable navigating suppliers. It would be nice if there was an updated BOM with **concrete** links to both DigiKey and Mouser.

cklimbac, 2024-06-07:
I got all the material for this project from ali at a reasonable price. So for this project there is no need to buy from dedicated distributors.

maximumrozema, 2024-06-07:
@cklimbac,
Do you mind adding AliExpress links for each item on the list? 
I think that's what @cornfeedhobo is asking for.

hrvach, 2024-06-07:
I'll try to update it when I find a few mins, currently working on a major revision to include a single firmware image for both boards, web UI for configuration and firmware staging/sharing. Also managed to get DMA UART support so this should leave more cycles for USB and other stuff.

cklimbac, 2024-06-08:
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/CKLIMB~1/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/CKLIMB~1/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#467886" vlink="#96607D">

Prices has been updated and links added, but keep in mind prices may change fast.
All prices includes shipping costs to my country, so it could be different to your destination.

Component | Name | Part | Qty | Purchase Qty | Purchase Price | Purchase   Price/pcs | Costs | URL
-- | -- | -- | -- | -- | -- | -- | -- | --
IC | U1, U2 | Raspberry Pi Pico | 2 | 10 | 25,16 | 2.516 | 5.032 | https://www.aliexpress.com/item/1005006193611752.html
IC | U4 | ADuM1201BRZ | 1 | 30 | 8.13 | 0.271 | 0.271 | https://www.aliexpress.com/item/1005005359273862.html
IC | U3, U5 | TPD4E1U06DBVR | 2 | 100 | 18.28 | 0.1828 | 0.3656 | https://www.aliexpress.com/item/1005006519571202.html
Resistor | R1, R2, R3, R4 | Res 0805 SMD 27ohm | 4 | 1000 | 2.7 | 0.0027 | 0.0108 | https://www.aliexpress.com/item/1005006606170262.html
Capacitor | C1, C2 | Cap 0805 SMD 100nF | 2 | 500 | 3.32 | 0.00664 | 0.01328 | https://www.aliexpress.com/item/1005005550357125.html
Capacitor | C3, C4 | Cap 4.7uF SMD 0805 | 2 | 500 | 3.38 | 0.00676 | 0.01352 | https://www.aliexpress.com/item/1005005550357125.html
Connector | J1, J4 | USB-A PCB conn. | 2 | 500 | 12.5 | 0.025 | 0.05 | https://www.aliexpress.com/item/1005006832730801.html
Connector | J2, J3 | Headers 2.54 1x03 | 2 | 1333 | 10.13 | 0.0075975 | 0.015195 | https://www.aliexpress.com/item/1005002020614208.html
PCB | PCB | Printed circuit board | 1 | 10 | 8.53 | 0.853 | 0.853 | https://www.elecrow.com/
|   |   |   |   |   | **SUM** | **6.62** | **USD**

</body>

</html>

cornfeedhobo, 2024-06-08:
Here's what I came up with, but don't have confidence in my capacitor and resistor choices.

| Component      | Part                | Mfr #              | Qty |
|----------------|---------------------|--------------------|-----|
|     U1, U2     | Raspberry Pi Pico   | SC0915             | 2   |
|     J1, J4     | USB-A PCB conn.     | 22-28-4034         | 2   |
|       U4       | TI ISO7721DR        | ISO7721DR          | 1   |
|     C1, C2     | Cap 0805 SMD 100nF  | GRM155R71C104JA88J | 2   |
| R1, R2, R3, R4 | Res 0805 SMD 27ohm  | RC0805FR-1327RL    | 4   |
|     U3, U5     | TPD4E1U06DBVR       | TPD4E1U06DBVR      | 2   |
|     C3, C4     | Cap 4.7uF SMD 0805  | CL21B475KQQVPJE    | 2   |
|     J2, J3     | Headers 2.54 1x03   | 22-28-4034         | 2   |

maximumrozema, 2024-06-08:
Wow, Digikey isn't far off, except for the quanities. I don't really want 1000 extra parts, so I might not choose AliExpress for that reason. Here's my list from Digikey.ca.
https://www.digikey.ca/en/mylists/list/ZF5XS3NU7I

cornfeedhobo, 2024-07-04:
@maximumrozema this is tremendously helpful. It just offers more data points to compare and shop around while making sure you still get the part needed. Thank you!